Klipper3d/klipper #2671: Fysetc Cheetah 1.1 sensorless homing Y does not work

Magoo0876, 2020-03-31:
I tried to setup sensorless homing but Y wont work. 
As mihero #2042 allready reported the Uart config for the Cheetha is wrong. I allready searched for the error and found the reason. Wrong solderd resistors for the uart id selection. The MSB and LSB are swopped. 
See the picture:
[image]

So the correct sertting is:

X0 Y2 Z1 E3

hth Magoo
[klippy.log](https://github.com/KevinOConnor/klipper/files/4411733/klippy.log)

csutcliff, 2020-04-07:
My 1.1b is configured the same as OP.

KevinOConnor, 2020-04-08:
Thanks, I updated the default config (commit 4ea7b21).

-Kevin

closed 2020-04-15 by KevinOConnor