Klipper3d/klipper #2672: Klipper service suddenly deads at first layer

ZhenyaKa, 2020-04-01:
If not dead at the first layer print goes well to end

I have second klipper service and print at it is not running at that time.
[klippyMark.log.zip](https://github.com/KevinOConnor/klipper/files/4412231/klippyMark.log.zip)
[image]

ZhenyaKa, 2020-04-05:
I think I found the reason of deading klipper service.

It deads when I ajust z_offset via custom menu at the first layer
### menu tune ###
[menu __tune]
type: list
#enable: toolhead.is_printing
name: Tune
items:
    .__speed
    .__flow
    .__offsetz

[menu __tune __offsetz]
type: input
name: "Offset Z:{0:05.3f} "
parameter: gcode.homing_zpos
input_min: -5
input_max: 5
input_step: 0.01
realtime: true
gcode: SET_GCODE_OFFSET Z={0:.3f}

[menu __temp __hotend0_target]
type: input
enable: extruder.is_enabled
name: "{0:4.0f}"
parameter: extruder.target
input_min: 0
input_max: 270
input_step: 1
input_step2: 10
gcode: M104 T0 S{0:.0f}

KevinOConnor, 2020-04-08:
This appears to be some kind of USB failure.  The best advice we have is at: https://www.klipper3d.org/FAQ.html#i-keep-getting-random-lost-communication-with-mcu-errors

-Kevin

ZhenyaKa, 2020-04-08:
I have long print jobs - by 4 hours each all the time second day.

But deading service I noticed after ajusting via menu z_offset and all time at the first layer.

closed 2020-04-24 by KevinOConnor