Klipper3d/klipper #2673: Ability to automatically FIRMWARE_RESET after disconnect

PedroC1999, 2020-04-01:
My printer (Ender 3 Pro) remains powered from the USB off of my OctoPrint server when the printer itself is unplugged. This is undesirable to me (and many others) so we modify our USB cables so that the power port isnt connected. This doesn't effect the ability to print or use the machine in any way.

However it means that the serial port becomes inactive/disconnected overtime we power the machine down. This results in a loss of communication to the MCU and thus requires us to execute a firmware restart before connecting the printer again (in order for the MCU connection to be attempted).

I am wondering if a feature/option could be added to Klipper, so that after a MCU disconnect, it would automatically perform a firmware reset, so that the firmware is ready for the printers serial connection straight away whenever power to the power is connected again.

Many thanks, Pedro

user747, 2020-04-01:
possible workaround mentioned here? 
https://github.com/KevinOConnor/klipper/issues/835#issuecomment-441186301

installed OctoKlipper and noticed my CR-10S see's Klipper after turning on from a cold state.

PedroC1999, 2020-04-01:
That is quite an interesting solution, however I run everything in Docker
and I am unsure where to even begin adapting that.

KevinOConnor, 2020-04-02:
This looks like a dup of #729 .

-Kevin

user747, 2020-04-06:
> That is quite an interesting solution, however I run everything in Docker and I am unsure where to even begin adapting that.

yes a little research into udev inside containers. i think it should be possible

ld3300, 2020-04-13:
This is interesting, because I am running an Ender 3 with a MKS GEN L board.  I have cut the power line on my USB cable and have the pi powered separately from the rest of the printer.  The pi is driving a solid state relay so I can use Octoprint commands to turn printer, including the MKS GEN L Board on and off.  When I power the printer back on it connects automatically and doesn't require a firmware_restart.  I wonder what the difference is.

KevinOConnor, 2020-04-24:
I'm going to close this in favor of the existing #729 / #835 .

-Kevin

closed 2020-04-24 by KevinOConnor