Klipper3d/klipper #2675: Can not update MCU 'mcu' config as it is shutdown

4noxx, 2020-04-01:
hello 

at the moment i ve a problem to connect my printer. i got this message here:

15:44:54.882: // Can not update MCU 'mcu' config as it is shutdown
15:44:54.883: // Once the underlying issue is corrected, use the
15:44:54.883: // "FIRMWARE_RESTART" command to reset the firmware, reload the
15:44:54.883: // config, and restart the host software.
15:44:54.884: // Error configuring printer
15:44:54.884: !! Can not update MCU 'mcu' config as it is shutdown
15:44:54.885: ok
15:45:13.362: Slow command added:#FIRMWARE_RESTART

My Board is a skr mini e3

[klippy.log](https://github.com/KevinOConnor/klipper/files/4415536/klippy.log)

Magoo0876, 2020-04-01:
hi, after a quick look your dev id looks weird, normally it looks like that or similar: 
serial: /dev/serial/by-id/usb-Klipper_Klipper_firmware_12345-if00

Try to reflash the board again with the settings described in the config for the skr mini e3.

The firmware should be compiled for the
STM32F103 with a "28KiB bootloader". Also, select "Enable extra
low-level configuration options" and configure "GPIO pins to set at
micro-controller startup" to "!PC13".

The "make flash" command does not work on the SKR mini E3. Instead,
after running "make", copy the generated "out/klipper.bin" file to a
file named "firmware.bin" on an SD card and then restart the SKR
mini E3 with that SD card.

4noxx, 2020-04-01:
i think my id is right.
[mcu]
serial = /dev/serial/by-id/usb-Klipper_stm32f103xe_35FFDB053043553524770651-if00
restart_method = command

it works sometimes, but something i have this connection errors.

> The firmware should be compiled for the
STM32F103 with a "28KiB bootloader". Also, select "Enable extra
low-level configuration options" and configure "GPIO pins to set at
micro-controller startup" to "!PC13".

The "make flash" command does not work on the SKR mini E3. Instead,
after running "make", copy the generated "out/klipper.bin" file to a
file named "firmware.bin" on an SD card and then restart the SKR
mini E3 with that SD card.
>  

that's exactly how I installed it

crocus80, 2020-04-06:
I'm having the exact same issue with the SKR Mini E3 V1.2.
Applied the same method several times, with different sdcards and usb cables...

KevinOConnor, 2020-04-08:
I recommend you upgrade to the latest version of Klipper (see the FAQ for upgrade instructions), then add `restart_method: cheetah` to the mcu section of the config, then do a full power restart of both printer and rpi.  If the problem persists after that, please attach the Klipper log file from that attempt.

-Kevin

KevinOConnor, 2020-04-08:
Sorry, I got the printers mixed up.  My previous advice would only be for a cheetah board, not the skr boards.

-Kevin

KevinOConnor, 2020-04-24:
I've seen reports of USB reconnect failures on the mini-e3 and similar boards.  It is not known the root cause of the issue - my best guess is problems on the bootloader that comes with the board.  The only known work around is to reset again (possibly via manually resetting power to the board) until it comes back up.

-Kevin

closed 2020-05-11 by KevinOConnor