Klipper3d/klipper #2676: Move exceeds maximum extrusion and no TMC2208 Creality Silent Board?

Osmodia666, 2020-04-01:
Hi,
trying to print I get an Error: Move exceeds maximum extrusion (0.943mm^2 vs 0.640mm^2)
I successfully installed klipper on my Ender 3 Pro with a genuine Creality Silent Board with TMC2208.
But unfortunately, 
I didn't find any info on how to config the stock TMC2208 board in klipper.

Maybe that is causing this error?

The Error is not coming from a issue with my slicer, because I tried printing something which I printed a dozen times before and this was the same file I printed with the old TH3D FW, which worked great.

GarthSnyder, 2020-04-01:
Most likely, this is just as the error message says: you are sending some G-code that would result in an extrusion cross-section larger than the active limit. Do you have a purge line in your preamble that starts with a blob? That's often the culprit.

See max_extrude_cross_section in the [extruder] section of example.cfg for details. You are not too far out of spec - you could probably just increase the limit.

Osmodia666, 2020-04-02:
I used the *.cfg for Ender 3 which is found here, and didn't change anything.

BlackStump, 2020-04-02:
I would check your slicer gcode
>The Error is not coming from a issue with my slicer, because I tried printing something which I >printed a dozen times before and this was the same file I printed with the old TH3D FW, which >worked great.

just to clarify you have printed the file a dozen times before with klipper?

Osmodia666, 2020-04-02:
Yes,
 I printed the file a dozen times without having any issues.
The printer has Octoprint installed for over a year now, the only thing I
changed was the firmware of the board.
The file has not been changed or resliced, it was on the SD card all the time.
It has to be something with the FW, because it worked for over a year with
TH3D FW.

Gilabite, 2020-04-02:
I had that error on my ender 3 when I was tuning pressure advance, I miss typed the value from 0.8 to 8.0, didnt notice and got that error.

BlackStump, 2020-04-02:
Please read the contact info and follow the instructions paying attention to using M112 immediately after the said error then post that log if you did that in the first instant as requested you would have saved everyone's time including your own!
https://www.klipper3d.org/Contact.html

ld3300, 2020-04-13:
@Osmodia666 I've had this problem with some specific purge code I was using with my Ender 3. I've also had problems with the way my slicer generates code for the purge block. just add `max_extrude_cross_section: 2` to your extruder config and see if the problem goes away.  The default value is 0.64 (4.0 * nozzle_diameter^2).  If you want to be more conservative you can increase it by smaller degrees until it works.  This isn't to take away from the points that @Hywelmartin  and @BlackStump bring up.  Please reslice your code as well with attention to your klipper settings, and if issues persist use the M112 steps so we can get enough information to help you.

KevinOConnor, 2020-04-24:
I'm going to close this as it appears the original question was answered (update the start g-code to not extrude so much or update the config to allow a larger extrusion area).

-Kevin

closed 2020-04-24 by KevinOConnor