Klipper3d/klipper #2677: Anycubic Kossel Plus + TMC2208 - Enhanced delta calibration - DELTA_ANALYZE SCALE

lk29, 2020-04-01:
try callibrate my Kossel Plus + TMC2208

trying to calibrate
print the docs/prints/calibrate_size.stl file

sizes of the obtained target 137.9

Do I understand correctly that I need 137.9 / 140 = 0.985

DELTA_ANALYZE SCALE=0.985
DELTA_ANALYZE CALIBRATE=extended
SAVE_CONFIG

restart klipper 

AND ... 

attempt to print again

and again the size is 137.9
What am I doing wrong.

BlackStump, 2020-04-02:
I think you need to read the Enhanced Delta Calibration document and follow it step by step, do not skip any steps and it will then make sense to you.
https://www.klipper3d.org/Delta_Calibrate.html

Hywelmartin, 2020-04-02:
The DELTA_ANALYZE SCALE is if you scaled the print in the slicer.. so in your case 1.0

closed 2020-04-08 by klipper-gitissuebot