Klipper3d/klipper #2681: M150 Gcode Macro

TheRunnerPi, 2020-04-03:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4425723/klippy.log)

Hi all,

i would like to have an Light ON / Light OFF Marco in my OctoPrint.

I have RGB LEDs and i can controll them with M150 R0 U0 B0 via OctoPrint.
Connected are the LED to the PINs of the Raspberry 
```
PIN=11 GPIO=17
PIN=13 GPIO=27
PIN=15 GPIO=22
```

[image]

But if i type that in the printer.cfg Klipper does not understand what i want.

Then i tried:

```
[output_pin LED_Rot]
pin: 11
[gcode_macro LICHT_AN]
	SET_PIN PIN=LED_Rot VALUE=1
```

But it Pin 11 isn´t the right one. 

Maybe its GPIO 17
```
[output_pin LED_Rot]
pin: 17
[gcode_macro LICHT_AN]
	SET_PIN PIN=LED_Rot VALUE=1
```

Unknown value '17' in enumeration 'pin'

I am done with my guesses, maybe you have a Answer for me :)

Regrads,
Daniel

FHeilmann, 2020-04-03:
the output pins relate to the pins on your 3D printing mainboard, not the raspberry pi.

Hywelmartin, 2020-04-03:
https://github.com/KevinOConnor/klipper/issues/1146

TheRunnerPi, 2020-04-03:
![Klipper](https://user-images.githubusercontent.com/15970898/78384701-f6134600-75da-
11ea-9f5d-eb51cce12814.JPG)

Okay, i found a solution that works for me. 
Typed the M150 gcode into OctoPrint/Klipper Macro Options.

Thanks and close.

closed 2020-04-03 by TheRunnerPi