Klipper3d/klipper #2686: Cannot compile on Raspbian Stretch for LPC1768

reklawdiov, 2020-04-04:
I keep getting the error indicated in the screenshot, the symlink for board appears to be invalid. I'm running the latest OctoPrint, manually built on Raspbian Stretch (not OctoPi image)
My controller board is a BTT SKR 1.3

[klippy.log](https://github.com/KevinOConnor/klipper/files/4432106/klippy.log)
[image]
[image]

FanDjango, 2020-04-05:
I just downloaded the current zip right now. Same config in `make menuconfig`
Works.

```
root@octo-a6:~/klipper-20200405# make menuconfig
scripts/kconfig/conf --olddefconfig /root/klipper-20200405/src/Kconfig
#
# configuration written to /root/klipper-20200405/.config
#
scripts/kconfig/mconf /root/klipper-20200405/src/Kconfig
configuration written to /root/klipper-20200405/.config

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

root@octo-a6:~/klipper-20200405# make
  Build Kconfig config file
#
# configuration written to /root/klipper-20200405/.config
#
  Creating symbolic link out/board
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/initial_pins.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/thermocouple.o
  Compiling out/src/i2ccmds.o
  Compiling out/src/lcd_st7920.o
etc etc etc
```

Seems to me make cannot create the symbolic link /out/board for some reason.

Try to create the symbolic link youself and see if stretch gives you an error indication, maybe.

Here is what mine has created:

```
root@octo-a6:~/klipper-20200405# cd out
root@octo-a6:~/klipper-20200405/out# ls -l
total 272
-rw-r--r-- 1 root root   4281 Apr  5 12:17 autoconf.h
lrwxrwxrwx 1 root root     34 Apr  5 12:17 board -> /root/klipper-20200405/src/lpc176x
drwxr-xr-x 2 root root   4096 Apr  5 12:17 board-generic
-rw-r--r-- 1 root root      0 Apr  5 12:17 board-link
-rw-r--r-- 1 root root  44803 Apr  5 12:17 compile_time_request.c
-rw-r--r-- 1 root root    583 Apr  5 12:17 compile_time_request.d
-rw-r--r-- 1 root root 110228 Apr  5 12:17 compile_time_request.o
-rw-r--r-- 1 root root  10643 Apr  5 12:17 compile_time_request.txt
drwxr-xr-x 4 root root   4096 Apr  5 12:17 include
-rwxr-xr-x 1 root root  19432 Apr  5 12:17 klipper.bin
-rw-r--r-- 1 root root   5401 Apr  5 12:17 klipper.dict
-rwxr-xr-x 1 root root  89460 Apr  5 12:17 klipper.elf
drwxr-xr-x 3 root root   4096 Apr  5 12:17 lib
drwxr-xr-x 3 root root   4096 Apr  5 12:16 scripts
drwxr-xr-x 4 root root   4096 Apr  5 12:17 src
```

Mabye that helps.

reklawdiov, 2020-04-05:
This is the link that's created by the makefile, since it performs an rm on any existing directory, I had to make my own and comment out that section of the makefile.

[image]

Here's my link
[image]

It also creates a link under klipper/out/board-generic/board that is invalid. Here's mine.
[image]

Original Makefile ~line 75
```makefile
$(OUT)board-link: $(KCONFIG_CONFIG)
	@echo "  Creating symbolic link $(OUT)board"
	$(Q)mkdir -p $(addprefix $(OUT), $(dirs-y))
	$(Q)touch $@
	$(Q)rm -f $(OUT)board
	$(Q)ln -sf $(PWD)/src/$(CONFIG_BOARD_DIRECTORY) $(OUT)board
	$(Q)mkdir -p $(OUT)board-generic
	$(Q)rm -f $(OUT)board-generic/board
	$(Q)ln -sf $(PWD)/src/generic $(OUT)board-generic/board
```

Changed to this and then created the symlinks manually.
```makefile
$(OUT)board-link: $(KCONFIG_CONFIG)
	@echo "  Creating symbolic link $(OUT)board"
	#$(Q)mkdir -p $(addprefix $(OUT), $(dirs-y))
	$(Q)touch $@
	#$(Q)rm -f $(OUT)board
	#$(Q)ln -sf $(PWD)/src/$(CONFIG_BOARD_DIRECTORY) $(OUT)board
	#$(Q)mkdir -p $(OUT)board-generic
	#$(Q)rm -f $(OUT)board-generic/board
	#$(Q)ln -sf $(PWD)/src/generic $(OUT)board-generic/board
```

compilation then succeeded without error but I don't consider this a fix.

FanDjango, 2020-04-06:
It looks as if $(PWD) is returning "" (nothing). 

Try putting a 
```
$(Q)echo $(PWD) $(PWD)/src/$(CONFIG_BOARD_DIRECTORY) $(OUT)board
```
line where the `ln...` is, to prove the `$(PWD)` returns nothing.

Then check `echo $PWD` on the command line to see what you get.

reklawdiov, 2020-04-08:
That's exactly what's happening. 

The value of the PWD var is empty, while echoing $PWD after make is /home/pi/klipper as expected.
[image]

FanDjango, 2020-04-08:
I found somewhere on the google-verse:

[here](https://jameswestby.net/mentors/commonpackagingmistakes/usingpwdnotcurdir.html)

> In debian/rules, when you want to refer to a location relative to the build directory, for instance to move or delete a file you need to use a variable that contains the current directory. One such variable is $(PWD), which is the $PWD environment variable from the shell.

>When a package is built it needs a root environment for some tasks. For this fakeroot is normally used. Some architectures don't have this, and so use sudo instead, giving them real root. in 2006 sudo was found to have a security vulnerability due to environment variables being passes to the process that is spawned as root. The fix was to change sudo to only pass some variables to the spawned process by default, and this list does not include $PWD. This means that if you use it in your debian/rules your package may fail to build on a Buildd that uses sudo.

>To work around this problem use $(CURDIR) rather than $(PWD). It will work exactly the same, but avoid these problems

If using `$(CURDIR)`  in all places instead of `$(PWD)` fixes this for you, it might be worth it to submit a PR for this. I am just surprised this is not seen by many others doing the compile/make under sudo, I have not researched whether this might be a sudo configurable thing by now.

reklawdiov, 2020-04-08:
That did the trick, replacing $(PWD) with $(CURDIR) worked with sudo make. It must have been one of the updates at some point, because I had compiled successfully on this system several months ago without any changes to the makefile. Thank you for your help!

closed 2020-04-08 by reklawdiov