Klipper3d/klipper #2687: Klipper does not respect pin_move_time when lowering the probe

fredizzimo, 2020-04-04:
It looks like klipper does not repsect the `pin_move_time` when lowering the probe. It can therefore move down even if there's an error.

The following sequence shows what happens with a `pin_move_time` of 0.675. 
```
Send: G28
Recv: ok
[...]
Send: probe
Recv: // Failed to verify BLTouch probe is raised; retrying.
Recv: // probe at 97.000,2.000 is z=1.262500
Recv: // Result is z=1.262500
Recv: ok
[...]
Send: probe
Recv: // Failed to verify BLTouch probe is raised; retrying.
Recv: // probe at 97.000,2.000 is z=1.195000
Recv: // Result is z=1.195000
Recv: ok
[...]
Send: probe
Recv: // Failed to verify BLTouch probe is raised; retrying.
Recv: // probe at 97.000,2.000 is z=1.122500
Recv: // Result is z=1.122500
Recv: ok
[...]
Send: probe
Recv: // Failed to verify BLTouch probe is raised; retrying.
Recv: // probe at 97.000,2.000 is z=1.050000
Recv: // Result is z=1.050000
Recv: ok
[...]
```
If the time is increased to 0.8 it correctly detects the error.

I also took a capture with a logic analyzer to see exactly what happened 
[bltouch_timing.zip](https://github.com/KevinOConnor/klipper/files/4432506/bltouch_timing.zip)

It can be opened with this program https://ideas.saleae.com/f/changelog/

I also added an image, so you don't necessarily need to open the capture

[image]

The capture shows that
* the pin down command was issued for around 0.56 seconds
* The steppers started to move after 0.59 seconds
* The probe reported an error after 0.66 seconds (which is lower than the `pin_move_time`, so it should have been able to detect the problem)
* The first attempt to move the probe up was after 0.8 seconds

So it should clearly starting to move the steppers too early, therefore failing to detect the problem.

All the tests were done using and Ender 3 printer, and a genuine BLTouch smart V2 probe.

A note about the firmware version I'm using. Like the klippy.log shows, I'm indeed using a custom firmware version, because I like to store my configuration along the firmware. I also have some changes to the menus, but otherwise it's completely stock with commit 0342c50033b998b191fb2aed733ec260eb6a8d81 merged. 
[klippy.zip](https://github.com/KevinOConnor/klipper/files/4432426/klippy.zip)

KevinOConnor, 2020-04-04:
On closer inspection, the recent commit 5c8d15bb caused a regression in lower_probe() timing.  So, I have reverted that commit (as of a028aeaf).

Hopefully that fixes the issue you are seeing.  Otherwise, you'll need to reproduce the error on the pristine code.

Thanks.
-Kevin

FanDjango, 2020-04-05:
I am still planning to test this coming week @fredizzimo with the newest code

FanDjango, 2020-04-05:
@KevinOConnor We should consider making sure that we wait a total of ~750ms before moving the stepper to see if an error occurs when deploying. It is more conservative than the 675 we currently use.

KevinOConnor, 2020-04-08:
FWIW, I'm fine with changing the default pin_move_time.

-Kevin

FanDjango, 2020-04-08:
K. Its on my todo list

KevinOConnor, 2020-04-15:
I'm going to close this as the original issue should now be fixed.

-Kevin

closed 2020-04-15 by KevinOConnor