Klipper3d/klipper #2689: mks sbase - pin p2.6 stuck in on / high

joshi-redbridge, 2020-04-05:
Im trying to have pin 2.6 (P2.6) as a second fan control. Thou it seems to be stuck in ON all the time.
When looking at the controller I can see the green LED light up at start and stays on no matter the command I send. P2.4 works as expected, but that leaves me with a coldend fan that is always on, which I don't want.
[klippy.log](https://github.com/KevinOConnor/klipper/files/4434491/klippy.log)

mental405, 2020-04-06:
You don't seem to have P2.6 defined in your config file. It is likely that pin is in a default state of HIGH or it is floating and causing the MOSFET to turn on. You need to define it in your config file in order to make it work.

Something like
```
[heater_fan hotend_fan]
pin:                        P2.6
heater:                     extruder
heater_temp:                60.0
fan_speed:                  1.0
max_power:                  1.0
kick_start_time:            0.100
shutdown_speed:             1.0
```

or

```
[fan]
pin:                        P2.6
kick_start_time:            0.100
```

joshi-redbridge, 2020-04-06:
Hmm - for some reason it's not the correct config that is shown in klippy.log. 
Here is the correct config that is the one that I use and is somewhat working.

```[stepper_x]
step_pin: P2.0
dir_pin: !P0.5
#enable_pin: P0.4
enable_pin: !P0.4
#step_distance: .0125
step_distance: .00625
endstop_pin: ~!P1.24
#endstop_pin: ^P1.25
position_endstop: 0
position_max: 220
homing_speed: 100

[stepper_y]
step_pin: P2.1
dir_pin: !P0.11
#enable_pin: P0.10
enable_pin: !P0.10
#step_distance: .0125
step_distance: .00625
#endstop_pin: ^P1.26
endstop_pin: ^!P1.27
position_endstop: 0
position_max: 200
homing_speed: 100

[stepper_z]
step_pin: P2.2
dir_pin: !P0.20
enable_pin: !P0.19
#Âstep_distance: .0025
step_distance: .00125
endstop_pin: ^!P1.28
#endstop_pin: ^P1.29
position_endstop: 0.5
position_max: 301
homing_speed: 150

[extruder]
step_pin: P2.8
dir_pin: !P2.13
enable_pin: !P4.29
step_distance: 0.0012045
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: P2.7
sensor_type: ATC Semitec 104GT-2
sensor_pin: P0.23
#control: pid
min_temp: 0
max_temp: 280
min_extrude_temp: 175
max_extrude_only_distance: 150.0
max_extrude_cross_section: 1000.0

[extruder1]
step_pin: P2.3
dir_pin: !P0.22
enable_pin: !P0.21
step_distance: 0.0012045
nozzle_diameter: 0.400
filament_diameter: 1.750
shared_heater: extruder
min_extrude_temp: 175
max_extrude_only_distance: 150.0
max_extrude_cross_section: 1000.0

[gcode_macro T0]
gcode:
  ACTIVATE_EXTRUDER EXTRUDER=extruder

[gcode_macro T1]
gcode:
  ACTIVATE_EXTRUDER EXTRUDER=extruder1

# [gcode_macro fans_off]
# gcode:
#   SET_PIN PIN=coldend_fan VALUE=0
#   SET_PIN PIN=fan VALUE=0

[heater_bed]
heater_pin: P2.5
sensor_type: EPCOS 100K B57560G104F
sensor_pin: P0.26
#control: pid
#pid_Kp: 22.2
#pid_Ki: 1.08
#pid_Kd: 114
max_delta: 4.0
min_temp: 0
max_temp: 130
max_power: 0.6
cycle_time: 0.1

[fan]
pin: P2.6
#pwm: True
kick_start_time: 0.1

[heater_fan coldend_fan]
pin: P2.4
heater: extruder
heater_temp: 45

[mcu]
serial: /dev/serial/by-id/usb-Klipper_lpc1768_0D60FF0E2025F6AE3FD89457C02000F5-if00

[printer]
name: Hypercube
kinematics: corexy
max_velocity: 200
max_accel: 3000
max_z_velocity: 5
max_z_accel: 100

[static_digital_output leds]
pins: P1.18, P1.19, P1.20, P1.21, P4.28

[mcp4451 stepper_digipot1]
i2c_address: 44
# Scale the config so that wiper values can be specified in amps.
scale: 2.25
## wiper 0 is X (aka alpha), 1 is Y, 2 is Z, 3 is E0
wiper_0: 1.0
wiper_1: 1.0
wiper_2: 1.0
wiper_3: 1.0
#wiper_4: 0.5
#
[mcp4451 stepper_digipot2]
i2c_address: 45
scale: 2.25
# wiper 0 is E1
wiper_0: 1.0

# "RepRapDiscount 128x64 Full Graphic Smart Controller" type displays
#[display]
#lcd_type: st7920
#cs_pin: P0.16
#sclk_pin: P0.15
#sid_pin: P0.18
#encoder_pins: ^P3.25, ^P3.26
#click_pin: ^!P1.30

[virtual_sdcard]
path: /home/pi/sdcard

[web_dwc2]
# optional - defaulting to Klipper
printer_name: Hypercube
# optional - defaulting to 127.0.0.1
listen_adress: 0.0.0.0
# needed - use above 1024 as nonroot
listen_port: 4750
#	optional defaulting to dwc2/web. Its a folder relative to your virtual sdcard.
web_path: dwc2/web

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [heater_bed]
#*# control = pid
#*# pid_kp = 19.044
#*# pid_ki = 0.279
#*# pid_kd = 324.942
#*#
#*# [extruder]
#*# control = pid
#*# pid_kp = 23.054
#*# pid_ki = 0.854
#*# pid_kd = 155.618```

jakep82, 2020-04-06:
If the config in the log doesn't match the config you're editing, then you're editing the wrong file.  Klipper is using the config shown in the log.  Also note that DWC2 is not supported.  You need to run unmodified code to get support here.

joshi-redbridge, 2020-04-06:
> If the config in the log doesn't match the config you're editing, then you're editing the wrong file. Klipper is using the config shown in the log. Also note that DWC2 is not supported. You need to run unmodified code to get support here.

DWC2 is not in use, it's something that just stuck along while I tried the same config on a different server ( A Tinkerboard).

And no, I have one config file on the server that runs klipper. I've Never used arduino config but still there is entries with arduino-config in the log. Why? Don't know.

The fact remains that P2.6 is NOT controlable from Klipper.

jakep82, 2020-04-06:
> Git version: 'v0.8.0-388-g302cd382-dirty'

In order for us to help, you need to post a log with pristine code from this repository, and a configuration showing P2.6 configured and non-functional.  The log you posted does not show P2.6 configured, and the code has been modified.

joshi-redbridge, 2020-04-07:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4444519/klippy.log)
`git pull`, removed the dwc-config lines from the configuraiton and verified that all started.
P2.6 is still in always high =/

ld3300, 2020-04-13:
hi @joshi-redbridge 
from the log it looks like your klipper version is still dirty and needs to be force updated.  SSH into your pi and try this:
`git fetch --all`
`git reset --hard origin/master`
then test it out and if there are still issues upload a new klippy.log

joshi-redbridge, 2020-04-14:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4473758/klippy.log)
Still no cigar.
This is what I did to reset.
```
  320 cd klipper
  321  git fetch --all
  322  git diff origin/master
  323  git reset --hard origin/master
  324  sudo reboot
  325  sudo systemctl stop klipper
  326  sudo echo '' > /var/log/klippy.log 
  327  cat /var/log/klippy.log
  328  sudo systemctl restart klipper
  329  cat /var/log/klippy.log

KevinOConnor, 2020-04-14:
As far as I know, the Klipper gpio code works as intended.  I've not seen any other reports of issues with pin "P2.6".  It is possible an electrical problem on the board (eg, a shorted wire or blown mosfet) is causing the problem on that particular board.

-Kevin

ld3300, 2020-04-15:
I've also seen a bad IC pin act that way.  If it got zapped it could be permanently on.

closed 2020-05-11 by KevinOConnor