Klipper3d/klipper #2690: BLTouch Genuine 3.1 / SKR 1.3

lolejar, 2020-04-06:
BOARD : SKR 1.3
BLTOUCH: Genuine v3.1
Problem : Probing occasionally fails with message 

```
Failed to verify BLTouch probe is raised; retrying.
Failed to verify BLTouch probe is raised; retrying.
BLTouch failed to raise probe
BLTouch failed to raise probe
```

I have completly new wiring. On Marlin probing in any number of consecutive probes doesn't fail. In Klipper it fails constantly at random probe.

Marlin video : https://photos.app.goo.gl/iesy21NgvG8eG97W6
Klipper video : https://photos.app.goo.gl/PS7oqoz4pV9svV8U9
Klipper config : https://pastebin.com/Reh8b5aA
Klippy log : https://pastebin.com/nFGE0phs

Could you please help me with this ? Thank you in advance.

FanDjango, 2020-04-06:
Try some things, please:

Just remove those that I can see in your config or set them to the default:

Also, get that latest build, much has been added/changed in the last week even. If you paste a klippy log, make sure it is complete, I cannot see the startup and thus have no indication of versions and dates etc.

```
pin_up_reports_not_triggered: True
pin_up_touch_mode_reports_triggered: False
pin_move_time: 0.3
```

Then report again, please.

lolejar, 2020-04-06:
Commenting out pin_move_time causes on probe errors that printer is not immediately halter but it somehow recovers after that state either it throws out error

```
Send: G29
Recv: // Failed to verify BLTouch probe is raised; retrying.
Recv: // Failed to verify BLTouch probe is raised; retrying.
Recv: // probe at 52.000,36.000 is z=2.830000
Recv: // probe at 135.500,36.000 is z=2.832500
Recv: // Failed to verify BLTouch probe is raised; retrying.
Recv: // Failed to verify BLTouch probe is raised; retrying.
Recv: // probe at 219.000,36.000 is z=2.825000
Recv: // probe at 219.000,124.000 is z=2.745000
Recv: // probe at 135.500,124.000 is z=2.745000
Recv: // probe at 52.000,124.000 is z=2.705000
Recv: // probe at 52.000,212.000 is z=2.810000
Recv: // probe at 135.500,212.000 is z=2.795000
Recv: // probe at 219.000,212.000 is z=2.685000
Recv: // Mesh Bed Leveling Complete
```

but either with this behaviour it can get stuck completely (see below - it's continuation of output from above)

```
Recv: // Bed Mesh state has been saved to profile [default]
Recv: // for the current session.  The SAVE_CONFIG command will
Recv: // update the printer config file and restart the printer.
Recv: ok
Send: M105
Recv: ok B:24.4 /0.0 T0:24.4 /0.0
Send: M105
Recv: ok B:24.4 /0.0 T0:24.5 /0.0
Send: BED_MESH_OUTPUT
Recv: // Mesh Leveling Probed Z positions:
Recv: // 0.080000 0.082500 0.075000
Recv: // -0.045000 -0.005000 -0.005000
Recv: // 0.060000 0.045000 -0.065000
Recv: Mesh X,Y: 7,7
Recv: Search Height: 10
Recv: Mesh Average: 0.02
Recv: Mesh Range: min=-0.0650 max=0.0828
Recv: Interpolation Algorithm: lagrange
Recv: Measured points:
Recv:   0.060000  0.065556  0.060556  0.045000  0.018889  -0.017778  -0.065000
Recv:   -0.000556  0.012191  0.016728  0.013056  0.001173  -0.018920  -0.047222
Recv:   -0.035556  -0.018735  -0.008086  -0.003611  -0.005309  -0.013179  -0.027222
Recv:   -0.045000  -0.027222  -0.013889  -0.005000  -0.000556  -0.000556  -0.005000
Recv:   -0.028889  -0.013272  -0.000679  0.008889  0.015432  0.018951  0.019444
Recv:   0.012778  0.023117  0.031543  0.038056  0.042654  0.045340  0.046111
Recv:   0.080000  0.081944  0.082778  0.082500  0.081111  0.078611  0.075000
Recv: 
Recv: ok
Send: M105
Recv: ok B:24.4 /0.0 T0:24.4 /0.0
Send: BED_MESH_OUTPUT
Recv: // Mesh Leveling Probed Z positions:
Recv: // 0.080000 0.082500 0.075000
Recv: // -0.045000 -0.005000 -0.005000
Recv: // 0.060000 0.045000 -0.065000
Recv: Mesh X,Y: 7,7
Recv: Search Height: 10
Recv: Mesh Average: 0.02
Recv: Mesh Range: min=-0.0650 max=0.0828
Recv: Interpolation Algorithm: lagrange
Recv: Measured points:
Recv:   0.060000  0.065556  0.060556  0.045000  0.018889  -0.017778  -0.065000
Recv:   -0.000556  0.012191  0.016728  0.013056  0.001173  -0.018920  -0.047222
Recv:   -0.035556  -0.018735  -0.008086  -0.003611  -0.005309  -0.013179  -0.027222
Recv:   -0.045000  -0.027222  -0.013889  -0.005000  -0.000556  -0.000556  -0.005000
Recv:   -0.028889  -0.013272  -0.000679  0.008889  0.015432  0.018951  0.019444
Recv:   0.012778  0.023117  0.031543  0.038056  0.042654  0.045340  0.046111
Recv:   0.080000  0.081944  0.082778  0.082500  0.081111  0.078611  0.075000
Recv: 
Recv: ok
Send: M105
Recv: ok B:24.4 /0.0 T0:24.5 /0.0
Send: G29
Recv: // Failed to verify BLTouch probe is raised; retrying.
Recv: // Failed to verify BLTouch probe is raised; retrying.
Recv: !! BLTouch failed to raise probe
Changing monitoring state from "Operational" to "Error: BLTouch failed to raise probe"
Send: M112
Send: N3 M112*34
Send: N4 M104 T0 S0*37
Send: N5 M140 S0*96
Changing monitoring state from "Error: BLTouch failed to raise probe" to "Offline (Error: BLTouch failed to raise probe)"
Connection closed, closing down monitor
```

FanDjango, 2020-04-06:
From when is this klippy version?

lolejar, 2020-04-06:
You mean klipper ? master branch - https://github.com/KevinOConnor/klipper/commit/d6034013e7f1dbe8676460656c0a2b7978e9618c

FanDjango, 2020-04-06:
You're missing a lot of BLTouch related stuff that will help diagnose, it not solve, your problem:
[image]

FanDjango, 2020-04-06:
So, if you can, please do me the favor...

lolejar, 2020-04-06:
Updated to latest version. The probing is still stucking like

> Send: G29
> Recv: // Failed to verify BLTouch probe is raised; retrying.
> Recv: // probe at 52.000,36.000 is z=2.822500
> Recv: // Failed to verify BLTouch probe is raised; retrying.
> Recv: // probe at 135.500,36.000 is z=2.827500

but then it recovers and continues. It seems like permanent stuck which puts printer into error mode is gone. I did 10x mesh generation. Lots of stuck (unresponsive/slow responsive) probes but then it recovers (either probe value is retrieved by octoprint) and it continues. Is this normal / desired behaviour ? I've never seen something like this with marlin where pin retracts always on first contact with bed.

FanDjango, 2020-04-06:
Good, now need to find out why it fails to verify the raised. Can you put a small video in?

FanDjango, 2020-04-06:
And the klippy log too

lolejar, 2020-04-06:
here is the video. 
it homes and gets stuck on first point. this is what it looks like when it fails to verify raised. 
https://photos.app.goo.gl/Gf7W4WPppRboP5ve9

FanDjango, 2020-04-06:
I see. And then it recovers and goes on the next point, where the same happens, right.

lolejar, 2020-04-06:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4438564/klippy.log)

lolejar, 2020-04-06:
it doesn't happen on all points, only on some and in random

FanDjango, 2020-04-06:
Homing is no problem, it seems.

Probe goes into error very quickly. klipper raise takes place to late. Strange. Can you test a first thing:

```
stow_on_each_sample: false
```
might help localize the problem better, I'd like a video again, this time two probe places.

Also please test a 

`PROBE_ACCURACY` after a `G28`

lolejar, 2020-04-06:
PROBE_ACCURACY (2 or 3 times) https://photos.app.goo.gl/HQReyNjjTKMCuhNS6
G29 (2x times) https://photos.app.goo.gl/uSz7bmkifkcWNQJGA

[klippy.log](https://github.com/KevinOConnor/klipper/files/4438636/klippy.log)

FanDjango, 2020-04-06:
Sorry, for some analysis need one more info, no video needed:

Does `PROBE_ACCURACY` fail just like the `G29` if you set `stow_on_each_sample: true` ?

It looks like there is some problem with probing (not homing) when trigger comes in and retract is too late or goes wrong somehow.

Need to look at the code this evening.

Until then it looks like you can do well by using `stow_on_each_sample: false`, right?

lolejar, 2020-04-06:
did 10x PROBE_ACCURACY (100 probes without single failure)
did G29 NEXT and it failed (just a slight stuck/lag like before) on HOMING and then on 1 of 9 points in mesh

FanDjango, 2020-04-07:
So, to sum this up before further action/analysis:

On your set-up, BOARD : SKR 1.3, BLTOUCH: Genuine v3.1, current klipper version, all `[bltouch]` parameters default, normal speeds:

1. With `stow_on_each_sample: true` (the default), `Z-HOMING` had a seldom failure, `PROBE_ACCURACY` works ok many times, `G29` has many failures as in the video

2. With `stow_on_each_sample: false` things are much better or perfect

lolejar, 2020-04-07:
Yes that is correct with the latest version as of a028aea

FanDjango, 2020-04-07:
So I have duplicated your set-up here and am ready for some testing.

Original video above is no longer available for view, so I need a short video of G29 (3 probes)

for case 1 
>With stow_on_each_sample: true (the default), Z-HOMING had a seldom failure, PROBE_ACCURACY works ok many times, G29 has many failures as in the video

lolejar, 2020-04-07:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4443291/klippy.log)
https://photos.app.goo.gl/uKN728fpEtKLGhbd9

FanDjango, 2020-04-07:
So a single failure to raise the probe quickly enough on the third position of your 3x3 mesh G29, which was recovered and it also should not affect the position accuracy as the endstop is triggered on the downward move by the mcu.

Is there any chance the underlying hardware which is running python/klippy has got some heavy loads? What is it running on? A normal RPi or what?

lolejar, 2020-04-07:
RPi 2B+

FanDjango, 2020-04-07:
Need some time. Don't delete the video.

FanDjango, 2020-04-07:
I've got a special branch with a few changes to `bltouch.py`

Do you maybe want to get it, and _just_ overcopy the `bltouch.py` module in `klippy/extras`? 

We will get some more log messages and there are some small changes to timing.

It is at [here](https://github.com/FanDjango/klipper/commits/BLTOUCH-TEST)

lolejar, 2020-04-07:
ok i've update bltouch.py, recompiled, uploaded new firmware to printer, updated config `withprobe_with_touch_mode = True`. Problem still persists.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4446157/klippy.log)

FanDjango, 2020-04-07:
Funny, I don't see the messages from
[image]
so it's not the right version of `bltouch.py`, and it is not surprising nothing has changed.
Note: Since that is only in klippy, you don't need recompile, upload etc. But you do need a service klipper restart.
Also, don't use the touch mode until we find the original problem.

lolejar, 2020-04-07:
sorry i accidentally downloaded the master version instead of yours. anyway the problem persists 2x G29 > 2 errors each in one mesh 

[klippy.log](https://github.com/KevinOConnor/klipper/files/4446277/klippy.log)

FanDjango, 2020-04-07:
Ok, some comments:
1. No retransmits or any other USB slowdowns to be seen
2. It is not the lastest `bltouch.py`, I pushed some more stuff. You need to get it fresh from GitHub. I'm adding to it as I work with you.
3. In this TEST branch, `pin_move_time` default is 0.740 - I can see you have specified 0.675 -> which is rounded up to 0.68
4. What are current parms? Be nice if the klippy log were complete. I always delete it and restart klipper service, reconnect and then do the G29/G28

FanDjango, 2020-04-07:
Please make a nice complete log file and then I think we will need @KevinOConnor on this, because:

1. Either the probe is faulty, but you say it works flawlessly on Marlin

2. There is a lower level timing problem and Kevin will know how to find it, will have suggestions where to place commands to get the timing information. It concerns the moment the trigger was issued by the probe (t_trig), the moment the trigger was seen by the MCU (t_endstop) and the moment that the probe was given a pin-up cmd (t_stow).

The probe needs (t_stow + 160ms) - (t_trig) < 740ms.

lolejar, 2020-04-08:
latest bltouch.py, 1x G29, 2 errors

[klippy.log](https://github.com/KevinOConnor/klipper/files/4448347/klippy.log)

FanDjango, 2020-04-08:
Thanks for the log

>  i started it using on marlin like one month ago. made it work generated tens of meshes (>50) without single lag or hesitation. i can provide video if needed with marlin that doing 10x mesh won't generate any type of error

I believe.

> also can provide marlin configs if desired

As a matter of interest, I would like to compare, yes

> I mean blue diode that is inside probe stays lit on marlin whole time and blinks when command is issued (retract, stow etc ...) on klipper it is off and blinks only when command is issued

That is true and it is normal. klipper turns **off** the PWM after sending a command. The blue LED is used by ANTCLABs to indicate the presence of the PWM and its brightness even actually indicates to a certain degree the "angle" of the servo signal.

Marlin keeps the PWM signal of the last command to be sent to the probe active on the pin, quasi repeating it over and over again.

This **is** a different way to operate but up to now there is no indication that one way is better than the other.

> Another difference I can tell is that red light in marlin is on during probing

That surprises me. When the probe is deployed, the red light should go off until the trigger happens. Please make a short video.

FanDjango, 2020-04-08:
First thing to change:

Activate Pull-Up. Maybe the trigger signal is seen, but it is seen to too late.

```
sensor_pin = ^P1.25
```

lolejar, 2020-04-08:
Ok. I've uploaded Marlin. Sorry about that red light that was just my impression. It's not true anyway. 

Marlin Probing video : https://photos.app.goo.gl/pQoj85ptcSe78BHm6

[Configuration_adv.txt](https://github.com/KevinOConnor/klipper/files/4449193/Configuration_adv.txt)
[Configuration.txt](https://github.com/KevinOConnor/klipper/files/4449194/Configuration.txt)

FanDjango, 2020-04-08:
> Sorry about that red light that was just my impression

No problem. Just wanted to be sure

Yes, endstops-pull up is on. So 

```
sensor_pin = ^P1.25
```

is the first step to take.

Second step to improve reliability is to turn on 5V mode just like in Marlin in the `[bltouch]` section:

```
set_output_mode: 5V
```

lolejar, 2020-04-08:
both flag ^ and 5V mode enable. problem persists on second mesh 1 error

[klippy.log](https://github.com/KevinOConnor/klipper/files/4449277/klippy.log)

FanDjango, 2020-04-08:
And now `G28` and then `PROBE_ACCURACY SAMPLES=100`. Any failures?

lolejar, 2020-04-08:
100 samples

`
Recv: // PROBE_ACCURACY at X:164.000 Y:112.000 Z:15.000 (samples=100 retract=2.000 speed=5.0 lift_speed=5.0)
Recv: // probe accuracy results: maximum 2.765000, minimum 2.740000, range 0.025000, average 2.756750, median 2.757500, standard deviation 0.003700`

1 error

FanDjango, 2020-04-08:
I would like to find out your timing leeway on your machine/set-up.

It means you would need to edit your `klippy/extras/bltouch.py` and make a few experiments. Do you feel that you can do that?

FanDjango, 2020-04-08:
In `bltouch.py`, find this place (should be in line 117+):
```
    def raise_probe(self):
        logging.info("BLTouch raise probe")
        self.sync_mcu_print_time()
        if not self.pin_up_not_triggered:
            # No way to verify raise attempt - just issue commands
            self.send_cmd('reset')
            self.send_cmd('pin_up', duration=self.pin_move_time)
            self.send_cmd(None)
            logging.info("BLTouch raise probe done")
            return
        for retry in range(3):
            check_start_time = self.send_cmd('pin_up',
                                             duration=self.pin_move_time)
            check_end_time = self.send_cmd(None)
            success = self.verify_state(check_start_time, check_end_time, False)
```

Add this line at line 128:
```
self.send_cmd(None, duration=0.5)
```

so it now looks like this:

```
    def raise_probe(self):
        logging.info("BLTouch raise probe")
        self.sync_mcu_print_time()
        if not self.pin_up_not_triggered:
            # No way to verify raise attempt - just issue commands
            self.send_cmd('reset')
            self.send_cmd('pin_up', duration=self.pin_move_time)
            self.send_cmd(None)
            logging.info("BLTouch raise probe done")
            return
        for retry in range(3):
            self.send_cmd(None, duration=0.5)
            check_start_time = self.send_cmd('pin_up',
                                             duration=self.pin_move_time)
            check_end_time = self.send_cmd(None)
            success = self.verify_state(check_start_time, check_end_time, False)
```

With this added 0.5 second "None" delay, the probe should fail **every** time.

Please try to find the value where failures occur very often by testing smaller values, like 0.1, 0.2, 0.3 etc. On my machine, failures begin at 0.36. I would like to know if on your machine, failures start at a much lower value, like maybe even 0.1 or 0.2.

Yes, I know even without this delay ( = 0.0 ) you already have **some** failures. So maybe you should start at 0.1, then 0.2 etc. and tell me from when you get a failure **every** probe.

FanDjango, 2020-04-08:
BTW I also have LPC1768 MCU, V3.1 probe but Debian on Tinkerboard S (is Rockchip 4 CPU)

lolejar, 2020-04-08:
I am software developer (11 years) / devops (2 years) guy so you can talk to me technical ;) just unfamiliar with klipper either I can write / read python scripts ;) 

< 0.3 occasional
= 0.35 many (but few pass)
= 0.36 even more  (but few pass)
=> 0.37 each fails

FanDjango, 2020-04-08:
> I am software developer (11 years) / devops (2 years) guy so you can talk to me technical ;)

Roger Wilco

My thoughts after all this:

Now it seem that we have exhausted all things I can see on the high-level .py side of things.

We could put one or two diagnostic messages into `probe.py` and get some more timing information but it would be redundant - the 0.37 (which gets rounded up to 0.38, see in your log) is similar to my machine and proves: Generally, the timing is the same as on other machines of this type, there is time to spare - but for some reason once in while, the pin-up command comes too late. A trigger **definitely** has been seen, otherwise the downward movement would not have been stopped.

There used to be some problems with "race conditions" and suchlike, but these have been fixed quite some time ago and there are no such indications in the klippy.log, but maybe Kevin can see more.

There used to be USB retransmits on a flaky connection pi->mcu (something Marlin would not be vulnerable to). This would cause retransmits (these cost time), but the log does not seem to indicate any of these.

There is a definite possibility of some task on the pi preempting the klippy software in adverse moment - statistically this must be happening often enough to perhaps detect it somehow on the linux side of things, otherwise the pure chance of it happening in the middle of a trigger-probe-retract sequence would be too low.

Leaving the parameters as they are right now and just add `stow_on_each_sample: True`: This will drastically reduce the amount of retracts that could cause retries. This would be an interim solution. Also you can use `probe_with_touch_mode: True`. This should also change the pin-up error probability (I would be curious to hear about that). And of course, you could even use **both** of those parameters.

I don't consider this issue closeable though. But it needs some time to devise a means to diagnose this.

FanDjango, 2020-04-08:
One very small thing I just found:

You have:
```
[mcu]
serial = /dev/serial/by-id/usb-Klipper_lpc1768_13700106018139AFF720075CC42000F5-if00
```

I successfully use:

```
[mcu]
serial = /dev/serial/by-id/usb-Klipper_lpc1768_13700106018139AFF720075CC42000F5-if00
baud: 250000
```

It is the default, I know. This hardware should be able to handle that. Try 115200. If it gets **better** then that would point to a flaky USB connection (big difference to Marlin). It's maybe something, you never know. But I don't think so.

FanDjango, 2020-04-08:
Another small thing comes to mind: klipper turns off the PWM between commands (and especially) after the pin deploy, while probing. You noticed that, see above.

Let's assume there is a bug in that code and the PWM output pin is "floating" when the PWM is turned off (big difference to Marlin), and I have seen your wiring. Maybe some signal is being picked up, confusing the BLTouch, causing it to go into ERROR state early?

If you think we should try that, we could add a very few lines of code to **NOT** turn off the PWM, just as an experiment.

Also, to reduce electrical noise for the probe, I turn off heaters when probing (see the `[homing_heaters]`config section for doing that).

```
[homing_heaters]
steppers: stepper_z, stepper_z1
heaters: heater_bed, extruder
```

lolejar, 2020-04-08:
Most of those cables are just temporary not part of permanent setup :) I just move ender 3 to be next to me on my work table so I can fiddle between work while on HO. Regarding heating it's turned off for me during probing. As for the baud rate didn't help :/ Thank you anyway appreciate your help a lot. I am thinking about buying another genuine probe to try out. Unfortunately none of my friends are having genuine one just cheap clones which work better than mine and that bothers me.

KevinOConnor, 2020-04-08:
FYI, I created a new work-bltouch-20200408 branch with additional debugging.  I recommend grabbing that branch, retrying, and then attaching the resulting Klipper log file here.

`cd ~/klipper ; git fetch ; git checkout origin/work-bltouch-20200408 ; make ; make flash ; sudo service klipper restart`

-Kevin

FanDjango, 2020-04-08:
I had a look at that debugging info in that branch that Kevin has put together and I strongly recommend you use that one @lolejar - it will give the timing information, which will either prove/disprove the probe is being stowed too late, we have narrowed it down to that question.

lolejar, 2020-04-09:
Hi huys attaching log. 

[klippy.log](https://github.com/KevinOConnor/klipper/files/4454228/klippy.log)

FanDjango, 2020-04-09:
A failing sequence edited out of that log:
```
BLTouch send pin_down at 212.451725
BLTouch send None at 213.031725
Stats 61477.0: gcodein=18 mcu: mcu_awake=0.004 mcu_task_avg=0.000011 mcu_task_stddev=0.000023 bytes_write=32730 bytes_read=29096 bytes_retransmit=9 bytes_invalid=0 send_seq=1881 receive_seq=1881 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=8 freq=99996767 heater_bed: target=0 temp=23.9 pwm=0.000 print_time=213.132 buffer_time=2.273 print_stall=0 extruder: target=0 temp=23.9 pwm=0.000 sysload=0.15 cputime=6.953 memavail=694752
bltouch probe @ 213.131725 (check=0.000500)
Stats 61478.0: gcodein=18 mcu: mcu_awake=0.004 mcu_task_avg=0.000011 mcu_task_stddev=0.000023 bytes_write=32771 bytes_read=29242 bytes_retransmit=9 bytes_invalid=0 send_seq=1884 receive_seq=1884 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=8 freq=99996768 heater_bed: target=0 temp=23.9 pwm=0.000 print_time=213.133 buffer_time=0.000 print_stall=0 extruder: target=0 temp=23.9 pwm=0.000 sysload=0.14 cputime=6.985 memavail=694752
Stats 61479.0: gcodein=18 mcu: mcu_awake=0.004 mcu_task_avg=0.000011 mcu_task_stddev=0.000023 bytes_write=32790 bytes_read=29368 bytes_retransmit=9 bytes_invalid=0 send_seq=1886 receive_seq=1886 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=99996769 heater_bed: target=0 temp=23.9 pwm=0.000 print_time=213.133 buffer_time=0.000 print_stall=0 extruder: target=0 temp=23.9 pwm=0.000 sysload=0.14 cputime=7.012 memavail=694752
Stats 61480.0: gcodein=18 mcu: mcu_awake=0.004 mcu_task_avg=0.000011 mcu_task_stddev=0.000023 bytes_write=33094 bytes_read=29569 bytes_retransmit=9 bytes_invalid=0 send_seq=1906 receive_seq=1906 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=99996770 heater_bed: target=0 temp=23.9 pwm=0.000 print_time=214.033 buffer_time=0.000 print_stall=0 extruder: target=0 temp=23.9 pwm=0.000 sysload=0.14 cputime=7.050 memavail=694752
Endstop trigger at 42.793533
mcu trigger position (-2846) -7.115000
BLTouch send pin_up at 214.698514
BLTouch send None at 215.378514
Stats 61481.0: gcodein=18 mcu: mcu_awake=0.004 mcu_task_avg=0.000011 mcu_task_stddev=0.000023 bytes_write=33344 bytes_read=29819 bytes_retransmit=9 bytes_invalid=0 send_seq=1927 receive_seq=1927 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=22 stalled_bytes=0 freq=99996770 heater_bed: target=0 temp=23.9 pwm=0.000 print_time=214.845 buffer_time=0.000 print_stall=0 extruder: target=0 temp=23.9 pwm=0.000 sysload=0.14 cputime=7.099 memavail=694752
Stats 61482.0: gcodein=18 mcu: mcu_awake=0.004 mcu_task_avg=0.000011 mcu_task_stddev=0.000023 bytes_write=33395 bytes_read=29942 bytes_retransmit=9 bytes_invalid=0 send_seq=1930 receive_seq=1930 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=99996773 heater_bed: target=0 temp=23.9 pwm=0.000 print_time=214.845 buffer_time=0.000 print_stall=0 extruder: target=0 temp=24.0 pwm=0.000 sysload=0.14 cputime=7.127 memavail=694752
Failed to verify BLTouch probe is raised; retrying.
BLTouch send reset at 216.615331
BLTouch send pin_up at 217.615331
BLTouch send None at 218.295331
```

The interesting part is (edited):

```
bltouch probe @ 213.131725 (check=0.000500)
Stats 61478.0: print_time=213.133
Stats 61479.0: print_time=213.133
Stats 61480.0: print_time=214.033
Endstop trigger at 42.793533
mcu trigger position (-2846) -7.115000
BLTouch send pin_up at 214.698514
BLTouch send None at 215.378514
```

@KevinOConnor I am stumped. What does `Endstop trigger at 42.793533` tell us?

I was looking to be able to discern _t(pin_up) - t(trigger)_

KevinOConnor, 2020-04-09:
> What does Endstop trigger at 42.793533 tell us?

Unfortunately, it means I messed up the debugging.  There should have been a call to `self._mcu.clock32_to_clock64()` in that debug output.  I did apply a fix to the testing branch for future runs.

Fortunately, we can reconstruct the actual time.  Although we don't have the high 32bits of the clock timestamp, we know it will always be a multiple of 2**32 and since we know the clock is 100Mhz, that means the actual timestamp is offset by some multiple of 42.94967296.

So, for that example, the actual timestamp is `4 * 42.94967296 + 42.793533 = 214.59222484`.  This result also checks out with the start of testing (`214.592225 - 213.131725 = 1.460500`, which is an even multiple of the interval .000500).

Which means the trigger was detected at 214.592225 and the pin_up command starts at 214.698514 - for a total delay of 106ms.

-Kevin

lolejar, 2020-04-10:
Latest branch checked out 
[klippy.log](https://github.com/KevinOConnor/klipper/files/4460838/klippy.log)

FanDjango, 2020-04-10:
Again 106ms, this time with correct timing log info:
```
Endstop trigger at 73.176174
mcu trigger position (-2874) -7.185000
BLTouch send pin_up at 73.283078
BLTouch send None at 73.963078
Stats 160592.8: gcodein=9 mcu: mcu_awake=0.004 mcu_task_avg=0.000011 mcu_task_stddev=0.000020 bytes_write=20855 bytes_read=18580 bytes_retransmit=9 bytes_invalid=0 send_seq=1190 receive_seq=1190 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=22 stalled_bytes=0 freq=99996805 heater_bed: target=0 temp=24.2 pwm=0.000 print_time=73.429 buffer_time=0.000 print_stall=0 extruder: target=0 temp=24.2 pwm=0.000 sysload=0.19 cputime=4.919 memavail=694348
Stats 160593.8: gcodein=9 mcu: mcu_awake=0.004 mcu_task_avg=0.000011 mcu_task_stddev=0.000020 bytes_write=20906 bytes_read=18711 bytes_retransmit=9 bytes_invalid=0 send_seq=1193 receive_seq=1193 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=99996808 heater_bed: target=0 temp=24.2 pwm=0.000 print_time=73.429 buffer_time=0.000 print_stall=0 extruder: target=0 temp=24.1 pwm=0.000 sysload=0.19 cputime=4.947 memavail=694348
Failed to verify BLTouch probe is raised; retrying.
```
So if you subtract, you get 106ms. Just as Kevin already deduced from the previous log which had the "bad" timing log entry.

--------------------
Now, this means the following:
1. The "pin-up" command **is not** being executed by the probe because it is **already** in an error state. **Proof**: Video shows the probe begins blinking directly after movement stops. Movement stops means: trigger is recognized.
2. Timing is now established: The "pin-up" command **is not** being sent too late (106ms after trigger) and thus this is not the reason for the probe going into an error state.

--------------------
What other things could theoretically happen to make the probe go into an error state at this very bad moment:

(All of these are just conjecture - and remember, the probe works ok in Marlin)

1. Since the PWM is turned off in klipper and not in Marlin, spurious noise on that pin/signal cable could be confusing the probe. Is the PWM pin set to Input and Pull-UP or to Output and High-Level when the PWM is turned off? Might be specific to this mcu type, of course (LPC1768). Could be tested against by not turning the PWM off after deploy or by adding a capacitor on the PWM/signal line to quieten it down, or by scrutinizing the PWM turn-off code for that mcu.

2. The timing information from the log might not reflect reality. One way to check this is perhaps to analyze the fail in the video frame by frame to get an estimate of how long it took from movement stop to blinking state begin. Really ~100ms?

3. I have never had the opportunity to test this, but perhaps the "pin-up" command is arriving **too early** and the probe is not yet ready to accept this so directly after its code was busy producing the output pulse and doing all sorts of internal stuff. Could easily be tested by making sure a "pin-up" command is never sent before at least 200ms have passed after a trigger event.

---------------

lolejar, 2020-04-10:
in case you want to do frame by frame analysis here is 60fps video https://photos.app.goo.gl/8uT8NVP3L4NFeGFDA

FanDjango, 2020-04-10:
@KevinOConnor 

This was fun, actually. Analyzed first a **fail probe**, then further down a **successful probe**

Method: Using open source MPC-HC (Media Player Classic Home Cinema)

_There is a context menu option to enable millisecond precision for time indicator, and you can easily step between frames in both forward (CTRL+Right) and backward (CTRL+Left) directions._

Stepping from frame to frame I could verify a 60 fps video was present by examining the time stamps of successive frames (between 16-18ms per frame, averaging 16.666ms).

**FAIL** timing:

"pin-down without None" PWM blue light to be seen from `25.658` to `26.191` = `0.553`. _give or take a 20ms cycle_
Trigger by probe (Red LED): somewhere between `27.749` and `27.778`
Very interesting: Movement actually visibly goes on, but much slower, for another 4 frames or so **!!!!** until `27.877`. Elasticity of belt? Some other error?
No Blue PWM light to be seen from frame to frame --> **conjecture: No pin_up signal ever sent out**
First BLTouch error indication (blink) begin at `28.613`

`28.613 - 27.778 = 0.835ms`

If one watches the video often enough, one can see the near-second of resting time before the error.

**SUCCESS** timing

22.514 Trigger signal, red LED comes on
------- **No further movement, nozzle virtually stops dead right away**
22.714 Blue light of the PWM goes on, raise-pin operation has begun
...and the rest is a happy end.

KevinOConnor, 2020-04-10:
I added more debugging to the work-bltouch-20200408 branch.  Could you grab that branch, reflash, and test again?

`cd ~/klipper ; git fetch ; git checkout origin/work-bltouch-20200408 ; sudo service klipper stop ; make ; make flash ; sudo service klipper start`

> Is the PWM pin set to Input and Pull-UP or to Output and High-Level when the PWM is turned off?

No - a PWM off results in a digital output low - no different than the low period between pwm pulses themselves.  The pwm code is in src/gpiocmds.c .

-Kevin

lolejar, 2020-04-10:
here you go

[klippy.log](https://github.com/KevinOConnor/klipper/files/4462393/klippy.log)

lolejar, 2020-04-10:
here you go (3 failed events) 2x mesh

[klippy.log](https://github.com/KevinOConnor/klipper/files/4462419/klippy.log)

FanDjango, 2020-04-10:
```
BLTouch send pin_down at 145.966437
BLTouch send None at 146.546437
#output: schedule pwm clock=1516752118 on=65000 off=1935000 new_clock=1711741814 prev_clock=1475409269 flags=60 lazy=0
Stats 186751.6: gcodein=48 mcu: mcu_awake=0.004 mcu_task_avg=0.000011 mcu_task_stddev=0.000020 bytes_write=35159 bytes_read=33661 bytes_retransmit=9 bytes_invalid=0 send_seq=2019 receive_seq=2019 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=8 freq=99996669 heater_bed: target=0 temp=25.1 pwm=0.000 print_time=146.646 buffer_time=2.221 print_stall=0 extruder: target=0 temp=25.1 pwm=0.000 sysload=0.08 cputime=7.608 memavail=691136
bltouch probe @ 146.646437 (check=0.000500)
Stats 186752.6: gcodein=48 mcu: mcu_awake=0.004 mcu_task_avg=0.000011 mcu_task_stddev=0.000020 bytes_write=35200 bytes_read=33792 bytes_retransmit=9 bytes_invalid=0 send_seq=2022 receive_seq=2022 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=8 freq=99996668 heater_bed: target=0 temp=25.0 pwm=0.000 print_time=146.647 buffer_time=0.000 print_stall=0 extruder: target=0 temp=25.1 pwm=0.000 sysload=0.08 cputime=7.641 memavail=691136
#output: schedule pwm clock=1744898817 on=0 off=0 new_clock=1769741814 prev_clock=1745741814 flags=14 lazy=1
Stats 186753.6: gcodein=48 mcu: mcu_awake=0.004 mcu_task_avg=0.000011 mcu_task_stddev=0.000023 bytes_write=35219 bytes_read=33958 bytes_retransmit=9 bytes_invalid=0 send_seq=2024 receive_seq=2024 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=99996669 heater_bed: target=0 temp=25.2 pwm=0.000 print_time=146.647 buffer_time=0.000 print_stall=0 extruder: target=0 temp=25.2 pwm=0.000 sysload=0.07 cputime=7.670 memavail=691136
Stats 186754.6: gcodein=48 mcu: mcu_awake=0.004 mcu_task_avg=0.000011 mcu_task_stddev=0.000023 bytes_write=35536 bytes_read=34164 bytes_retransmit=9 bytes_invalid=0 send_seq=2045 receive_seq=2045 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=99996666 heater_bed: target=0 temp=25.1 pwm=0.000 print_time=147.597 buffer_time=0.000 print_stall=0 extruder: target=0 temp=25.1 pwm=0.000 sysload=0.07 cputime=7.710 memavail=691136
Endstop trigger at 148.123937
mcu trigger position (-2881) -7.202500
BLTouch send pin_up at 148.232667
BLTouch send None at 148.912667
#output: schedule pwm clock=1928456241 on=147500 off=1852500 new_clock=1938364843 prev_clock=1769741814 flags=60 lazy=0
#output: schedule pwm clock=1944001069 on=0 off=0 new_clock=2006364842 prev_clock=1944364843 flags=14 lazy=1
Stats 186755.6: gcodein=48 mcu: mcu_awake=0.004 mcu_task_avg=0.000011 mcu_task_stddev=0.000023 bytes_write=35795 bytes_read=34472 bytes_retransmit=9 bytes_invalid=0 send_seq=2065 receive_seq=2065 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=99996668 heater_bed: target=0 temp=25.2 pwm=0.000 print_time=148.378 buffer_time=0.000 print_stall=0 extruder: target=0 temp=25.1 pwm=0.000 sysload=0.07 cputime=7.762 memavail=691136
Stats 186756.6: gcodein=48 mcu: mcu_awake=0.004 mcu_task_avg=0.000011 mcu_task_stddev=0.000023 bytes_write=35824 bytes_read=34603 bytes_retransmit=9 bytes_invalid=0 send_seq=2068 receive_seq=2068 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=99996668 heater_bed: target=0 temp=25.2 pwm=0.000 print_time=148.378 buffer_time=0.000 print_stall=0 extruder: target=0 temp=25.2 pwm=0.000 sysload=0.07 cputime=7.791 memavail=691136
Failed to verify BLTouch probe is raised; retrying.
BLTouch send reset at 150.152308
BLTouch send pin_up at 151.152308
BLTouch send None at 151.832308
#output: schedule pwm clock=2120682864 on=219000 off=1781000 new_clock=2130328926 prev_clock=2006364842 flags=60 lazy=0
#output: schedule pwm clock=2140904139 on=147500 off=1852500 new_clock=2230328926 prev_clock=2142328926 flags=62 lazy=1
Stats 186757.6: gcodein=48 mcu: mcu_awake=0.004 mcu_task_avg=0.000011 mcu_task_stddev=0.000023 bytes_write=35888 bytes_read=34798 bytes_retransmit=9 bytes_invalid=0 send_seq=2072 receive_seq=2072 retransmit_seq=2 srtt=0.001 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=8 freq=99996671 heater_bed: target=0 temp=25.2 pwm=0.000 print_time=148.378 buffer_time=0.000 print_stall=0 extruder: target=0 temp=25.1 pwm=0.000 sysload=0.07 cputime=7.833 memavail=691136
#output: schedule pwm clock=2239314875 on=0 off=0 new_clock=2298328926 prev_clock=2240328926 flags=14 lazy=1
Stats 186758.6: gcodein=48 mcu: mcu_awake=0.003 mcu_task_avg=0.000009 mcu_task_stddev=0.000013 bytes_write=35902 bytes_read=34959 bytes_retransmit=9 bytes_invalid=0 send_seq=2073 receive_seq=2073 retransmit_seq=2 srtt=0.001 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=99996671 heater_bed: target=0 temp=25.2 pwm=0.000 print_time=148.378 buffer_time=0.000 print_stall=0 extruder: target=0 temp=25.1 pwm=0.000 sysload=0.07 cputime=7.861 memavail=691136
Endstop trigger at 151.833308
```

KevinOConnor, 2020-04-10:
Everything in the log indicates Klipper is operating correctly.  I put together another test branch that will log every pwm pin change:

`cd ~/klipper ; git fetch ; git checkout origin/work-bltouch-debug-20200410 ; sudo service klipper stop ; make ; make flash ; sudo service klipper start`

This branch isn't suitable for actual printing due to the volume of logging.

-Kevin

[7 more replies]

closed 2020-05-24 by KevinOConnor