Klipper3d/klipper #2694: Shutdown after Z-move

dvanp, 2020-04-07:
Since this morning my printer goes into Shutdown after a z-move (on both z steppers).
I'm running an SKR 1.3 with 2208 in UART mode, which has been working great for some months now. This morning during the first print of the day it went into shutdown and I discovered that this was caused by a Z-move, because when I homed the printer after the shutdown it went into shutdown again when homing Z.
After that I did a stepper buzz on X and Y, which gave no issues, and then tried both z steppers, which both resulted in shutdown.

Things I've tried:
- replacing the stepper drivers
- the stepper motors
- reflashing firmware to SKR
- reflashing clean OctoPi (1.4.0) and klipper on RPI 3B

After the clean install I did a Stepper buzz on the Z axis, after which the printer went in shutdown immediately and that's reflected in the log:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4444213/klippy.log)

The only thing I can think off besides what I've already tried is replacing the SKR, but I'm hoping to find a simpler solution here :)

KevinOConnor, 2020-04-08:
This error is the result of a quirk in the STEPPER_BUZZ command - it can sometimes lead to this error on steppers with very fine resolution and low acceleration settings.  If you increase the acceleration in your config temporarily (eg, to 3000) this error should go away for STEPPER_BUZZ.

We'd need to see the log from the failed homing attempt to understand what went wrong there (homing would not exhibit the problem above).

Also, you have two different step distances for z and z1 - that's unusual (but shouldn't cause a problem).

-Kevin

dvanp, 2020-04-09:
I just powered on the printer and homed it, without any issues. 
"Let's print!", I thought.
Printer: "Nope."

Immediately after starting the gcode it goes into shutdown. I've attached the log.
[klippy.log](https://github.com/KevinOConnor/klipper/files/4455541/klippy.log)

So it's not the z-axis that's causing the issues..

I'll check all the wiring again in the mean time, because it occurred to me it could be a wiring issue. My last check didn't reveal anything, but maybe I missed it.

dvanp, 2020-04-09:
> I'll check all the wiring again in the mean time, because it occurred to me it could be a wiring issue. My last check didn't reveal anything, but maybe I missed it.

And miss it I did. It seems to have been the power connector which was a bit loose. I retightened it and the printer is back to work!

Closing this!

> Also, you have two different step distances for z and z1 - that's unusual (but shouldn't cause a problem).

Fixed this as well, of course. Thanks a lot Kevin, for both you help and the great firmware!

closed 2020-04-09 by dvanp