Klipper3d/klipper #2695: progress bar shows 0%

4noxx, 2020-04-07:
hello

i use a SKR E3 Mini with a RepRap 12864 Display and Repetier Server.

Problem: The progress bar shows only 0%. All other works perfect.
Repetier shows the correct value, but not Klipper / Display.

[klipper.log](https://github.com/KevinOConnor/klipper/files/4446701/klipper.log)

KevinOConnor, 2020-04-08:
In order for the lcd display to show a progress bar, either the print has to be from a virtual_sdcard, or the user interface (OctoPrint or other) must send M73 commands to Klipper that explicitly state what the current progress is.

-Kevin

4noxx, 2020-04-08:
thx, i dont know if Repetier can send M73 commands, i think no.
so, then i doesnt need the status bar, can i delete or disable it from my mainscreen?

mcmatrix, 2020-04-08:
> thx, i dont know if Repetier-Server can send M73 commands, i think no.
> so, then i doesnt need the status bar, can i delete or disable it from my mainscreen?

@4noxx Yes, it can but you need to modify the configuration file manually.
If you log via ssh to your repetier-server then go `/usr/local/Repetier-Server/firmware/`
I assume that you're using Klipper firmware in repetier-server config. 

Open file `Klipper.xml` for editing.
Remove comments from the line that is surrounded by red rectangle.
[image]
I also added comments to the next line (don't know if this is needed)
Comment markup surrounding a tag``

4noxx, 2020-04-09:
@mcmatrix 

thx, works :)

closed 2020-04-09 by 4noxx