Klipper3d/klipper #2697: Fysetc cheetah 1.1b - Print stops with "Lost communication with MCU 'mcu'"

csutcliff, 2020-04-08:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4447792/klippy.log)

Installed this board, seemed to all be working OK on homing, calibration and a few small short test prints.

First two larger prints have both failed at similar time durations. Attached is a log example.

What can I do to troubleshoot?

KevinOConnor, 2020-04-08:
The best information we have on this is in the FAQ: https://www.klipper3d.org/FAQ.html#i-keep-getting-random-lost-communication-with-mcu-errors

-Kevin

csutcliff, 2020-04-11:
After some research and experimenting I think I have a solution for this board. Compiling the firmware for a 256000 baud rate and setting

```
[mcu]
baud: 256000
```

seems to work, I have a couple of days of clean printing with that.

It is based on a forum post from the creater of the board saying that 152000 and 256000 are the most reliable baud rates for flashing firmware (his intended use of the serial port) combined with some LKML threads and kernel patches for the CH431 usb chip that cover the Linux driver's reliability when the baud rates are not in sync with the internal clock.

KevinOConnor, 2020-04-24:
I'm going to close this as it looks like a workaround was found for the original problem report.

-Kevin

closed 2020-04-24 by KevinOConnor

grigorig, 2020-04-29:
The Linux kernel driver version that most people still use (on Raspberry Pi) still has issues with baudrate calculations. Baudrate calculations were inaccurate in many cases and this was only fixed quite recently. It is a bit odd, though, as a baudrate of 250K can be hit easily and perfectly on both STM32 and CH341 and I think the old driver did this well. So this is a bit of a mystery to me. Can you link the relevants discussions @csutcliff?

grigorig, 2020-09-26:
This has now been closed for a long time, but I've found a likely cause: bad noise suppression with ferrite beads. The GND line on the USB connection of Cheetah boards actually has a ferrite bead in it. Bad idea, because USB is a high-speed signal and it will be attenuated. In addition you get bad grounding overall. The culprit is the bead L4. Just remove it and bridge the pads.

Overall the Cheetah boards are horribly engineered. Avoid them if you can!