Klipper3d/klipper #2698: Unknown value 'P0.24' in enumeration 'pin'

vikbessems, 2020-04-08:
- Board: Makerbase MKS Gen L V1.0
- Delta printer: FLSUN Kossel Delta
- OctoPrint 1.4.0 running on OctoPi 0.17.0

I followed the online setup, when making my config file I took the example-delta config and added in Pin values from generic-mks-**S**genl, they should be the same as the mks gen l?

After running restart on octopi I am greeted with the message Unknown value P0.24 in inumeration 'pin'. Which is the sensor_pin of [heater_bed].

[klippy.log](https://github.com/KevinOConnor/klipper/files/4452055/klippy.log)
[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/4452096/printer.cfg.txt)

KevinOConnor, 2020-04-08:
The Gen L is not at all similar to the sgen-L.  Use the config/generic-ramps.cfg file as an example for the MKS Gen L.

-Kevin

vikbessems, 2020-04-08:
Thanks for the quick response issue solved!

closed 2020-04-08 by vikbessems