Klipper3d/klipper #2701: Unable to connect

Stuie101, 2020-04-09:
New to all this.

Trying to install Klipper to a Delta Printer using a SKR 1.3 Board

Firmware was installed to the SKR 1.3 using the SD card method.

Trying to find the usb connection using ls /dev/serial/by-id/ fails see below:

pi@octopi:~ $ ls /dev/serial/by-id/
ls: cannot access '/dev/serial/by-id/': No such file or directory
pi@octopi:~ $ ls /dev/serial/by-id/*
ls: cannot access '/dev/serial/by-id/*': No such file or directory
pi@octopi:~ $ cd klipper
pi@octopi:~/klipper $ ls /dev/serial/by-id/
ls: cannot access '/dev/serial/by-id/': No such file or directory
pi@octopi:~/klipper $ ls /dev/serial/by-id/*
ls: cannot access '/dev/serial/by-id/*': No such file or directory
pi@octopi:~/klipper $

Searched the forum and set the following in printer.cfg through octoklipper

[mcu]
serial: serial: /dev/ttyUSB0
baud: 115200

This looks like it allows connection to be established, resulting in the following on the screen in octoklipper:

08:22:20 Connected to host via /tmp/printer @115200bps
08:22:20 Firmware version: v0.8.0-410-ga028aeaf

but clicking the get status button in Octoklipper  or trying to home, or any movement, resulting in the following on the screen in octoklipper:

08:22:25 Unable to connect
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Error configuring printer
Klipper state: Not ready
08:22:25 Unable to connect

[klippy.log](https://github.com/KevinOConnor/klipper/files/4454625/klippy.log)

Stuie101, 2020-04-09:
Have remade and install firmware via SD Card this time to include USB.

Same issue but new log
[klippy.log](https://github.com/KevinOConnor/klipper/files/4454980/klippy.log)

Stuie101, 2020-04-11:
Removing SD card seems to have fixed this problem. Thought it was like Marlin and the SD card had to remain as EEPROM. apparently  not the case.

SD Card out, reboot Pi and printer,  ls /dev/serial/by-id/*, quick copy paste to printer.cfg and connection made.

Now finding SAVE_CONFIG is not saving anything to the printer.cfg.

PID_CALIBRATE runs fine but SAVE_CONFIG does nothing.

I want to move on to DELTA_CALIBRATE etc but SAVE_CONFIG kinda needs to be working for that to have any point to it.

ld3300, 2020-04-13:
Can you do a PID_CALIBRATE, then SAVE_CONFIG, then M112 and post the klippy.log

Stuie101, 2020-04-13:
Hi id3300

On a facebook group I was told to check permissions and check that the cfg file was not open.
This appeared to do the trick and allowed PID for hotend and heatbed to be saved.

After that however it refused to save end stop phase Probe calibration or delta calibration.

I was then advised to ls -l ~/printer.cfg. 

and then chmod 666 /home/pi/printer.cfg

But it still does not save properly.

chmod 666 /home/pi/printer.cfg produced 3 x rw when I ran ls -l ~/printer.cfg after it.

I tried an end stop phase calibration, but nothing saved.

However this time when the everything booted up, a really bad bed level that I did earlier in the afternoon appeared to have saved, but it didnt show up at the time.

So currently cfg file has PID for hotend and heat bed and a bad bed level, the system seems hit and miss as to when it will save and then when the save will show up in the cfg.

I think I am doing something wrong process wise.

I run a calibration, data is returned, I enter CONFIG_SAVE in the terminal go to octoklipper plugin to see the CFG file and nothing is added.

In light of the above wIll doing a PID_CALIBRATE, then SAVE_CONFIG, then M112 and posting the klippy.log still help you diagnose he issue. or would something else be more helpful?

Alan Lord Check the permissions and ownership of your config file.
Hide or report this
Like
 · Reply · 1d
Stuart Powles
Stuart Powles Thanks for the quick response.
I am assuming I should tick the other 2 W boxes?
Edit or delete this
No photo description available.
Like
 · Reply · 1d
Stuart Powles
Stuart Powles Ticked both boxes ran PID Tune still nothing saved.
Send: PID_CALIBRATE HEATER=heater_bed TARGET=60 WRITE_FILE=1…See More
Edit or delete this
Like
 · Reply · 1d
Paul Arnold
Paul Arnold The pid values after the save_config will appear at the very bottom of your config file.
Dont leave your config file open in an editor while saving via the terminal
Hide or report this
Like
 · Reply · 1d
Stuart Powles
Stuart Powles Yeah it aint putting anything at the end of the printer config file.
Good point on the file being open else where. I had a lot of stuff open so it printer config could have been open somewhere. Closed all files now and will try again.
1
Edit or delete this
Like
 · Reply · 1d
Stuart Powles
Stuart Powles Many thanks guys that has fixed it. 👍
Now onto PROBE_CALIBRATE and DELTA_CALIBRATE.😃
Edit or delete this
Like
 · Reply · 1d
Andrew Moran
Andrew Moran U dont have to remove the sd card for usb to work
Hide or report this
Like
 · Reply · 20h
Stuart Powles
Stuart Powles Really that seemed to be the fix. It was indicated in a post on the github issues page
Edit or delete this
Like
 · Reply · 20h
Andrew Moran
Andrew Moran Stuart Powles I haven't removed mine works just just fine
Hide or report this
Like
 · Reply · 20h
View 1 more reply
Stuart Powles
Stuart Powles Still having issues with SAVE_CONFIG though got heatbed PID saved with out issue. Extruder PID took several attempts and endstop phase , probe calibration and delta calibration just wont save at all
Edit or delete this
Like
 · Reply · 19h
Christopher Wilson
Christopher Wilson Sounds more like a permissions issue with the printer.cfg vs what user you're running Klipper under.
Hide or report this
Like
 · Reply · 16h
Hide 13 Replies
Stuart Powles
Stuart Powles Dont recall ever being asked to create a user for Klipper.
How would I check or change it?
Edit or delete this
Like
 · Reply · 16h
Christopher Wilson
Christopher Wilson ls -l ~/printer.cfg. Mine is -rw-r--r--. If you're the only one running on whatever computer you have klipper installed on, you can verify it isn't a permission issue by doing "chmod 777 ~/printer.cfg" if that suddenly starts saving the data, then it is a permissions issue.
Hide or report this
Like
 · Reply · 16h
Stuart Powles
Stuart Powles pi@octopi:~ $ ls -l ~/printer.cfg
-rw-r--r-- 1 pi pi 5023 Apr 12 16:00 /home/pi/printer.cfg
Edit or delete this
Like
 · Reply · 16h
Christopher Wilson
Christopher Wilson cat /etc/default/klipper
Hide or report this
No photo description available.
Like
 · Reply · 16h
Christopher Wilson
Christopher Wilson /etc/init.d/klipper reads the defaults in /etc/default/klipper.
Hide or report this
Like
 · Reply · 16h
Stuart Powles
Stuart Powles Loosing me going way over my head. dumb it down to an IQ of 70 🤔
Edit or delete this
Like
 · Reply · 16h
Stuart Powles
Stuart Powles Typed in cat /etc/default/klipper and got:
_USER=pi…See More
Edit or delete this
Like
 · Reply · 16h · Edited
Christopher Wilson
Christopher Wilson try "chmod 666 /home/pi/printer.cfg". this makes the file where everyone can read/write to the file. Then restart klipper and try your calibration and SAVE_CONFIG again.
Hide or report this
Like
 · Reply · 16h
Stuart Powles
Stuart Powles 666 or 777 youve changed it from the comment further up or is one a typo?
Edit or delete this
Like
 · Reply · 16h
Christopher Wilson
Christopher Wilson 666 is read, write, no execute for you, your group, and everyone. 777 is read, write, execute. There's no reason to execute. 777 is the default thing to try if you have any issues, but as this isn't an executable file or a directory, you don't need execute permissions. So go with 666.
Hide or report this
Like
 · Reply · 15h
Stuart Powles
Stuart Powles Will give it a go and report back. Thanks for your help
Edit or delete this
Like
 · Reply · 15h
Stuart Powles
Stuart Powles chmod 666 /home/pi/printer.cfg produced 3 x rw when I ran ls -l ~/printer.cfg after it.
Tried an end stop phase calibration, but nothing saved
However this time when the everything booted up a really bad bed level that I did earlier in the afternoon appeared to have saved, but it didnt show up at the time.

theopensourcerer, 2020-04-13:
You need to perform a SAVE_CONFIG and then upload the klippy.log file. Maybe there will be an error recorded in the log which will help diagnosis. (klippy.log should be in /tmp).

Stuie101, 2020-04-13:
Ok so this morning I did a PID of the hot end which appear to save, followed by a End stop phaes calibrate which appeared not to have saved when I restarted the firmware following the m112.

Klippy log attached.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4469157/klippy.log)

Stuie101, 2020-04-13:
Ok so this is weird Just thought I would have another go while waiting for a response, so ran 

the End Stop Phase Calibrate again,
followed by save config
Followed by M112
then reconnected
then hit a restart firmware

And then checked Config in Octoklipper and nothing had saved.

Undeterred I then ran:

the Delta Calibrate,
followed by save config
Followed by M112
then reconnected
then hit a restart firmware

And then checked Config in Octoklipper and delta had saved and so had the End stop phase.

The numbers look horrible but there are some known mechanical issues that also need resolving so I am not too surprised or concerned by that. It is concerning that the saved data does not show up straight away or simply does not save on occasions.

Klippy log attached

ld3300, 2020-04-13:
I haven't used it myself, but other users have complained about issues with octoklipper and config files. Are you able to access the file system of your pi and open the config file with notepad++?

Stuie101, 2020-04-13:
Yeah have WinSCP and Notepad++.

I just thought that that was the whole point of octoklipper, to make it simple to do it all from the single web based  interface?

ld3300, 2020-04-13:
I think that was the intent, but the repository hasn't been updated in 2 years, so it probably has some unexpected issues.  The plugin isn't developed by KevinOConnor.

tony3d778, 2020-04-13:
Hi guys,

Following the first issue on this ticket, I have the exact same trouble as @Stuie101 identifying the serial port on the pi. I also have the SKR 1.3 board (on an ender 3) and trying to find the usb connection using ls /dev/serial/by-id/ is just impossible... I tried everything from trying differant iterations of that command to find the serial port, completly reinstalling Klipper (uninstalled and reinstalled), also tried to take out the SD card and restarting the system to see if the pi then take the command and every other tricks on this issue available on the internet but nothing... I'm a bit frustrated I have to say lol. 

Can anyone help me on that? Would be so much appreciated, don't know how many hours I spent. 

I attached the most recent log file I have, thank you!!

[klippy.log](https://github.com/KevinOConnor/klipper/files/4472266/klippy.log)

ld3300, 2020-04-15:
Can you try `ls /dev` and see if "serial" shows up.  If not will you try swapping out the USB cable and do it again? Just to confirm, the SKR board needs to be powered during these steps.

tony3d778, 2020-04-15:
@ld3300 Okay just tried it. Here is the result : 
[image]

ld3300, 2020-04-17:
It looks like the usb connection isn't working, otherwise we would see serial in that list.  Did you try  different USB cable?  Maybe different ports.

tony3d778, 2020-04-17:
@ld3300 I was finally able to install it correctly, I tought I had the Mini E3 v1.3 but is was actually the v1.1... stupid I am! After I was able to install it I had problems with the stepper making a lot of noise when axis was moving and finally, I deciced to go back to Marlin 2.0. Maybe I'll try it again later! 
Thanks for the help tough :)

ld3300, 2020-04-18:
No problem.  I do the best I can.  You might need to add the config sections for your steppers if they are TMC220xs.  That should resolve motor issues.

KevinOConnor, 2020-04-24:
I'm going to close this as it looks like the original reporter found the root cause (different board).

-Kevin

closed 2020-04-24 by KevinOConnor

rafikipl, 2021-07-27:
Hi,

I have a similar problem, I have a skr mini e3 v2.0 board, I checked it. normally with Marlin, the USB port is detected, in octopi it is detected, but after uploading the clipper according to the SKR instructions, it is no longer visible, the clipper is prepared according to the recommendations, that is:
# This file contains common pin mappings for the BIGTREETECH SKR mini
# E3 v2.0. To use this config, the firmware should be compiled for the
# STM32F103 with a "28KiB bootloader" and USB communication. Also,
# select "Enable extra low-level configuration options" and configure
# "GPIO pins to set at micro-controller startup" to "! PA14".

# The "make flash" command does not work on the SKR mini E3. Instead,
# after running "make", copy the generated "out / Klipper.bin" file to a
# file named "firmware.bin" on an SD card and then restart the SKR
# mini E3 with that SD card.

after uploading I get:
ls: cannot access '/ dev / serial / by-id / *': No such file or directory

log in the attachment
Someone can help?

P.S. from the sticker it looks like the disc is dated 2021.5.19
I can upload Marlin every time and everything works.
[klippy.log](https://github.com/KevinOConnor/klipper/files/6887910/klippy.log)