Klipper3d/klipper #2702: Configuring a custom reset switch?

dothebart, 2020-04-09:
on other solutions, one can configure a reset switch:
http://smoothieware.org/switch#setting-up-a-reset-button
It generaly soft-wires the a button to a generic reset command ( src/modules/utils/simpleshell/SimpleShell.cpp )
and under the hood this also invokes `NVIC_SystemReset` as `src/generic/armcm_reset.c` does in klipper. 
Since klipper will only load the configuration into the board, this would probably have to be configured before compiling the firmware.

Any thoughts?

ld3300, 2020-04-13:
Are you using Octoprint?  If so install the Action Commands plugin (not the same as Action Command Prompt) and you can configure the button with a gcode "RESPOND" to send an action command back to Octoprint that can then execute a gcode or system command.  Does that get you what you need?

dothebart, 2020-04-14:
Hi, 
Yes - I use octoprint. But I have situations where it fails to communicate with the MCU.

KevinOConnor, 2020-04-14:
It's possible to configure a "kill button" in the display config section.  It's also possible to configure a [gcode_button] config section that invokes a shutdown.

-Kevin

dothebart, 2020-04-23:
But these will only take effekt once the python klipper daemon was able to communicate with the MCU? I'm sugesting something which will be there from ever on in the lifetime of the MCU.

KevinOConnor, 2020-04-24:
FWIW, Klipper is designed to be a 3d printer firmware that runs on a Raspberry Pi (or similar low-cost general purpose computer).  It's not a goal of mine to add features for the case where a Raspberry Pi is not present.

-Kevin

KevinOConnor, 2020-05-11:
I'm going to close this as it looks like the original question was answered (it's possible to configure an action to take on a button press in the host config file).

-Kevin

closed 2020-05-11 by KevinOConnor