Klipper3d/klipper #2704: Error:  Internal error on command:"G1"

tominabox1, 2020-04-09:
Seems to be a few instances of this error in the past but I sliced this file twice (slightly different models) and the error occurred nearly the same time partway through the 2nd to last layer.  Included Gcode 

[WD9_Bottom_Leftlll.gcode.zip](https://github.com/KevinOConnor/klipper/files/4457267/WD9_Bottom_Leftlll.gcode.zip)

[klippy.log](https://github.com/KevinOConnor/klipper/files/4457255/klippy.log)

justh1n10, 2020-04-09:
Are you using Cura 4.6 by any chance? I did get the same exact error today using 4.6.

tominabox1, 2020-04-09:
I'm on 4.5

justh1n10, 2020-04-11:
Here is my log + gcode, caused the same error: 
[klippy.log](https://github.com/KevinOConnor/klipper/files/4464346/klippy.log)
[CE5_mosquito mount.zip](https://github.com/KevinOConnor/klipper/files/4464348/CE5_mosquito.mount.zip)

ld3300, 2020-04-13:
Are you sure it isn't the Heater Extruder issue? from log (line 14492):

> Heater extruder not heating at expected rate
Transition to shutdown state: Heater extruder not heating at expected rate
See the 'verify_heater' section in config/example-extras.cfg
for the parameters that control this check.

KevinOConnor, 2020-04-15:
It looks like you are running code other than the code found in this repository.  You'll need to reproduce the error using the pristine code found here, or you'll need to report the error to the dwc2 developers.

-Kevin

tominabox1, 2020-04-21:
I've updated my klipper code to the current code. I have had several successful print since this error so have no problem closing this right now.

closed 2020-04-21 by tominabox1