Klipper3d/klipper #2707: Print speed throttling based on nozzle pressure

theshamot, 2020-04-10:
Hello,
when printing fast, my extruder stepper is loosing stepps from time to time on long straight lines, because it is not strong enough to force that big amount of filament throught nozzle for long time.

The question is:
Since klipper is calculating pressure inside nozzle for Pressure advance, is it possible to retrieve calculated pressure or even slow down print speed when there is "pressure exceeding preset limit" detected?

Thank you :)

Hywelmartin, 2020-04-10:
there isn't any software fix for this... get a longer heater block..

you are simply extruding more than your hotend can handle..

theshamot, 2020-04-11:
but is there a way to retrieve calculated pressure then?

Hywelmartin, 2020-04-11:
I don't think the calculated pressure is of any use, due to that the viscosity is radically different when the polymer isn't totally melted.. have you calculated the volume flow...???

lbibass, 2020-04-14:
If you have problems with skipping steps, if you're using Slic3r or one of its many variants, (slic3r++, PrusaSlicer) you can set a max volumetric flowrate. Try that.

KevinOConnor, 2020-04-14:
As indicated, this isn't currently implemented.  I'm sure it's possible to implement, but an interested developer would need to take it on and submit it.

-Kevin

KevinOConnor, 2020-04-14:
This looks like a dup of #2610 .

-Kevin

KevinOConnor, 2020-04-24:
I'm going to close this in favor of the existing #2610. 

-Kevin

closed 2020-04-24 by KevinOConnor