Klipper3d/klipper #2708: Klipper + Arduino Uno/Nano

lelibreauquotidien, 2020-04-10:
Hello.
I have an arduino uno / nano and a raspberry pi 3b.
I would like to know if it is possible to use klipper only with my arduino nano / uno and my raspberry?
Thank you.

Hywelmartin, 2020-04-10:
i'm using a Nano.. so yes

gabork, 2020-04-13:
It is very much possible use Klipper with an Arduino Uno. I use one as an expansion board to my Mega+RAMPS. If you are using it with a CNC Shield, you can look up the pin configuration for Klipper here: https://www.thingiverse.com/thing:3490163

The question is, what do you want to build, and what can you build with **only** the listed parts. There are CNC Shields available for both the Uno and the Nano, so a CNC/laser engraver machine sounds possible with only those microcontrollers. On the other hand, there are no 3D printing shields for them AFAIK, so you'd have to make a custom solution for the heating elements, at which point using an Arduino Mega with a RAMPS shield sounds more practical.

KevinOConnor, 2020-04-24:
I'm going to close this as it looks like the original question was answered (it is possible).

-Kevin

closed 2020-04-24 by KevinOConnor