Klipper3d/klipper #2709: Can't connect klipper

Mckevinho, 2020-04-11:
Hi,

I use marlin for long time and i wanna change to Klipper because i wanna use 2 boards in one printer, to have enough pins for all the features ;).
I installed Klipper on my raspberry pi4 and flashed the firmware to my MKS gen L board without problems.
I installed also the octoklipper plugin on octoprint. Now i try to connect with klipper but its doesn't work. Klipper is new for me and i can't find the solution on my own. I get always next message.

Unable to connect
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Error configuring printer
Klipper state: Not ready
05:02:46 Unable to connect

For so far as i guess is it something with USB connection. The strange thing is octoprint also doesn't see the printer.cfg file on the raspberry.

I hope someone can help me.
[klippy.log](https://github.com/KevinOConnor/klipper/files/4463988/klippy.log)

Kevin

neophrema, 2020-04-11:
Silly questions, 
you did remove the SD card, right?
Did you configure octoprint according to the manual of klipper?
https://www.klipper3d.org/Installation.html
`Unable to open port: [Errno 2] could not open port /dev/ttyACM0: [Errno 2] No such file or directory: '/dev/ttyACM0' `
this seems odd to me, shouldnt there be something about "`/tmp/printer`"?

sincerely,
v.

BlackStump, 2020-04-11:
https://www.klipper3d.org/FAQ.html#wheres-my-serial-port

Mckevinho, 2020-04-11:
Hi,

Yeah i did the "/tmp/printer" as serial port in octoprint and also at the addiotional serial ports box.
in the behaviour topic, i marked this checkbox: 

Cancel any ongoing prints but stay connected to the printer
Please note that if you choose this, OctoPrint will still disconnect from the printer in case of fatal errors reported by your firmware (e.g. kill() called, fatal:).

 this serial port (/dev/ttyACM0) was first in the printer.cfg file but i changed it to /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0

i did all this configuration steps in octoprint like the manual descibed.

BlackStump, 2020-04-11:
In your printer.cfg this line needs to be checked against the actual id of your connection, hence why I sent you the link  - where is my serial port.
>The general way to find a USB serial port is to run ls /dev/serial/by-id/* from an ssh terminal on the host machine. It will likely produce output similar to the following:
/dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0

Mckevinho, 2020-04-12:
thx for response :smiley: 

in the printer.cfg I puted next:
[mcu]
serial = /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0

now there is a new error with the connection. Something like he doesn't get response and he stops with connecting.
Starting with klipper is not easy :wink: 

I've uploaded a new klippy.log

[klippy.log](https://github.com/KevinOConnor/klipper/files/4465901/klippy.log)

BlackStump, 2020-04-12:
what does it show when you log into your Pi and run this command

ls /dev/serial/by-id/*

KevinOConnor, 2020-04-15:
It looks like you've changed the baud rate in "make menuconfig".  Run "make menuconfig" again and do not enable "low-level options".  Then reflash the board.  See the FAQ for further details.

-Kevin

Mckevinho, 2020-04-15:
This is the USB connection i get
[image]

So i made also a new menuconfig without the low level option
[image]

You can see i have also an printer.cfg file
[image]

but this message stay :(
[image]

Mckevinho, 2020-04-15:
I found the problem. It was the capital letter in "Printer.cfg".
Now he connects without problem🎉

Thanks all for helping me😊

Tomorrow I will change the board in my printer and test the klipper with a print.
The current board (skr1.4) stays remember the marlin while he has klipper on the sd now.

So I will changed it to the MKS gen L board. That works with the klipper.😊

KevinOConnor, 2020-04-24:
I'm going to close this as it looks like the solution was found.

-Kevin

closed 2020-04-24 by KevinOConnor

bogeyf14, 2021-06-28:
it sounds like OP didn't solve the original problem and instead used USB.

A solution that worked for me was to modify the printer.conf
instead of using "serial: /dev/ttyAMA0"
I have to use      "serial: /dev/serial0"