Klipper3d/klipper #2710: Ender 3 Pro installation problem

johncmolyneux, 2020-04-11:
I'm attempting to get Klipper running on my Ender 3 Pro.  I have followed the instructions on this page:

https://www.klipper3d.org/Installation.html

I get as far as the make flash command and get the following error...

avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x65

(1 of 10, up to 10 of 10)

I've attached the log file, as requested.  I do hope you can help me get this up and running.
Thanks.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4464268/klippy.log)

BlackStump, 2020-04-11:
https://www.klipper3d.org/FAQ.html#the-make-flash-command-doesnt-work

https://www.klipper3d.org/Bootloaders.html

johncmolyneux, 2020-04-11:
Thanks for the reply.  I've already worked through the suggestions in the first link.  I am having problems with OctoPrint reconnecting as soon as I click "disconnect", so I've been stopping the service via SSH before trying to flash Klipper.

Also, I don't think the 2nd link will be any help as I've previously flashed a bootloader and updated the firmware to a derivative of Marlin 2 (TH3D).  I'll have a thorough read through it and see if there's anything in there that will help.

KevinOConnor, 2020-04-24:
I'm going to close this as it looks like the conversation has concluded.  Some boards are flashed differently and it doesn't look like anyone here knows the particular sequence for that board.

-Kevin

closed 2020-04-24 by KevinOConnor

johncmolyneux, 2020-04-24:
That's fine.  I've not actually had chance to look into it any further yet
as I need to keep my printer working right now and don't want to risk
killing it, even for a short amount of time.  I'll revisit this one day and
probably come back for help then.

Thanks :)