Klipper3d/klipper #2711: Extruder steps differ due to pressure at higher speeds

trollkarlen, 2020-04-11:
Is this correct due to the extruder getting too cold or is it something that can be compensated with like pressure advance ?
Maybe its something that can get manual help to the physics with a settings in config like
step_distance: 0.00975
step_distance_500: 0.00935

//T

Hywelmartin, 2020-04-11:
#2707 looks like the same... get a hotend that support the mass flow you are trying  to extrude
#2610

trollkarlen, 2020-04-11:
A little in the same ball park as #2610.

Its not only hotend issue(it can be) but a fluid dynamics problem that can
be compensated for so higher speed prints get better quality.

//T

KevinOConnor, 2020-04-14:
As indicated this looks like a dup of #2610. 

-Kevin

KevinOConnor, 2020-04-24:
I'm going to close this in favor of the existing #2610 .

-Kevin

closed 2020-04-24 by KevinOConnor

trollkarlen, 2020-04-24:
Sounds good.

//R