Klipper3d/klipper #2712: Gcode not recognized?

ddmkr666, 2020-04-11:
I cannot print any file with klipper.
The gcode commands seem to be unrecognized?

Changing monitoring state from "Operational" to "Starting"
Send: N0 M110 N0*125
Recv: ok
Changing monitoring state from "Starting" to "Printing"
Send: N1 M82*24
Recv: ok
Send: N2 M140 S60*81
Recv: ok
Send: N3 M104 S230 ?T0*28
Recv: !! Error on 'N3 M104 S230 ?T0*28': unable to parse 230 ?
Changing monitoring state from "Printing" to "Cancelling"
Send: N4 M108*46
Recv: ok
Send: N5 G91*20
Recv: // Unknown command:"M108"

[klippy.log](https://github.com/KevinOConnor/klipper/files/4464524/klippy.log)

theophile, 2020-04-11:
Why is there a question mark in that line?

ddmkr666, 2020-04-11:
There isn't. It's how it looks in the actual gcode:
`;FLAVOR:Marlin
;TIME:6993
;Filament used: 11.2366m
;Layer height: 0.2
;MINX:73.515
;MINY:73.534
;MINZ:0.2
;MAXX:161.485
;MAXY:161.466
;MAXZ:50
M82 ;absolute extrusion mode
M140 S60 ; start preheating the bed
M104 S215 ﻿T0 ; start preheating hotend
M190 S60 ; heat to Cura Bed setting 
G28 ; home
G29; ABL
G92 E0 ; Reset Extruder`

This exact same g-code printed on Marlin without any problems.

theophile, 2020-04-11:
On the "start preheating hotend" line, either remove the "T0" or put it before the "S215."

BlackStump, 2020-04-11:
you should read what gcode Klipper supports, Klipper is not Marlin.
https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md

ddmkr666, 2020-04-12:
Klipper's goal is to support the G-Code commands produced by common 3rd
party software (eg, OctoPrint, Printrun, Slic3r, Cura, etc.)
Well, that's what's written there. So why doesn't it support default Cura
gcode?

WhereIsTheExit, 2020-04-12:
I had the same problem. For me the cause was CHEP's start G-code for heating up the hotend and the heated bed at the same time. I just copy pasted it from his site, which inserted two 0xfeff characters (byte order mark?) into the G-code. Klipper doesn't recognize this and halts with an error (which is good, Marlin silently ignores it). 
The characters are after the variables:
[image]
Delete the spaces in your printer config and manuall type them back in, fixed it for me.

ddmkr666, 2020-04-12:
Ah, that makes sense. Thanks.

KevinOConnor, 2020-04-24:
I'm going to close this as it appears the original poster found a solution.

-Kevin

closed 2020-04-24 by KevinOConnor