Klipper3d/klipper #2713: CoreY/X ?

catchra, 2020-04-11:
@KevinOConnor Hi I am just getting started with Klipper and so far it is great but the only problem is the lack of the CoreYX kinematics.
I did look at `./klipper/klippy/kinematics/corexy.py` but i have no idea on how to modify it for coreyx.

is it possible to add coreyx to klipper?
I did try to switch the `[stepper_x]` and `[stepper_y]` in my printer.cfg file but it still moves the wrong axis just in the opposite direction.

Henke1983, 2020-04-12:
Have you looked in here: https://github.com/KevinOConnor/klipper/blob/master/config/example-corexy.cfg

Under [printer] you have to change the kinematics to corexy else it will not work, have you tried that?

I don´t think you should change the corexy.py - because thats based on calculations how the movement for corexy are done.

mkuf, 2020-04-12:
> 
> I did try to switch the `[stepper_x]` and `[stepper_y]` in my printer.cfg file but it still moves the wrong axis just in the opposite direction.

That sounds like one of your steppers is moving in the wrong direction.
Try inverting the step direction of one of them to make the right axis move.

chris666uk1, 2020-04-12:
here is my config with d-bot and duet if that helps

`[stepper_x]
step_pin: PD6
dir_pin: !PD11
enable_pin: !PC6
step_distance: .0100
endstop_pin: ^!PC14
position_endstop: 0
position_max: 290

[tmc2660 stepper_x]
cs_pin: PD14
spi_bus: usart1
microsteps: 16
run_current: 0.900
sense_resistor: 0.051

[stepper_y]
step_pin: PD7
dir_pin: !PD12
enable_pin: !PC6
step_distance: .0100
endstop_pin: ^!PA2
position_endstop: 290
position_max: 290

[tmc2660 stepper_y]
cs_pin: PC9
spi_bus: usart1
microsteps: 16
run_current: 0.900
sense_resistor: 0.051

[stepper_z]
step_pin: PD8
dir_pin: !PD13
enable_pin: !PC6
step_distance: .000861
endstop_pin: probe:z_virtual_endstop
#endstop_pin: PD10  # E0 endstop
#endstop_pin: PC16  # E1 endstop
position_endstop: 0
position_max: 200

[tmc2660 stepper_z]
cs_pin: PC10
spi_bus: usart1
microsteps: 16
run_current: 0.900
sense_resistor: 0.051

# Define a probe
[probe]
pin: !PC1
x_offset:  25
#z_offset:  4

[extruder]
step_pin: PD5
dir_pin: !PA1
enable_pin: !PC6
step_distance: .002413
nozzle_diameter: 0.300
filament_diameter: 1.750
heater_pin: !PA20
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC15
control: pid
pid_Kp: 36.156
pid_Ki: 2.363
pid_Kd: 138.298
min_temp: 0
max_temp: 250

[tmc2660 extruder]
cs_pin: PC17
spi_bus: usart1
microsteps: 16
run_current: 1.000
sense_resistor: 0.051

#[extruder1]
#step_pin: PD4
#dir_pin: PD9
#enable_pin: !PC6
#heater_pin: !PA16
#sensor_pin: PC12
#...
#[tmc2660 extruder1]
#cs_pin: PC25
#spi_bus: usart1
#sense_resistor: 0.051
#...

[heater_bed]
heater_pin: !PA19
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC13
control: watermark
min_temp: 0
max_temp: 130

[fan]
pin: PC23  # FAN0

#[heater_fan nozzle_cooling_fan]
#pin: PC26  # FAN1

#[heater_fan board_cooling_fan]
#pin: PA0  # FAN2

[mcu]
serial: /dev/serial/by-id/usb-Klipper_sam4e8e_00313753364B37373035303537303230-if00

[printer]
kinematics: corexy
#   This option must be "corexy" for corexy printers.
max_velocity: 300
max_accel: 3000
max_z_velocity: 30
max_z_accel: 30

# Example bed_mesh config section
[bed_mesh]
mesh_min: 25,0
mesh_max: 290,290
probe_count: 4,4

[gcode_macro G29]
gcode:
    G28
    BED_MESH_CALIBRATE`

Henke1983, 2020-04-12:
Have you tried to remove the ! in front of the dir_pin?

[stepper_x]
step_pin: PD6
dir_pin: !PD11 <-- changing direction

catchra, 2020-04-12:
@Henke1983  @mkuf my print head is moving in the correct direction the problem is if i move the x axis the y axis will move instead.

I also realize that forgot to say what hardware i am using.
arduino mega with ramps 1.4 with 5 pololu a4988 drivers the hotend is a e3d titan aero
the printer is a hypercube evolution with a dual z.

this is my printer.cfg

```
[stepper_x]
step_pin: ar54
dir_pin: ar55
enable_pin: !ar38
step_distance: .0125
endstop_pin: ^ar2
endstop_pin: ^ar3
position_endstop: 0
position_max: 300
homing_speed: 50

[stepper_y]
step_pin: ar60
dir_pin: !ar61
enable_pin: !ar56
step_distance: .0125
endstop_pin: ^ar15
endstop_pin: ^ar14
position_endstop: 0
position_max: 300
homing_speed: 50

[stepper_z]
step_pin: ar46
dir_pin: ar48
enable_pin: !ar62
step_distance: .0025
endstop_pin: ^!ar18
endstop_pin: ^!ar19
position_endstop: -0.0
#endstop_pin: probe:z_virtual_endstop
position_max: 300
position_min: -1

[stepper_z1]
step_pin: ar36
dir_pin: ar34
enable_pin: !ar30
step_distance: .0025
position_max: 200

[extruder]
step_pin: ar26
dir_pin: ar28
enable_pin: !ar24
step_distance: .002
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: ar10
sensor_type: PT100 INA826
sensor_pin: analog13
control: pid
pid_Kp: 22.2
pid_Ki: 1.08
pid_Kd: 114
min_temp: 10
max_temp: 250

[heater_bed]
heater_pin: ar8
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog14
control: watermark
min_temp: 10
max_temp: 130

[fan]
pin: ar9

[mcu]
serial: /dev/serial/by-id/usb-Arduino__www.arduino.cc__Arduino_Mega_2560_64938323231351B08231-if00
pin_map: arduino

[printer]
kinematics: corexy
#   This option must be "corexy" for corexy printers.
max_velocity: 300
max_accel: 3000
max_z_velocity: 25
max_z_accel: 30

# Common EXP1 / EXP2 (display) pins
[board_pins]
aliases:
    # Common EXP1 header found on many "all-in-one" ramps clones
    EXP1_1=ar37, EXP1_3=ar17, EXP1_5=ar23, EXP1_7=ar27, EXP1_9=<GND>,
    EXP1_2=ar35, EXP1_4=ar16, EXP1_6=ar25, EXP1_8=ar29, EXP1_10=<5V>,
    # EXP2 header
    EXP2_1=ar50, EXP2_3=ar31, EXP2_5=ar33, EXP2_7=ar49, EXP2_9=<GND>,
    EXP2_2=ar52, EXP2_4=ar53, EXP2_6=ar51, EXP2_8=ar41, EXP2_10=<RST>
    # Pins EXP2_1, EXP2_6, EXP2_2 are also MISO, MOSI, SCK of bus "spi"
    # Note, some boards wire: EXP2_8=<RST>, EXP2_10=ar41

# See the sample-lcd.cfg file for definitions of common LCD displays.

[display]
lcd_type: st7920
cs_pin: EXP1_4
sclk_pin: EXP1_5
sid_pin: EXP1_3
encoder_pins: ^EXP2_3, ^EXP2_5
click_pin: ^!EXP1_2
kill_pin: ^!EXP2_8

[output_pin beeper]
pin: EXP1_1
```

mkuf, 2020-04-12:
> 
> @Henke1983 @mkuf my print head is moving in the correct direction the problem is if i move the x axis the y axis will move instead.
> 

That is because only _one_ of your Motors is turning in the wrong direction. 
Invert the dir_pin of either stepper_x **or** stepper_y

catchra, 2020-04-12:
@mkuf thx for your response but that didn't work, I tried all combinations and it either moved in the wrong direction or crashed into the side of the printer.

mkuf, 2020-04-12:
It moved in the wrong direction, but the correct axis?

catchra, 2020-04-12:
no It moved in the wrong direction and wrong axis

mkuf, 2020-04-12:
This is strange. 
Have a look at https://corexy.com/theory.html

This boils down to:
Both Motors physically turn the same direction -> X Move
Motors physically turn in different directions -> Y-Move

If one Motor does not move in the direction it is supposed to, _the axis will be swapped_.

You mentioned that you're using a hypercube evolution. This one? https://www.thingiverse.com/thing:2254103
If yes, than it is a bog standard CoreXY impelementation and should not require any special Code to work with Klipper

I'm not familliar with Marlin, the only Info i could find about CoreYX in Marlin was 'CoreXY but inverted', which Interpret as Motors in the front vs. Motors in the Back.

theopensourcerer, 2020-04-12:
> If yes, than it is a bog standard CoreXY implementation and should not require any special Code to work with Klipper

FWIW I've been using klipper on my HEVO for ages with no problems at all.

locaJulia, 2020-04-14:
That is because not only does the DIR matter, but also the connection itself. Try also flipping the stepper connectors

aotd1, 2020-04-16:
> no It moved in the wrong direction and wrong axis
**coreYX** in Marlin is only **coreXY** with inverted direction pins or/and swapped stepper connections.

closed 2020-04-18 by klipper-gitissuebot