Klipper3d/klipper #2714: filament sensor regression

aschor, 2020-04-12:
Hi, before updating to latest master, (and to  dmbutyugin's smoothing branch), I had no issue what so ever with my filament sensor.

but NOW I have : basically, it does not behave as it should, not at all : 
- I start the printer, I heat the nozzle, then launch a macro to remove my filament => the event "runout" fires in the log, BUT IT SOULD NOT, since I'm not printing (I've not even homed the printer)

- when I insert my new filament, nothing happens, the "insert" section of runout configuration does not run, BUT IT SHOULD, because I'm not printing.

If I home, then do the same "tests" :  
- when I remove my filament, the printer triggers a "pause" and moves the head to pause position, but it shouldn't because I'm not printing.
- insert gcode is still not triggered.

is 3ed081f2b500359d5aacff7ec262c8aa6b6691af the culprit ?

attached a "klippy.log" where you can see python errors when the runout tries to pause a non printing printer ... 
[klippy.log](https://github.com/KevinOConnor/klipper/files/4466375/klippy.log)

relevant configuration :

[filament_switch_sensor filament_sensor]
pause_on_runout: True
runout_gcode:
        M118 Filament Runout Detected
        M600 X100 Y-3 Z10
insert_gcode:
        M118 Filament Load Detected
        M118 lance un LOAD_FILAMENT manuellement
        G4 S10
        PRE_LOAD_FILAMENT
event_delay: 3.0
switch_pin: !P1.24  #skr 1.3 z+ endstop

[gcode_macro PRE_LOAD_FILAMENT]
gcode:
    M117 PRE Loading Filament...BMG
    M83
    G1 E20 F1000
    #G1 E40 F100
    M82
    M400
    M117 Load Complete

ld3300, 2020-04-13:
My printer is busy, so I can't test with the latest to confirm the issues, but you can add to your macro where you remove the filament at the beginning:
`SET_FILAMENT_SENSOR SENSOR=filament_sensor ENABLE=0`
Then set the filament sensor Enable to 1 when you are ready for it.

aschor, 2020-04-13:
Thanks. But if I do that, the "insert" gcode will not launch either when I insert filament.

ld3300, 2020-04-13:
This is interesting because I am having the opposite response.  Without anything printing my filament sensor doesn't acknowledge removed filament, but does acknowledge inserted filament.  I also have a pullup on my pin because of the way the switch is wired. Give me a few to try to look through the code and see what's up.

ld3300, 2020-04-13:
I looked over your original post again, are you on the Klipper master, or one of dmbutyugin's branches? If so which one?  There are several that have smoothing in the name.

aschor, 2020-04-13:
Hi @ld3300 thanks for looking at this.

I'm on scurve-smoothing branch (and dmbutyugin confirmed he didn't touch anything. on the contrary on master there were several commits).

Maybe, for some reason, klipper "thinks" that I'm printing (even un-homed) ? Maybe it has to do with octoprint querying non stop what it's doing. 

But only thing I'm sure : it changed nothing but klipper update. I was before more or less on  commit  7f812ea642c22cf510e5c498d9ed2f4470518d91

dmbutyugin, 2020-04-13:
@aschor yes, I did not change a thing regarding the filament sensor or anything related to print state tracking in Klipper. However, it can be easier for all if you tried to reproduce the same issue with the mainline Klipper code, and attach the log from that attempt. I think that would take fairly little work on your end, as it is very easy to reproduce and you don't need to print anything.

aschor, 2020-04-14:
Hi again. So I did some tests and the results are not what I expected :

1)  sudo  service klipper stop ; git fetch ; git checkout master ; echo > /tmp/klippy.log ; sudo  service klipper start ;
then I heated the toolhead, unload macro .... reinsert the filament and ..... and bam ! the "insert" code of the config ran ! So filament sensor worked as intended on master !

So 2nd test on master : I Home All, move the head a few times, then again "unload macro", followed by filament re-insertion .......... and again it worked as expected

here is the log : 
[klippy.log.ok.txt](https://github.com/KevinOConnor/klipper/files/4473699/klippy.log.ok.txt)

2) sudo  service klipper stop ; git fetch ; git checkout scurve-smoothing  ; echo > /tmp/klippy.log ; sudo  service klipper start ;

first difference : as soon as klipper has finished Building C code module c_helper.so and outputing the current config, it starts spitting non stop lines with supposed gcode (Stats 1501.6: gcodein=772 mcu: ...etc), where master does not do that (until octoprint is connected)

then I heat toolhead -> unload macro -> reinsert the filament and ........... nothing, the "insert" section of filament sensor is not fired.

here is the log too
[klippy.log.ko.txt](https://github.com/KevinOConnor/klipper/files/4473730/klippy.log.ko.txt)

so fore 'some' reason, it seems it has something to do with the scurve-smoothing branch somehow, don't you agree @dmbutyugin  ? Do you want me to do other tests ?

dmbutyugin, 2020-04-14:
@aschor thanks for double-checking. I will take a look at it then, maybe there is some unexpected interference with scruve code in the end. Could you also do me a favour and check `scurve-c-combine-smoothpa` branch, does it have this unexpected behaviour of the filament sensor too? I was planning to install the filament sensor, but did not do it yet, so I will have to think of something on my end.

aschor, 2020-04-14:
ho @dmbutyugin 

just tested branch scurve-c-combine-smoothpa (on commit 58c5ba2264622ed220bf8a934e25c59342d6f73d) and it works

dmbutyugin, 2020-04-14:
@aschor thanks for reporting, `scurve-smoothing` branch should be fixed now. In a sense that `insert_gcode` macro is executed now. However, note that

>  - I start the printer, I heat the nozzle, then launch a macro to remove my filament => the event "runout" fires in the log, BUT IT SOULD NOT, since I'm not printing (I've not even homed the printer)

still happens, on master too (and you can see it in your klippy.log.ok.txt):
> Filament Sensor filament_sensor: runout event detected, Time 596.02
> action:paused

The best explanation I can offer you is when you execute `UNLOAD_FILAMENT` macro, your printer is, in fact, printing. As the extruder is pulling out the filament, still executing that macro, the filament sensor triggers when filament is pulled from it. This causes 'filament runout' event, if that makes sense. It is possible, though I did not check it, that before 3ed081f2b500359d5aacff7ec262c8aa6b6691af this was not happening and Klipper did not consider executing a macro as printing for the filament runout. I think this issue (if an issue at all) should be addressed separately, since this happens on pristine Klipper codebase as well.

aschor, 2020-04-15:
great ! thanks ! thanks for the explanation too :) (I'll retry as soon as I can)

So @ld3300 , what do we do ? Do I leave this issue open ? (I think so, since the printer should no try to pause and move if it is not homed and throw python errors in logs)

Arksine, 2020-04-16:
It seems that most issues have been cleared up, outside of a runout event firing when an UNLOAD_FILAMENT macro is executed.  This isn't an issue per se, if your filament sensor's switch triggered while the extruder gears are still running then code correctly interprets that as a runout.   The solution is to temporarily disable the filament sensor in your UNLOAD macro:
```
[gcode_macro UNLOAD_FILAMENT]
gcode:
  SET_FILAMENT_SENSOR SENSOR=filament_sensor ENABLE=0
  M117 Unloading Filament...
  M83
  G1 E10 F200
  G1 E-1.0 F1000
  G1 E1.5 F1000
  G1 E-1.5 F1000
  G1 E2.0 F1000
  G1 E-100 F3000
  G1 E-100 F3000
  G1 E-100 F3000
  G1 E-100 F3000
  G1 E-100 F3000
  G1 E-80 F3000
  M82
  M400
  M117 Remove Filament Now!
  SET_FILAMENT_SENSOR SENSOR=filament_sensor ENABLE=1
```

dmbutyugin, 2020-04-16:
This seems like the most reasonable solution to me. Klipper itself has insufficient knowledge to disable a filament sensor during macro execution - it might not always be correct, unless explicitly disabled by the user.

aschor, 2020-04-17:
Hi @Arksine @dmbutyugin , yes but ... no. And maybe that's where the issue is :

in fact, this does not work. because the filament sensor is several centimeters above the wheels (bondtech, or anything : all runout sensors are before the extruder wheels). Running the macro DOES NOT trigger the sensor nor the event : the extruder can NOT push filament beyond it's wheels hence can not push beyond filament sensor ; the event is triggered while manually removing the filament, and while there is NO motor turning, not even the extruder one.

So basically, disabling and enabling the sensor would both happen before what happens in the logs. Klipper sould know that if no motor is running, then the runout event should not be triggered (but the insert event should)

Arksine, 2020-04-17:
Klipper will not trigger the runout event if no motors are turning.  I realize that it is not possible for the gears to push the filament past the filament sensor switch, but it is certainly possible for a person to pull it past the switch while the gears are still running.

Update your macro as I suggested and try to reproduce the issue.   If its still there please post the log.

aschor, 2020-05-02:
thanks, closing this

closed 2020-05-02 by aschor