Klipper3d/klipper #2715: bltouch - zlift is not high enough when creating bed mesh, BLT pin is hiting the bed while the pin is falling down

Mak3rde, 2020-04-12:
i am looking for a way to increase the z lift when probing the heatbed, 
when creating the bedmesh, i have set the probing to probe each point twice.

The problem is that the print head is not lifted far enough after the first probe.  
As a result, during the pin down maneuver, the pin doesn't have enough space to fall down freely, and is hitting the bed.

Fortunately, the whole thing doesn't seem to be that bad, I get a clean bed mesh, 
but still it would be nice if I could lift z just a little higher.

I would just need the information where in the source code I can change the value

thx a lot for you work and this awesome firmware

Mak3rde, 2020-04-12:
no infos in the log because no error, but the bot wants a log file so here it is

[klippy.log](https://github.com/KevinOConnor/klipper/files/4466865/klippy.log)

BlackStump, 2020-04-12:
Just as you have stated in your log "there_is_noting_to_see_here" 
RTFM
https://www.klipper3d.org/

jakep82, 2020-04-12:
https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L578

Mak3rde, 2020-04-13:
@BlackStump  as i have described in the text in the first post, there is no error as such and no error message, so the log is completely superfluous, and i have read the doc 

@jakep82 
thanks for trying to help but as i described the problem is the second lift move after first probe on the same spot.

the

> #horizontal_move_z: 5

#   The height (in mm) that the head should be commanded to move to
#   just prior to starting a probe operation. The default is `5.`

seems to be only used when changing the cooridnate, but unfortunately does not seem to be used for the second sample at the same place.

sequence of events:

1. head  goes up to **horizontal_move_z: 5** 
2. head is movin to next probe spot, blt pin is dropped 
3. first probe will happen without  problems
4. head is lifted, but not high as before its not the defined **horizontal_move_z: 5**
5. blt pin is dropped, but due to lack of space the pin hits the bed
6. second probe will happen
7. head  goes up to **horizontal_move_z: 5** 
8. head is movin to next probe spot, blt pin is dropped 

and so on its just the second lift that seems to be not high enough

Mak3rde, 2020-04-13:
@BlackStump 
looks like it wouldn't have been enough to read the doc otherwise I wouldn't have asked, strange way of addressing someone asking for help, there was nothing in the log file that would have helped so it was superfluous, otherwise the onfo would have been attached...

@jakep82 
thanks a lot, was probably too late yesterday i just didn't find this, happy easter

I think this can be closed.

closed 2020-04-13 by Mak3rde