Klipper3d/klipper #2716: error from start

norwestie, 2020-04-12:
My Octopi is online in my browser either by octopi.local or the LAN address. But when I try to SSH via Terminal, this is the error log:

Last login: Sun Apr 12 11:46:44 on ttys000

The default interactive shell is now zsh.
To update your account to use zsh, please run `chsh -s /bin/zsh`.
For more details, please visit https://support.apple.com/kb/HT208050.
JEFFs-iMac:~ jeffxxxx$ ssh pi@192.168.170.146
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
@    WARNING: REMOTE HOST IDENTIFICATION HAS CHANGED!     @
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
IT IS POSSIBLE THAT SOMEONE IS DOING SOMETHING NASTY!
Someone could be eavesdropping on you right now (man-in-the-middle attack)!
It is also possible that a host key has just been changed.
The fingerprint for the ECDSA key sent by the remote host is
SHA256:inbq+zddyDJ7itTbLkgmYAMQsa6QH50pmSfcnZCA4.
Please contact your system administrator.
Add correct host key in /Users/jeffxxxx$.ssh/known_hosts to get rid of this message.
Offending ECDSA key in /Users/jeffhood/.ssh/known_hosts:1
ECDSA host key for 192.168.170.146 has changed and you have requested strict checking.
Host key verification failed.

> JEFFs-iMac:~ jeffxxxx$ git clone https://github.com/KevinOConnor/klipper
> xcrun: error: invalid active developer path (/Library/Developer/CommandLineTools), missing xcrun at: /Library/Developer/CommandLineTools/usr/bin/xcrun

JEFFs-iMac:~ jeffxxxx$ 

What am I doing wrong?

theopensourcerer, 2020-04-12:
remove the host key from your known_hosts file:

> Offending ECDSA key in /Users/jeffhood/.ssh/known_hosts:1

(You probably ssh'd to this IP address some time ago but it was a different RPi or you have changed the sd card and/or re-installed raspbian or whatever...)

And then it looks like you have a problem on your mac somewhere...

> xcrun: error: invalid active developer path (/Library/Developer/CommandLineTools), missing xcrun at: /Library/Developer/CommandLineTools/usr/bin/xcrun

I'm not a mac user so have no idea what this error means. Google would be your friend... e.g.

https://www.google.com/search?client=ubuntu&channel=fs&q=xcrun%3A+error%3A+invalid+active+developer+path&ie=utf-8&oe=utf-8

benedekt, 2020-04-12:
If you want to clone Klipper to your iMac, you can install Git through the Apple way:
`xcode-select --install`

But I don't think that you want to do that. You shall install Klipper on the pi.

Addressing your the SSH problem, I would suggest you to run:
`rm .ssh/known_hosts`
The IP address probably changed of your Pi, or the same hostname was used before to SSH somewhere with your mac, and this is a security measure to prevent fraud (MITM attacks).
Deleting known hosts is technically the bad way, but since you don't use SSH for anything else, you can take the easy route.

theopensourcerer, 2020-04-13:
> Addressing your the SSH problem, I would suggest you to run:
rm .ssh/known_hosts

The OP (probably) only needs to delete line 1 from known_hosts.

norwestie, 2020-04-13:
Thank you all for the replies. I haven't tried a redo on the Mac but instead ran it with PuTty on my PC. Worked great!

closed 2020-04-13 by norwestie