Klipper3d/klipper #2717: G28 problem

Kowy54, 2020-04-12:
When i send G28, i get error "Must home axis first: 0.000 0.000 5.000 [0.000]"
I dont know whats wrong.
[klippy.log](https://github.com/KevinOConnor/klipper/files/4466982/klippy.log)

jakep82, 2020-04-12:
Your homing_override is wrong. Try the override below

```
[homing_override]
set_position_z: 0
gcode: 
	G90
	G1 Z5 F500
	G28 X
	G28 Y
	G1 X85 Y85 F2000 ; Change the X and Y coordinates to the center of your print bed
	G28 Z
	G1 Z10

Kowy54, 2020-04-13:
My homing_override is same as you wrote. But nothing :(

jakep82, 2020-04-13:
No it isn't the same.  The set_postion_z should not be nested inside the gcode section.  From your log:

```
[homing_override]
gcode = 
	G90
	G1 Z5 F500
	G28 X
	G28 Y
	G1 X85 Y85 F2000 ; Change the X and Y coordinates to the center of your print bed
	G28 Z
	G1 Z10
	set_position_z: 0.0

Kowy54, 2020-04-13:
Still same error

closed 2020-04-13 by Kowy54