Klipper3d/klipper #2718: First layer tilting randomly on delta kinematics, but manual G-CODE works

benedekt, 2020-04-12:
I have an interesting issue on my delta printer. 

The problem:

- **Manual** programming of the movement works, within the calibrated accuracy. Since I use the full-manual method for calibrating my delta, I move the effector with direct G-CODE a lot. I confirm, that movements are perfectly OK, even on the extreme edges of the printbed.

- **Printing** starts too high (around +2mm), then bottoms out at some other places of the print. Also the "lowspots" are pretty random - if I start the same print right after the previous code was stopped, and I load a new design, the lowspots will probably move, and the highspot comes with the starting line. The perimetes of the lowspots are not deterministic (ie. not concentrated around the center, or any of the towers.)

What I have tried so far:
- **Repeatability** testing: I confim that all endstops are working, all belts are OK, error near a tower or on the center is smaller than 0.05mm.
- Different **slicer**: I mainly use simplify3d, but now I downloaded Cura. It does the same thing.
- Manual tracing: after some calculations and tinkering I traced the brimlines with manual movements without extrusion. It was perfect, but when I tried printing from the file which should draw the  same brim, it raised again.
- Borax silicate glass: removed the aluminium **heated bed**, which I would not trust when it comes to flatness. It's the same with the glass.

When I do the leveling, I check it with a feeler gauge set, which goes down to 0.05mm, so I am pretty confindent that my calibration is ok.

My printer is an Anycubic kossel linear plus, retrofitted with TMC2100 drivers. It was calibrated with a manual procedure: height adjustment at the endstops, radius (concave-convex) calibration, then printer height correction. **This an old and well worn system, so my issue might be mechanical.**

The exact same models were printed successfully on this printer before, using Marlin FW.

Might this be a bug in Klipper? All comments and suggestions are more than welcome.

I have attached my full log of this boot with many unsuccessful tries (just for system parameters), and my configuration for this delta crap.
I have also attached the initial section of a GCODE, which produces the error on my machine.
[klippy.log](https://github.com/KevinOConnor/klipper/files/4467127/klippy.log)
[printer.txt](https://github.com/KevinOConnor/klipper/files/4467129/printer.txt)
[badcode.txt](https://github.com/KevinOConnor/klipper/files/4467142/badcode.txt)

KevinOConnor, 2020-04-15:
Have you run the calibration procedure described at: https://www.klipper3d.org/Delta_Calibrate.html ?  If not, I recommend using that calibration procedure.  If you continue to have issues, please attach the Klipper log file from the session where you ran `DELTA_CALIBRATE METHOD=manual`.

-Kevin

benedekt, 2020-04-15:
I've tried the calibration procedure, and the result was absurd: it calculates the value of the radius to be around 146mm and the arm length to 271mm, which are obviously and measurably wrong.

`#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [printer]
#*# delta_radius = 146.089539
#*#
#*# [stepper_a]
#*# angle = 210.019029
#*# arm_length = 271.500000
#*# position_endstop = 311.963432
#*#
#*# [stepper_b]
#*# angle = 336.731682
#*# arm_length = 271.500000
#*# position_endstop = 309.843588
#*#
#*# [stepper_c]
#*# angle = 90.000000
#*# arm_length = 271.500000
#*# position_endstop = 317.351725`

And those `position_endstop` things look pretty interesting too. My towers cannot be this much different. The spacings were measured with gauge blocks, and as a rough guess they should be within 1mm (worst case), 0.1mm (probably) compared to each other. 

I will capture a full log tomorrow. 
Thanks for your time and effort.

benedekt, 2020-04-16:
Here is the log:
[autocalibration.klippy.log](https://github.com/KevinOConnor/klipper/files/4485470/autocalibration.klippy.log)

KevinOConnor, 2020-04-19:
I'd guess there is either a mechanical error (eg, too low current to a driver, loose grub screw, too high friction on a tower, bad stepper, or bad stepper driver) or a significant error in the config (eg, arm length, step_distance, etc.).

If you run the DELTA_CALIBRATE command again and continually get the same (highly tilted) results then you could just try with those parameters.  If you continually get different results then it indicates something is mechanically loose (or loosing steps).

-Kevin

closed 2020-05-11 by KevinOConnor