Klipper3d/klipper #2719: Recv: // Lost communication with MCU 'mcu'

swordfish45, 2020-04-12:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4467144/klippy.log)

[image]

[4-12-20_octoprint_Log.txt](https://github.com/KevinOConnor/klipper/files/4467161/4-12-20_Log.txt)

During print, lost communication with MCU. Octoprint running on raspberry pi 4. Noticed 2 webcam streams on the pi are no longer available. Octoprint UI continues to work as expected.

ld3300, 2020-04-13:
Last time I had a problem like this it was a failing power supply that was running my pi.  It was allowing the voltage to dip low enough that it couldn't maintain connection on the USB ports.  Try running a print without the cameras connected, and if that doesn't work try a different power supply as well.

swordfish45, 2020-04-13:
I have the Canakit power supply that came with the Pi4

KevinOConnor, 2020-04-15:
The best information we have on this is in the FAQ: https://www.klipper3d.org/FAQ.html#i-keep-getting-random-lost-communication-with-mcu-errors

-Kevin

closed 2020-04-24 by KevinOConnor