Klipper3d/klipper #2721: USB disconnect does not shut down heater

Eeshwar-Krishnan, 2020-04-12:
This is something I have noticed due to several recent failures with my 3d printer. A few days ago I caught my 3d printer frozen with smoke pouring out of the nozzle and had to turn it off. After that it happened again, smoke with me luckily catching it on camera and stopping it before anything bad happened. Finally, today, I went to change the filament and set the temp of the nozzle to 200. Around 195 I accidentally pulled the usb while grabbing a pair of flush cutters, and noticed it seconds afterwards. It took about 15-20 seconds to restart klipper and for the lcd to come back online, and when it did the temperature of the printer read 215.

I have tuned my PID previously and it has a nominal overshoot of 1-2 degrees, nowhere near the 15 degrees it did when it dc'd. Afterwards I kind of connected the dots and realised that all the times the printer had started smoking was because of the usb cord getting snagged for one reason or another (all accidents). 

The main issue at heart is that a USB disconnect does not seem to power off the heater. I do not have a klipper log file because I only realised it now and forgot to grab the log file at the time. I am hesitant to recreate the issue for obvious safety concerns, but if absolutely needed I can. I was planning on maybe testing it by splicing a 12v fan to the output and intentionally disconnecting the pi, but again I am hesitant to test that for safety reasons. 

Intended behaviour would be, of course, the heater shutting off on a USB disconnect (perhaps through some sort of heartbeat system or similar, I do not know). It is extremely alarming to me that it does not do so, and I would really like to know if this is a bug or intended behaviour.

The board I was using was an MKS GEN L 1.0, with a good USB cable (tested, over a 5 hour upload interval it dropped no signals). I remember there being a related issue #1101 to this issue, and similar to that I have had USB disconnects that result in the heater shutting off. The next time it happens (I hope it doesn't) I will be sure to pull the log file and pass it on.

KevinOConnor, 2020-04-14:
Unfortunately, without a log there is very little we can do.  I recommend you run the steps at https://www.klipper3d.org/Config_checks.html .  The "verify M112" section is specifically designed to verify that the micro-controller shutdown logic will perform correctly.

It is also possible that the board has an electrical fault that causes it to enable a mosfets when the usb power is not engaged - if so, I recommend testing for that and to not use the board in that case.

-Kevin

Eeshwar-Krishnan, 2020-04-19:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4499969/klippy.log)

It happened again. Here is the log file from the most recent time

> Transition to shutdown state: MCU 'mcu' spontaneous restart
Dumping gcode input 0 blocks
gcode state: absolute_coord=True absolute_extrude=True base_position=[0.0, 0.0, 0.0, 6009.37982] last_position=[51.36, 169.011, 3.5, 6221.25958] homing_position=[0.0, 0.0, 0.0, 0.0] speed_factor=0.0125 extrude_factor=1.0 speed=95.625
Virtual sdcard (1956684): '\nG1 X51.529 Y186.743 E210.06473\nG1 X51.828 Y187.456 E210.10191\nG1 X52.213 Y188.127 E210.13909\nG1 X52.678 Y188.744 E210.17627\nG1 X53.215 Y189.299 E210.21345\nG1 X53.818 Y189.784 E210.25063\nG1 X54.476 Y190.189 E210.28780\nG1 X55.179 Y190.511 E210.32498\nG1 X55.916 Y190.742 E210.36216\nG1 X56.677 Y190.881 E210.39934\nG1 X57.449 Y190.924 E210.43652\nG1 X58.220 Y190.871 E210.47370\nG1 X58.979 Y190.723 E210.51088\nG1 X59.713 Y190.482 E210.54806\nG1 X60.412 Y190.152 E210.58524\nG1 X61.065 Y189.738 E210.62242\nG1 X61.661 Y189.246 E210.65960\nG1 X62.192 Y188.684 E210.69678\nG1 X62.649 Y188.060 E210.73396\nG1 X63.025 Y187.385 E210.77113\nG1 X63.315 Y186.668 E210.80832\nG1 X63.513 Y185.921 E210.84549\nG1 X63.618 Y185.155 E210.88267\nG1 X63.627 Y184.382 E210.91985\nG1 X63.540 Y183.614 E210.95703\nG1 X63.358 Y182.863 E210.99421\nG1 X63.085 Y182.140 E211.03139\nG1 X62.724 Y181.456 E211.06857\nG1 X62.281 Y180.823 E211.10575\nG1 X61.769 Y180.255 E211.14253\nG1 X59.058 Y177.537 E211.32713\nG1 X56.422 Y174.759 E211.51134\nG1 X53.855 Y171.916 E211.69555\n'
Upcoming (1957708): 'G1 X51.360 Y169.011 E211.87976\nG1 X50.412 Y167.852 E211.95177\nG1 X48.937 Y166.045 E212.06396\nG1 X46.588 Y163.020 E212.24817\nG1 X'
Exiting SD card print (position 1957739)
MCU 'mcu' shutdown: Command request
clocksync state: mcu_freq=16000000 last_clock=149248875862 clock_est=(1390944.507 148791881327 16000270.122) min_half_rtt=0.000421 min_rtt_time=1390687.328 time_avg=1390944.506(844.023) clock_avg=148791881327.948(13504596992.663) pred_variance=1166935.192
Dumping serial stats: bytes_write=4254372 bytes_read=640174 bytes_retransmit=9 bytes_invalid=0 send_seq=78513 receive_seq=78513 retransmit_seq=2 srtt=0.004 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0
Dumping send queue 100 messages

KevinOConnor, 2020-04-21:
It looks like you are using external code (dwc2).  You'll need to reproduce the problem on the pristine code from this repo and provide that log, or report the issue to the dwc2 developers.

That said, I'd encourage you to confirm the steps described in Config_checks.md and stop using the hardware until you can identify the problem.

-Kevin

closed 2020-05-11 by KevinOConnor

Eeshwar-Krishnan, 2021-07-27:
For anyone that is finding this in the future, I wanted to write that I redid my cable management and updated klipper, and as of today it doesn't look like its dc'ing anymore (I haven't run any prints > 4 hours yet, will report back when i do). If this doesn't fix the issue, switching over to marlin seemed to stop the disconnects at the time, but I still do not know what exactly caused the issue.