Klipper3d/klipper #2722: Distorted print paths

gabork, 2020-04-13:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4467434/klippy.log)

Hi!

I have built a custom Core-XY printer with 3 Z axes. XY and extruder driven through a Mega+RAMPS board with TMC2208 drivers, and the Z driven though a Uno + CNC Shield with A4988 drivers.

I'm experiencing a very weird issue where the printer seems to deliberately misprint the GCode. I say deliberately, because the distortion of the toolhead paths is very consistent and repeatable. Infill lines would often not reach the perimeters (but on other spots they do), concentric paths can have gaps or even cross over each other. When I restart a print, the exact same artifacts appear on the exact same spots. The machine otherwise seems to be quite precise, for example when printing a single perimeter with no infill. The prints come out oriented correctly.

I have a simple GCode that demonstrates the issues with concentric paths. It's all 100% concentric infill with a long straight section and two circular sections on each end: [tensioning_bar.gcode.zip](https://github.com/KevinOConnor/klipper/files/4467494/tensioning_bar.gcode.zip) On the first layer it both presents a crossing path that other wise would be concentric, and a gap in the inner loop, when I print it. This GCode omits `G28`, so I could iterate faster.

My current guess is that something goes wrong in the trajectory generation inside Klipper, probably due to misconfiguration on my part. Looking through my config file, I don't think I do anything unorthodox there, but I let you be the judge of that.

I checked a lot of things, because it's a custom designed machine, and I first suspected a mechanical issue. Here's what I investigated so far:

- Happens even with fresh Klipper from master. The attached log says the repo is dirty, because I used the `install-debian.sh`, and it did not have execute permission. Here's the output of `git diff`: [klipper.diff.txt](https://github.com/KevinOConnor/klipper/files/4467570/klipper.diff.txt)
- The GCode by itself seems to be fine. I tried both Cura and PrusaSlicer and inspected it with different GCode viewers.
- Happens regardless of host. I tried both with OctoPrint and Printrun/Pronsole. No difference.
- It's not affected by speed. The toolhead takes the same path with 100mm/s and slowed down to 40mm/s with `M220 S40`.
- It isn't skipped steps or the printer being sluggish. When the print strays from the desired path, it meets back with with the starting position at the seam. No layer shifts either.
- For these reasons, I don't think it's caused by the TMC2208 drivers, but it is also the first time I used them. They are cooled by an always on 5V fan and have the larger heatsinks. (This fan does not appear in the printer config, because it is directly attached to a 5V pin on the Mega)
- It isn't affected by position on the printbed, it prints exactly the same when offset in XY. So it isn't just a distorted spots.
- Not an adhesion problem, the lines don't lift along the curves. The bed is levelled with `Z_TILT_ADJUST` and the print sticks well at every spot along the paths.
- Happens on any height, not just the first layer.
- Not a flow rate issue. I'm printing 1.75mm PLA at 220°C with a 0.4mm nozzle and a direct drive volcano extruder. The brand of PLA doesn't matter either.
- The printhead isn't loose, it does not wobble, not dragged while printing. This was my first guess at a potential cause, but no matter how I test it, it seems solid.
- Tightening the belts has no effect on the issue. 

I haven't tried a different firmware yet, because it would take a lot of rework on the machine.

I'll try to draw something to illustrate the results I'm getting.

Thanks in advance for any clues on the issue!

gabork, 2020-04-13:
Here is an illustration of the issue, drawn over the previously attached GCode (I left the middle section out, it's just straight lines):
[image]

Almost as if the paths would have some sort of inertia, making curves sharper or something like that, but as I mentioned, it prints the same when printing slow.

When I wrote "When the print strays from the desired path, it meets back with with the starting position at the seam", I meant the error doesn't accumulate. On the image, it obviously doesn't meet up with the start point anymore. However, it will do concentric, consistently distorted loops as the print progresses. The layers above distort the exact same way as the first layer.

Also, I use a Raspberry 2B and a kinda slow SD card, because that's what I had on hand.

ld3300, 2020-04-13:
If I get some energy later I will try to adapt your house for my printer and see what happens. I'm recovering from Covid-19 and only get a few hours of energy a day still.

ld3300, 2020-04-13:
Sorry, my print bed isn't big enough to run the code.  If you would like to reslice it to fit in 230x230 and see if you get the same effect on your end then I could test it as well.

gabork, 2020-04-13:
I'm not sure if I understand. The GCode should fit well within 230x230.

Although it is a fair point, I will try to reslice it placed closer to the origin point. My slicer was set to 190x210 I think, but the part was centered on the bed, so it won't even reach that far.

gabork, 2020-04-13:
Interesting, Cura didn't generate `G21` into the GCode, so a viewer might interpret it in inches rather than in millimeter. But the coordinates are in fact in millimeter.

Here is the extent of the print according to the GCode header (in millimeter):
```
;MINX:87.508
;MINY:76.432
;MINZ:0.3
;MAXX:112.5
;MAXY:173.573
;MAXZ:14.9
```

So you could bring it close to the origin with `SET_GCODE_OFFSET X=-80 Y=-70`

ld3300, 2020-04-13:
I looked at it in Notepad++ and it had some X coordinates in the 250s

gabork, 2020-04-14:
That can't be right, it says the **MAX** X is 112.5, and the **MIN** X is only 87.508

Either way I resliced the model, now it doesn't reach beyond even 30x100: 
[tensioning_bar_v2.gcode.zip](https://github.com/KevinOConnor/klipper/files/4475162/tensioning_bar_v2.gcode.zip)

I also included `G21` in the start GCode (which tells the printer that the measurements are in milimeter), so there should be no confusion about the print size. I did not have the chance to print it yet, but I will try later today.

I did not expect anyone to print the GCode. I mostly attached it for illustration. But if anyone attempts it, thank you. My issue is already visible on the first layer, if present at all, so one doesn't even have to print the whole thing.

Also, I suspect very few (if any) other people have the same problem as me, as I couldn't find similar issue tickets. I must have stumbled upon an edge case, or made a rookie mistake in the config.

KevinOConnor, 2020-04-14:
FWIW, I haven't seen any reports like this.  My initial reaction is that it sounds like a mechanical issue.

One suggestion would be to try increasing the current on the tmc2208 drivers, do not enable a stealthchop_threshold (only use spreadcycle), and do not use a lower hold_current.

-Kevin

gabork, 2020-04-14:
Thanks for the tips! I tried the following config variations for the TMC2208 drivers, but nothing changed in the print result:
- Set `stealthchop_threshold` to 0. No change.
- Set the `run_current` and `hold_current` to the same value, ranging from 0.5 to 0.8. At 0.5 the steppers were losing steps, at 0.8 they had a faint whine. No other change.
- Set the `microsteps` in the range 4 to 32 (adjusting `step_distance` accordingly). No change.
- Set `interpolate` from True to False. The steppers became louder, but no other change.

To completely eliminate the possibility of the TMC drivers causing the problems, I swapped them out for A4988 drivers. When I move them with `FORCE_MOVE` they move fine, but when I try to home the machine the steppers don't move and make a loud, toneful buzz. Here is the log: 
[klippy_2020_04_14.log](https://github.com/KevinOConnor/klipper/files/4477998/klippy_2020_04_14.log)

That's all I could try today.

BlackStump, 2020-04-14:
I also suspected it is a mechanical build issue, make sure your drive belts are parallel to your axis.
You may find the pull of the steppers is skewing your axis.
I have a corexy and there is no problem with the tool paths, print dimensions are very good.
Try printing a 30mm circle and see if your actually get a circle.

gabork, 2020-04-15:
Looks like it isn't a stepper driver issue, it produces the same print with A4988s.

The A4988 drivers were buzzing, because for the jumper configuration I set, my `step_distance` resulted in a too fast homing speed. (In my defense, the RepRap wiki was a bit misleading on the jumpers: https://reprap.org/wiki/RAMPS_1.4#Stepper_Driver_Boards)

@KevinOConnor Thanks for confirming that others did not experience similar issues. I will continue debugging the mechanics. If you want, you can close the ticket as it currently seems unrelated to Klipper.

@BlackStump Thanks, I will test these next!

gabork, 2020-04-17:
It *was* loose belts. There was a broken part that let the belts slip and loosen after I tightened them. Sorry for bothering you with this.

closed 2020-04-17 by gabork

gabork, 2020-04-19:
For anyone who stumbles upon this ticket and has a similar problem, also check if your belts stretched. My belts stretched even though I barely tensioned them. It could be a bad quality belt, or it could be fatigue because of how my belts are routed. I replaced the belts to different type and I'll see how these hold up.

I didn't understand why my results didn't get better consistently after tightening the belts multiple times. One reason was, that a part deformed, and eventually broke, that let a belt slip. The other was that was that the belt was loosening and stretching on its own.