Klipper3d/klipper #2724: New Module : gcode_sound

mental405, 2020-04-13:
I have authored a [module](https://github.com/mental405/klipper/blob/master/klippy/extras/gcode_sound.py) that plays sound through the host OS's audio system using gcode commands. It also has a few hooks into different events which I plan to expand later to either include more events or have the event names parsed from the config section. Right now it is using the external pygame library to spawn a playback thread that runs externally to the klipper host service. The thread is called and the sound starts playing back. This is still a work in progress and I wanted to get some feedback regarding the external dependencies and threading before I spent much more time on it for public release or if I should just keep it all to myself.

In order to get pygame to work, there are some external dependencies that have to be downloaded via apt and pip. 

I could probably use something else instead of pygame but it is pretty well documented and does a good job of it.

I guess my question is, what are peoples thoughts on external libraries like pygame being pulled in for modules, and how do I need to sort out the dependency for the install scripts if I submit a PR for this module?

In regards to threading, is there any reason I should avoid spawning a thread in klipper if it keeps the main thread from being blocked? Especially in the case of sound playback to an external library where you might want the sound to continue to be heard (such as the case of a  "start print" tune or something)

KevinOConnor, 2020-04-15:
FWIW, I can only give some very high-level guidance.

I'd say we probably don't want to add a dependency like pygame to Klipper - the desired functionality is not core to the operation of a 3d printer and I'd guess it is not a good trade-off to have everyone install it for the few people that may use that feature.  Also, given the real-time nature of some parts of the host software, I'd be leery about running code from large external 3rd party libraries that may not be optimized for latency.

Unfortunately, threads don't work particularly well in Python due to the "global interpreter lock".  Unfortunately, a thread in Python can and will prevent other, higher priority, threads from running.  It depends on the work load, but if there is a conflict the typical solution is to use the multiprocess module or to run everything in an external program.

-Kevin

mental405, 2020-04-15:
I am going to look at possibly getting the code for playsound.py working. It doesn't use nearly the amount of dependencies as Pygame, but it does rely on gstreamer from a command line call for sound playback which means I might not be able to abort playback once it has started. 

I agree with you on including it as a main module as it is very much a niche thing. It would probably be better if I just maintained it in my own branch unless there is significant interest from the community to develop it further.

In terms of the parallel execution, I will look into changing it to use multiprocessing instead of the standard threading.

KevinOConnor, 2020-05-11:
Any further updates on this?

-Kevin

mental405, 2020-05-11:
I have not really had time to work on this beyond the initial implementation. Another fellow and I are working on finding a way to do it without all the dependencies of pygame. I will close the issue since my questions were answered and submit a PR when and if I get it workable. I mean.. it works now, it is just not a clean installation.

closed 2020-05-11 by mental405