Klipper3d/klipper #2726: Can't SET_SERVO to any of my servos

cyclonesurfer, 2020-04-14:
x-special/nautilus-clipboard
copy
file:///home/pat/klippy.log

x-special/nautilus-clipboard
copy
file:///home/pat/printer.cfg

I can't SET_SERVO any servo I set, please help.  It always returns 

Error:  The value 'TEST' is not valid for SERVO

gabork, 2020-04-14:
The names might be case sensitive. Try it with lowercase (it's lowercase in your printer.cfg):
```
SET_SERVO SERVO=test ANGLE=21
```

cyclonesurfer, 2020-04-14:
I tried lowercase, uppercase, and a mix of cases while trying to figure out how to make it work

ld3300, 2020-04-15:
I think you need to try with a fresh install to master branch with no DWC modifications to make sure it isn't interfering.  It's really hard to troubleshoot when your code is modified from the master branch.

Spacemarine2018, 2020-04-15:
I had a similar problem here: https://github.com/KevinOConnor/klipper/issues/2146

It was a problem with case sensitivity and DWC. Problem was gone when I switched to Octoprint.

I also reported it as an issue at DWC, but I think they didn't care.

cyclonesurfer, 2020-04-15:
I followed the directions on klipper.info for klipper+DWC2.

I'd really like to ditch octoprint if possible...

Spacemarine2018, 2020-04-15:
Then file a bug report for DWC2 and see if they respond.

cyclonesurfer, 2020-04-15:
This seems like a problem with klipper and not dwc2 to me.

Spacemarine2018, 2020-04-16:
Did you verify this by using octoprint instead of DWC?

KevinOConnor, 2020-04-24:
This github tracker is exclusively for use with the code in this github repo.  You'll need to either reproduce the error on the pristine code and attach the corresponding log, or you'll need to report the error to the dwc2 developers.

-Kevin

closed 2020-04-24 by KevinOConnor