Klipper3d/klipper #2727: Help with custome menus

dash3811, 2020-04-14:
I have copied a menu format from someone else that I wanted to see if I liked, but when I implemented it I kept getting an error. Any help would be appreciated. 

[klippy.log](https://github.com/KevinOConnor/klipper/files/4473001/klippy.log)

mcmatrix, 2020-04-14:
One thing I found.
```
[menu __calibration_bed_mesh_calibrate]
type = command
name = Generate bed mesh
width = 18
show_back = False
enter_gcode = 
	G28
	BED_MESH_CALIBRATE
	SAVE_CONFIG
```
only lists have `enter_gcode`. For `command` its just `gcode`

Xenomorphdelombre, 2020-04-14:
Anyone else got this? I updated and even tho I can use the printer, I can't use the printer screen/control it crashes the printer.... 
This is what I have right now

`[menu __main]
type: list
name: Main Menu
items:
    __tune
    __octoprint
    __sdcard
    __control
    __calibration
    __temp
    __filament
    __prepare
    __test

[menu __calibration]
type: list
name: Calibration
items:
    __calibration_home_all_axes
    __calibration_bed_mesh_calibrate
    __calibration_screws_calculate
	__calibration_probe_calibrate
    __calibration_probe_accuracy
	__calibration_save_config
    __general_firmware_restart

[menu __calibration_accept]
type: command
name: Accept
gcode: ACCEPT

[menu __calibration_abort]
type: command
name: Abort
gcode: ABORT
action: back

[menu __calibration_screws_calculate]
type: command
name: Screw tilt calculate
gcode:
    G28
    SCREWS_TILT_CALCULATE

[menu __calibration_probe_accuracy]
type: command
name: Test accuracy
gcode:
    G28
    G0 X164 Y134 Z10 F6000
    PROBE_ACCURACY 

[menu __calibration_save_config]
type: command
name: Save config
gcode: SAVE_CONFIG

[menu __general_firmware_restart]
type: command
name: Restart firmware
gcode: FIRMWARE_RESTART

[menu __calibration_home_all_axes]
type: command
name: Home XYZ
gcode: G28

[menu __calibration_probe_calibrate]
type: list
show_back: False
name: Adjust Z offset
enter_gcode:
    G28
    G0 Z10 F6000
    PROBE_CALIBRATE
items:
    __calibration__toolhead_zpos
    __calibration_probe_calibrate_testz_minus, __calibration_probe_calibrate_testz_plus 
    __calibration_probe_calibrate_testz_minus_minus, __calibration_probe_calibrate_testz_plus_plus
    __calibration_probe_calibrate_testz_minus_1, __calibration_probe_calibrate_testz_plus_1
    __calibration_probe_calibrate_testz_minus_point_1, __calibration_probe_calibrate_testz_plus_point_1
    __calibration_accept
    __calibration_save_config
    __calibration_abort

[menu __calibration__toolhead_zpos]
type: item
width: 16
name: "Z = {0:.3f}"
cursor: \x20
parameter: toolhead.zpos

[menu __calibration_probe_calibrate_testz_minus]
cursor: \x20
type: command
width: 7
name: "   -"
gcode: TESTZ Z=-

[menu __calibration_probe_calibrate_testz_plus]
cursor: \x20
type: command
name: "   +"
width: 7
gcode: TESTZ Z=+

[menu __calibration_probe_calibrate_testz_minus_minus]
cursor: \x20
type: command
name: "   --"
width: 7
gcode: TESTZ Z=--

[menu __calibration_probe_calibrate_testz_plus_plus]
cursor: \x20
type: command
name: "   ++"
width: 7
gcode: TESTZ Z=++

[menu __calibration_probe_calibrate_testz_minus_1]
cursor: \x20
type: command
name: "  -1.0"
width: 7
gcode: TESTZ Z=-1

[menu __calibration_probe_calibrate_testz_plus_1]
cursor: \x20
type: command
name: "  +1.0"
width: 7
gcode: TESTZ Z=+1

[menu __calibration_probe_calibrate_testz_minus_point_1]
cursor: \x20
type: command
name: "  -0.1"
width: 7
gcode: TESTZ Z=-0.1

[menu __calibration_probe_calibrate_testz_plus_point_1]
cursor: \x20
type: command
name: "  +0.1"
width: 7
gcode: TESTZ Z=+0.1

[menu __calibration_bed_mesh_calibrate]
type: command
name: Generate bed mesh
width: 18
show_back: False
gcode: 
	G28
	BED_MESH_CALIBRATE
	SAVE_CONFIG`

Coffee0297, 2020-04-15:
there's been a breaking update to klipper so no old copied menues work, see config_changes for info here about

dash3811, 2020-04-15:
All the menu items have the proper parameters and values and what did say “deck” or “card” have been changed so the breaking update isn’t the reason.

Xenomorphdelombre, 2020-04-15:
What would it be then? Is there something I missed? I ended up moving the Screw tilt adjust into a Macro button in Klipper instead since this is the only one I'm using

ld3300, 2020-04-15:
Can you upload a new klippy log.  The information you pasted above is quite different from what was in your original log.

dash3811, 2020-04-15:
I attached a recent klippy log when I enable a portion of the menu. I have also found that the printer will connect on initial power on however when I press the button/wheel to access the menu from the printer is when the crash occurs.

mcmatrix, 2020-04-15:
```ini
[menu __calibration_bed_mesh_calibrate]
type = command
name = Generate bed mesh
width = 18
show_back = False
enter_gcode = 
	G28
	BED_MESH_CALIBRATE
	SAVE_CONFIG
```
replace `enter_gcode = ` by `gcode =`

Xenomorphdelombre, 2020-04-15:
In the logs, with the changes you made for the menu to work withot DECK or CARD it still error out:

> Unhandled exception during connect
> Traceback (most recent call last):
>   File "/home/pi/klipper/klippy/klippy.py", line 134, in _connect
>     self._read_config()
>   File "/home/pi/klipper/klippy/klippy.py", line 127, in _read_config
>     self.try_load_module(config, section_config.get_name())
>   File "/home/pi/klipper/klippy/klippy.py", line 116, in try_load_module
>     self.objects[section] = init_func(config.getsection(section))
>   File "/home/pi/klipper/klippy/extras/display/__init__.py", line 9, in load_config
>     return display.load_config(config)
>   File "/home/pi/klipper/klippy/extras/display/display.py", line 201, in load_config
>     return PrinterLCD(config)
>   File "/home/pi/klipper/klippy/extras/display/display.py", line 90, in __init__
>     self.menu = menu.MenuManager(config, self.lcd_chip)
>   File "/home/pi/klipper/klippy/extras/display/menu.py", line 1115, in __init__
>     self.load_menuitems(config)
>   File "/home/pi/klipper/klippy/extras/display/menu.py", line 1573, in load_menuitems
>     item = cfg.getchoice('type', menu_items)(self, cfg, name)
>   File "/home/pi/klipper/klippy/extras/display/menu.py", line 480, in __init__
>     raise error("Missing or empty 'gcode' option")
> error: Missing or empty 'gcode' option
> Internal error during connect: Missing or empty 'gcode' option

dash3811, 2020-04-15:
That’s already been done please look at the recent klippy log and comments to see current issue.

dash3811, 2020-04-15:
That’s not the current error.

Xenomorphdelombre, 2020-04-15:
I have the same problem. After editing the custom menus in printer.cfg with what mcmatrix gave, I still have a crash whenever I click the Knob on the printer screen. Nothing happens then it become unresponsive. I need to powercycle the printer AND the Pi to be able to connect again. I ended up removing the menu and I'm printing now so I can't reproduce. 

What is the most recent error log after this?

dash3811, 2020-04-15:
I have fixed the issue. Under the [menu __main] it should look like this:

[menu __main]
type: list
name: Main Menu
items:
    __tune
    __octoprint
    __sdcard
    __control
    __calibration
    __temp
    __filament
    __prepare
#    __test

It is currently working on mine.

ld3300, 2020-04-17:
That is interesting.  Test was removed, but not included in the config changes file.

galosre, 2020-04-19:
@dash3811 Can you post an example of your working menu with calibration items?

ld3300, 2020-04-20:
I think the code he posted like 5 days ago works, the issue turned out to be the the [menu __main] had a __test section in it, which was part of the Klipper code until mid last month when it was removed, so the menu structure was looking for something that no longer existed.

dash3811, 2020-04-20:
ID3300 is correct. Just use the original code from the first post and remove the test menu section and all should work.

closed 2020-05-11 by KevinOConnor