Klipper3d/klipper #2728: Can't build controller FW

countAleister, 2020-04-14:
Hello everyone!
Don't judge strictly please, I'm very new to the Klipper and too much out of programming (more that 8y already).
Pi4 B. Following all the step-by-step instructions I'm to set it all up but got stuck at the controller FW step. After everything is configured, I'm facing following  in the end of the process:
`
pi@octopi:~/klipper $ make
  Building out/compile_time_request.o
Traceback (most recent call last):
  File "./scripts/buildcommands.py", line 609, in <module>
    main()
  File "./scripts/buildcommands.py", line 600, in main
    ctr_dispatch[cmd](req)
  File "./scripts/buildcommands.py", line 168, in decl_initial_pins
    + ','.join(self.initial_pins))
  File "./scripts/buildcommands.py", line 138, in decl_constant_str
    name, value = req.split(None, 2)[1:]
ValueError: need more than 1 value to unpack
make: *** [Makefile:91: out/compile_time_request.o] Error 1
`

Everything is freshly installed and updated. Any thoughts?
Thank you!

[klippy.log](https://github.com/KevinOConnor/klipper/files/4476395/klippy.log)

ld3300, 2020-04-15:
Hi @countAleister did you take one of the example config files that matches your machine and change any values that may be different for your machine, then save it as printer.cfg in your home directory?

countAleister, 2020-04-15:
@ld3300 Hello
Yes, sure, sorry, forgot to mention, I'm using to SKR 1.4 Turbo (LPC1769 120MHz) and custom CoreXY printer. 

Answering your previous question yes, i also tried it but it's the next step after the firmware is built as far as I can get so it doesn't matter. Anyway, I tried with the same result.

I bet you would like to see this params...
[image]

ld3300, 2020-04-15:
Just to clarify, it looks like it is the "make" command that is getting an error.  Is that correct?

countAleister, 2020-04-15:
> Just to clarify, it looks like it is the "make" command that is getting an error. Is that correct?

Exactly so, sir

KevinOConnor, 2020-04-15:
Errors like the above are usually indicative of a corrupted SD card image.  You can try running "make clean" to see if it clears the error - otherwise you may need to rewrite the sd card.

-Kevin

countAleister, 2020-04-17:
@KevinOConnor 
Dear Kevin, 
"make clean" didn't help. rewrite didn't help as well despite there are no errors during the installation
So I just gave up and downloaded the bin from some forum))) Works fine.

KevinOConnor, 2020-04-24:
I'm going to close this as it looks like the original poster has found a workaround for the issue.

-Kevin

closed 2020-04-24 by KevinOConnor