Klipper3d/klipper #2729: Can't install klipper properly

jlur3n, 2020-04-14:
Hi
I´ve been using klipper for a couple of months now with ender 3 pro and its been working great. Until this weekend when I tried to implement S-curves and followed the instructions in this [thread]
(https://github.com/KevinOConnor/klipper/pull/2030#issuecomment-568019070) .
(Probably shouldn't since my programming experience is non existing) 
After that the printer screen went blank I could connect but couldnt read the temps any more. 

I made a stupid decision to reinstall everything only now realizing the logs would probably give the answer... (facepalm)

Now I face another problem because it seems like I cannot get all required packages when trying to reinstall klipper on a brand new install of octopi. And I cannot connect to my printer any longer.

This is what putty is showing:

[image]

There is a 30ish more packages failing to install from this server.
Any ideas how to get around the problem?

mental405, 2020-04-14:
This looks like an issue with APT and not Klipper.
You could try running 
``sudo apt-get update
sudo apt-get upgrade`` 
from the terminal and see if it does the same thing.

It could be an issue with your pi's network configuration or possibly the repositories are unreachable. If this has been going on for a few days, it is probably the former, if it started happening today, you might wait a while and try again.

Short of reinstalling the octopi image again, there isn't really a Klipper issue here since that step is a standard OS command to make sure all of your packages are up to date. If the package repos cannot be reached, it is indicative of a bigger issue than the Klipper install script.

jlur3n, 2020-04-14:
Thanks for your answer!
It has been going on for a few days so I can't wait any longer :)
The commands did work better but it still tried the same mirror on some packages.

Fortunately I'm behind a VPN and could change server to another country and by doing so accessing another working mirror. So now everything works again , big thanks!

Oh, and I also re-ran the commands for installing the S curve branch and this time it worked.

closed 2020-04-14 by jlur3n