Klipper3d/klipper #2730: Can't get serial ID after successful SD card flash on SKR Mini E3 v1.2

6moons, 2020-04-14:
Hello. New to Klipper and all of this. I tried to get Klipper to work on my Ender 3 with an SKR Mini E3 v1.2 and RPi 3B, but even after a successful SD card flash (I see firmware.cur on the SD card after flash), using `ls /dev/serial/by-id/*` returns "ls: cannot access '/dev/serial/by-id/*': No such file or directory," so I can't update my printer.cfg file with my unique serial. I sought help from the helpers in the Discord channel -- they had me try `apt-get update` followed by `apt-get upgrade`, `sudo ls -a /dev/serial/by-id/*`, `/dev/serial/by-path/*`, `sudo ls -a /dev/serial/by-path/*`. We then tried `git checkout b921b8a1` and `git checkout tags/v0.8.0`, but neither worked. `dmesg` outputted [this](https://gyazo.com/72e95d869695037a0b44f186a04d55ef) before and after updating/upgrading the linux packages. As far as I and the Discord folks could tell, my [config menu](https://gyazo.com/472e047e6e6afd68055948c4de170f41) lined up with the example file for the SKR Mini E3 v1.2. We troubleshot for roughly 3 hours to no avail, so one of the Discord guys told me to post here. Any help would be appreciated. I look forward to using Klipper.

BlackStump, 2020-04-14:
you have disabled "use USB for communication" in make menuconfig section, did you do that intentionally because you are not using a USB cable to connect the mcu to the pi?

6moons, 2020-04-14:
I have it disabled because it's turned off by default and the example file for my board doesn't say anything about it. I showed the Discord guys and they didn't think anything was wrong with it.

BlackStump, 2020-04-15:
[image]

This is default, you need to check the USB

6moons, 2020-04-16:
it's not ticked by default, but ty.

closed 2020-04-16 by 6moons