Klipper3d/klipper #2731: cutting on longer moves( custom board )

sqrtmo, 2020-04-15:
Hello, 

I am working on my custom controller designed to work with klipper, it is almost ready and I will share my design as soon as possible. 

Everything seems to work, but I am observing a strange behaviour on Z and E axes - for some reason longer movements are executed on steps. In this [video](https://youtu.be/HfXGXPmNh-s) I am executing a 100 mm Z move in a single command - as you can see it makes several pauses before get to the end of the move. Steps are not consistent also. It seems that after reboot the system it is abled to make some clean moves and then starts splitting the movement. It also seems to me that the problem appears only on longer moves Z and E because of the ratio there.

My setup is: STM32F407VGT (100 pin package) connected to raspberry pi 4 UART0(GPIO14 & GPIO15).
I have disabled the serial console on raspberry and set dtoverlaly=UART0 in /boot/config.txt

Any suggestions on what can cause the problem and how to debug?

[klippy.log](https://github.com/KevinOConnor/klipper/files/4479320/klippy.log)

KevinOConnor, 2020-04-15:
I'd guess your run_current is too low for the stepper drivers.  You probably need to tune that.

-Kevin

Spacemarine2018, 2020-04-15:
I'd remove the X-Axis from the system to make the Z-axis spin freely. This way you will be able to make sure it's not a mechanical problem.

sqrtmo, 2020-04-16:
Hello, thanks a lot for the answers! 

Actually I have missed the TMC2130 Rref resistor so my AIN_IREF pin is hanging. 

@KevinOConnor your comment led me in the right direction. 

Thanks for your help. I guess I will have more questions in the future.

closed 2020-04-16 by sqrtmo