Klipper3d/klipper #2732: Problems with arc movements when using offset (g2/g3 does not get executed)

Spacemarine2018, 2020-04-15:
I'm running a simple g-code, which runs fine and behaves as expected:

G1 x63 y61.150
G3 x63.000000 y61.150000 i0.000000 j-2.450000

All it does is go to a certain coordinate and perform a 360° loop there

However, if I set an offset SET_GCODE_OFFSET X=0 Y=10 Z=0, the g3 command can not get processed anymore. Some offsets work, some don't. Offset X=0 y=1 works fine.

The error message from the console is: 
could not tranlate from 'N2 G3 x63.000000 y61.150000 i0.000000 j-2.450000*2'

Without the offset I get no such error message and a correct movement.

(Note: it was directly copied from the console, including the missing s in tranlate)

[klippy.log](https://github.com/KevinOConnor/klipper/files/4482964/klippy.log)

KevinOConnor, 2020-04-19:
Perhaps @achmed20 has some thoughts?

-Kevin

KevinOConnor, 2020-04-23:
I think this should be fixed now (commit 402110f6).

-Kevin

closed 2020-05-02 by KevinOConnor