Klipper3d/klipper #2733: "extruder_stepper" usage & functionality

theophile, 2020-04-16:
I have an Ender 3 to which I have attached 6 extruders, each with its own stepper, and a 6-in-1-out splitter feeding a single-nozzle hotend. It is, shall we say, less reliable than I would like, and I attribute some of the blame to overly long bowden runs (>1m). I had an idea to add a direct-drive extruder right above the hotend, which would be "fed" with filament from one of the six upstream extruders. Then the "feeder" extruder and the "drive" extruder would execute all extrusion moves in sync with one another during the print. This should allow for the same responsiveness at the nozzle as in direct-drive system, while preserving the multi-material functionality.

It looks like the "extruder_stepper" config section is what I want. To test I set one up to have extruder 6 mirror extruder 1, and it worked more or less perfectly. The "drive" extruder consistently loads from the "feeder" extruder and both work perfectly in tandem with one another.

But for this to work, I need the "drive" extruder to be able to sync with _all of the _extruders,__ not just one. Unfortunately, if I try to set up two different "extruder_stepper" sections using the same stepper pins, it tells me I can't reuse pins. Is there any way around this? I feel like I'm _this close_ to making it work!

A second question: Is there a way to define a "tmc2208" section for an extruder_stepper? I tried but it said "my_extra_stepper" was not a known stepper.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4484464/klippy.log)

KevinOConnor, 2020-04-19:
I'd guess the closest equivalent would be one `[extruder]` config section and 6 `[extruder_stepper extra_stepper1]` config sections.  However, that would require the ability to disable the 5 inactive extruder_steppers when they are not in use, which currently isn't implemented.

There was some discussion of this in #2171 and in #2598.  Perhaps @ld3300 has further ideas.

-Kevin

KevinOConnor, 2020-04-19:
To define a tmc2208 config section, it should look like `[tmc2208 extruder_stepper my_extruder_stepper]`.

-Kevin

ld3300, 2020-04-20:
This is actually something on my mind, I am hoping to implement a multi drive system eventually (playing with design ideas).  There is a chance that once you know the direct drive extruder has grabbed the filament you could disable the feeding motor and maybe there won't be enough resistance to cause any issues for the feeding motor.  You would definitely want to use the largest inner diameter tubes between feeder and driver as you can implement to avoid resistance and give some slop room which should make loading and retracting work better.

My plan right now is to have a sensor close to the extruder motor that is a known distance from the extruder, so once I hit that sensor I know I need to go x distance to put pressure against the extruder motor to try and pull the filament.  Haven't worked out all the details yet, but that's one thought.

theophile, 2020-04-20:
I tried disabling the feeder motor, but it still provides enough resistance to cause slippage and inconsistent extrusion. Until someone smarter than me implements a shared extruder system, my other thought is to figure out some way to use a servo to take the tension off the feeder extruder.

ld3300, 2020-04-21:
I might be able to give this a shot, but my projects are building up.  Looking through the code quickly I haven't determined the way to do it, but I think I got a hint when I found the dual stepper setup for things like dual z.  It can't be exactly the same because that can't be changed at run time, but it might get me there eventually.  Probably a native python program could figure it out in a few hours.  I'm a little bit of a guess and test process right now.

theophile, 2020-04-21:
So I had a couple ideas for how this might be implemented. Keep in mind I'm not a programmer so I have no idea how feasible these approaches would be.

1. @KevinOConnor suggested in #2171 that borrowing something like the shared heater concept might be called for.
2. Maybe the existing extruder_stepper implementation could be tweaked so that extruder_stepper could be tied either to a particular stepper _or_ to the active stepper, whatever that happens to be.
3. A variation of the above, maybe an extruder_stepper could be configured to accept an array of extruders, not just one, to sync to.

These ideas assume there is only one extruder active at any given time. If it is theoretically possible for multiple independent extruders to be active simultaneously (such as with a mixing hotend), there would have to be a mechanism for conflict avoidance.

I'm a "guess and test" guy too and I was wondering if # 2 above might be the easiest to try with this approach.

theophile, 2020-04-27:
@sapell, you wrote the extruder_stepper bits. Would it be feasible to allow an extruder to sync to whatever the active extruder is at any given time, assuming there is only one extruder active at a time?

theophile, 2020-04-29:
I had another idea for an approach to possible implementation. Extended gcodes can already be used to override certain parameters that are otherwise defined in printer.cfg, such as (for example) SET_EXTRUDER_STEP_DISTANCE, which will override the step_distance figure from printer.cfg. Maybe something similar could be done to allow the overriding of the "extruder" parameter within an "extruder stepper" section in printer.cfg, something like (for example) SET_STEPPER_EXTRUDER EXTRUDER=<config_name>

Then a toolchange script could do something like this:
[gcode for cooling moves, long retraction of filament from hotend by "extruder"]
ACTIVATE_EXTRUDER EXTRUDER=extruder1
SET_STEPPER_EXTRUDER EXTRUDER=extruder1
[gcode for long insertion of filament and priming of hotend by "extruder1"]

theophile, 2020-04-29:
@ld3300, I was looking more closely at how SET_EXTRUDER_STEP_DISTANCE was implemented to see if I could accomplish the idea in my previous comment, when I noticed that you authored the commit that implemented SET_EXTRUDER_STEP_DISTANCE. Then I re-read @KevinOConnor's initial comment in this thread and realized that that's why he mentioned you here in the first place! 

Do you think it would work to change the extruder that the extruder stepper syncs to "on the fly" using extended gcode, similar to how the SET_EXTRUDER_STEP_DISTANCE functions?

theophile, 2020-04-30:
Okay, I rolled up my sleeves and took a shot at it, and actually got somewhere! Here's what I've come up with so far:

```
--- extruder_stepper.bak	2020-04-29 23:06:55.913775207 +0100
+++ extruder_stepper.py	2020-04-30 02:37:23.703599923 +0100
@@ -9,17 +9,30 @@
 class ExtruderStepper:
     def __init__(self, config):
         self.printer = config.get_printer()
+        self.name = config.get_name()
         self.extruder_name = config.get('extruder', 'extruder')
         self.stepper = stepper.PrinterStepper(config)
         self.stepper.set_max_jerk(9999999.9, 9999999.9)
         self.stepper.setup_itersolve('extruder_stepper_alloc')
         self.printer.register_event_handler("klippy:connect",
                                             self.handle_connect)
+        gcode = self.printer.lookup_object('gcode')        
+        gcode.register_mux_command("SET_EXTRUDER_STEPPER", "EXTRUDER_STEPPER", None,
+                                   self.cmd_SET_EXTRUDER_STEPPER,
+                                   desc=self.cmd_SET_EXTRUDER_STEPPER_help)
     def handle_connect(self):
         extruder = self.printer.lookup_object(self.extruder_name)
         self.stepper.set_trapq(extruder.get_trapq())
         toolhead = self.printer.lookup_object('toolhead')
         toolhead.register_step_generator(self.stepper.generate_steps)
-
+    cmd_SET_EXTRUDER_STEPPER_help = "Set extruder stepper"
+    def cmd_SET_EXTRUDER_STEPPER(self, params):
+        gcode = self.printer.lookup_object('gcode')
+        self.extruder_name = gcode.get_str('EXTRUDER', params)
+        extruder = self.printer.lookup_object(self.extruder_name)     
+        self.stepper.set_trapq(extruder.get_trapq())
+        toolhead = self.printer.lookup_object('toolhead')
+        toolhead.register_step_generator(self.stepper.generate_steps)
+        gcode.respond_info("Extruder stepper now syncing with '%s'" % (self.extruder_name))
 def load_config_prefix(config):
     return ExtruderStepper(config)
```

It kinda works. I can use the gcode command `SET_EXTRUDER_STEPPER EXTRUDER=extruder1` and it will make "my_extra_stepper" begin syncing to "extruder1" instead of "extruder." And as long as the previous extruder's moves ultimately returned that extruder to its starting position, the next extruder (and the newly synced my_extra_stepper) will pick right up in sync.

However, if the previous extruder is not returned to its starting position, then after activating the next extruder and issuing SET_EXTRUDER_STEPPER EXTRUDER=<next_extruder>, upon the next extrude command, the firmware enters shutdown mode with this error:

```
stepcompress o=3 i=0 c=76 a=0: Invalid sequence
stepcompress o=3 i=0 c=76 a=0: Invalid sequence
Exception in flush_handler
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/toolhead.py", line 388, in _flush_handler
    self.flush_step_generation()
  File "/home/pi/klipper/klippy/toolhead.py", line 334, in flush_step_generation
    self.move_queue.flush()
  File "/home/pi/klipper/klippy/toolhead.py", line 171, in flush
    self.toolhead._process_moves(queue[:flush_count])
  File "/home/pi/klipper/klippy/toolhead.py", line 330, in _process_moves
    self._update_move_time(next_move_time)
  File "/home/pi/klipper/klippy/toolhead.py", line 289, in _update_move_time
    m.flush_moves(mcu_flush_time)
  File "/home/pi/klipper/klippy/mcu.py", line 744, in flush_moves
    self._name,))
error: Internal error in MCU 'mcu' stepcompress
Transition to shutdown state: Exception in flush_handler
```

I'm hoping someone who actually knows what they're doing can show me what I'm missing/doing wrong.

sapell, 2020-04-30:
> @sapell, you wrote the extruder_stepper bits. Would it be feasible to allow an extruder to sync to whatever the active extruder is at any given time, assuming there is only one extruder active at a time?

I believe it could be done e.g. if one was to implement a kind of 'virtual extruder' module that would always be (point to?) the currently active extruder. This virtual extruder could then be referred to in the extruder stepper config section. I wouldn't consider this feature to be best implemented as a modification in the extruder stepper module, but rather as a separate module.

It could be worth to consider whether the extruder stepper implementation could be altered such that one could define multiple steppers for one extruder stepper. It should be carefully thought out to not cause any undesired issues (e.g. 'race conditions' if multiple of the synchronized master steppers are stepping simultaneously) with that. Does @KevinOConnor have any thoughts on whether this would be feasible

Anyhow, I will not be able to commit to the actual implementation.

Best regards,
- Simo

KevinOConnor, 2020-05-02:
> However, if the previous extruder is not returned to its starting position, then after activating the next extruder and issuing SET_EXTRUDER_STEPPER EXTRUDER=<next_extruder>, upon the next extrude command, the firmware enters shutdown mode

I suspect you'd need to read the toolhead position (`pos = toolhead.get_position()`), extract the extruder position (`pos[3]`), and then set the stepper's kinematic position accordingly (`stepper.set_position()`).  I haven't looked at this closely though.

-Kevin

theophile, 2020-05-02:
Thank you @KevinOConnor, that got me over one hurdle, although now I'm stuck again after spending the entire day on trial-and-error tactics since I have no idea what I'm doing. Here's the extruder_stepper.py as I've hacked on it to the moment:

```
# Code for supporting multiple steppers in single filament extruder.
#
# Copyright (C) 2019 Simo Apell <simo.apell@live.fi>
#
# This file may be distributed under the terms of the GNU GPLv3 license.
import logging
import stepper

class ExtruderStepper:
    def __init__(self, config):
        self.printer = config.get_printer()
        self.name = config.get_name()
        self.extruder_name = config.get('extruder', 'extruder')
        self.stepper = stepper.PrinterStepper(config)
        self.stepper.set_max_jerk(9999999.9, 9999999.9)
        self.stepper.setup_itersolve('extruder_stepper_alloc')
        self.printer.register_event_handler("klippy:connect",
                                            self.handle_connect)
        self.printer.register_event_handler("extruder:activate_extruder",
                                            self.handle_activate)
    def handle_connect(self):
        extruder = self.printer.lookup_object(self.extruder_name)
        self.stepper.set_trapq(extruder.get_trapq())
        toolhead = self.printer.lookup_object('toolhead')
        toolhead.register_step_generator(self.stepper.generate_steps)
    def handle_activate(self):
        gcode = self.printer.lookup_object('gcode')
        toolhead = self.printer.lookup_object('toolhead') 
        extruder = self.printer.lookup_object('toolhead').get_extruder()       
        pos = toolhead.get_position()
        last_epos = pos[3]
        self.stepper.set_position((0., 0., 0., 'last_epos'))
	self.stepper.set_trapq(extruder.get_trapq())
def load_config_prefix(config):
    return ExtruderStepper(config)
```

In my previous iteration posted above, I had tried to set it up so that a gcode command could be used to set the extruder that the extruder_stepper syncs to. But I ran into problems whenever I had allowed an extruder to step "by itself," independently of the extruder_stepper, so I tweaked things a bit so that every time a new extruder is activated, the extruder_stepper syncs to that new active extruder.

Currently, this works so that I can step any desired distances/directions on 'extruder,' then without having to return it to "0," I can activate 'extruder1' and it will pick up stepping in sync with the extruder_stepper. I can move on to 'extruder2,' 'extruder3,' and so on in the same manner, with the extruder_stepper staying in sync with each in turn.

But if I ever go _back_ to an extruder that had previously moved, upon the next extrusion move, the firmware enters shutdown with a similar "invalid sequence" error like so:

```
stepcompress o=3 i=0 c=1193 a=0: Invalid sequence
Exception in flush_handler
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/toolhead.py", line 388, in _flush_handler
    self.flush_step_generation()
  File "/home/pi/klipper/klippy/toolhead.py", line 334, in flush_step_generation
    self.move_queue.flush()
  File "/home/pi/klipper/klippy/toolhead.py", line 171, in flush
    self.toolhead._process_moves(queue[:flush_count])
  File "/home/pi/klipper/klippy/toolhead.py", line 330, in _process_moves
    self._update_move_time(next_move_time)
  File "/home/pi/klipper/klippy/toolhead.py", line 289, in _update_move_time
    m.flush_moves(mcu_flush_time)
  File "/home/pi/klipper/klippy/mcu.py", line 744, in flush_moves
    self._name,))
error: Internal error in MCU 'mcu' stepcompress
Transition to shutdown state: Exception in flush_handler
Dumping gcode input 17 blocks
Read 1110424.716467: 'N0 M110 N0*125\n'
Read 1110424.751569: 'N0 M110 N0*125\n'
Read 1110424.827839: 'N0 M110 N0*125\n'
Read 1110424.836284: 'N1 M115*39\n'
Read 1110424.854868: 'N2 M21*18\n'
[... 27 lines cut ...]
```
Recv: // mcu: stepper_x:0 stepper_y:0 stepper_z:0
Recv: // stepper: stepper_x:0.000000 stepper_y:0.000000 stepper_z:0.000000
Recv: // kinematic: X:0.000000 Y:0.000000 Z:0.000000
Recv: // toolhead: X:0.000000 Y:0.000000 Z:0.000000 E:10.000000
Recv: // gcode: X:0.000000 Y:0.000000 Z:-0.493333 E:10.000000
Recv: // gcode base: X:0.000000 Y:0.000000 Z:0.000000 E:0.000000
Recv: // gcode homing: X:0.000000 Y:0.000000 Z:0.000000
``` 

Then, if I execute `ACTIVATE_EXTRUDER EXTRUDER=extruder1`, followed by `ACTIVATE_EXTRUDER EXTRUDER=extruder`, and then another `GET_POSITION`, I get this:

```
Recv: // mcu: stepper_x:0 stepper_y:0 stepper_z:0
Recv: // stepper: stepper_x:0.000000 stepper_y:0.000000 stepper_z:0.000000
Recv: // kinematic: X:0.000000 Y:0.000000 Z:0.000000
Recv: // toolhead: X:0.000000 Y:0.000000 Z:0.000000 E:9.995664
Recv: // gcode: X:0.000000 Y:0.000000 Z:-0.493333 E:9.995664
Recv: // gcode base: X:0.000000 Y:0.000000 Z:0.000000 E:9.995664
Recv: // gcode homing: X:0.000000 Y:0.000000 Z:0.000000
```

Though there have been no move commands, simply switching to a different extruder and back again causes the firmware to return a different E position. Maybe this has something to do with it? To be fair, I tried completely disabling the "extruder_stepper" section in my config to rule out my own meddling, and these value shifts still occurred, so the rest of the firmware doesn't seem to be bothered by it.

I would be grateful for any further assistance anyone might be able to provide in helping bring this across the finish line. Thanks all!

KevinOConnor, 2020-05-03:
Those stepcompress errors are because the stepper got out of sync with its position.  Alas, I don't know "off the top of my head" why that would be.

At a minimum though, in retrospect, using the toolhead.get_position() is not a good idea.  What you probably want is something like `self.stepper.set_position([extruder.stepper.get_commanded_position(), 0., 0.])`.

-Kevin

theophile, 2020-05-03:
Thank you @KevinOConnor! That was the missing link. I've got it working reliably now so that in addition to being able to define a fixed 'extruder_stepper' in the config as currently, I can also put a "sync_to_active: True" in the config to have it always sync the 'extruder_stepper' to the active extruder, _and_ I can link any 'extruder_stepper' to any 'extruder' on the fly with gcode.

Now to see if I can submit a proper pull request...

KevinOConnor, 2020-05-24:
Okay, lets close this then as it seems like the conversation has moved to #2824 . 

-Kevin

closed 2020-05-24 by KevinOConnor