Klipper3d/klipper #2738: Klippy crash when probing from Z=0

proto3, 2020-04-17:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4492140/klippy.log)

Klippy crashes when PROBE is called from Z lowest position. The problem looks clear, there is a divide by zero error while calculating probing rate, but I'm not sure on how to fix it properly.

*File "/home/lucas/projects/klipper_proj/klipper/klippy/homing.py", line 42, in _calc_endstop_rate
    return move_t / max_steps
ZeroDivisionError: float division by zero*

ld3300, 2020-04-18:
What is the probe for?  It isn't set up as being for any of your axes.  ie:
```
# Define a probe
[probe]
pin: ar30
z_offset: 2.345

# Example settings to add to stepper_z section
[stepper_z]
endstop_pin: probe:z_virtual_endstop
position_min: -2
```
How does your probe have a Z offset of zero from your nozzle?

We need to know a lot more about your printer setup to be much help.  What is your printer? what is your probe?  What is your probe setup like?  Is there a reason the probe isn't also your endstop?

proto3, 2020-04-19:
Just to put it into context, I'm currently working on a fork of Klipper dedicated to CNC plasma cutter. Everything which is related to plasma stay on my fork, but in cases that are generic enough, I try to push it to the official Klipper and rebase my fork on it.

This explain my zero offset because in my case the "nozzle" acts as a probe by electrical contact. I have to use PROBE prior to every cut to detect workpiece and position the plasma torch at the correct height.
My endstop is different from the probe and it is placed on top of Z axis to prevent crash on material while homing.

I made this fix #2750, which works fine for me. Now _calc_endstop_rate just return 1.0 default value when distance is 0 and rate has no meaning.

KevinOConnor, 2020-04-19:
Thanks.  I think this should be fixed now (commit 06846313).

-Kevin

proto3, 2020-04-19:
It works fine now, I close the issue.

closed 2020-04-19 by proto3

ld3300, 2020-04-20:
That's a cool project.  Thanks for sharing the details.