Klipper3d/klipper #2739: Parallel motors/split sensors for 28 extrudes printing simultaneously

countAleister, 2020-04-17:
Dear all,

I'm looking for the solution to run the machine featuring 28 extruders printing simultaneously the same part so we will be having 28 extruder motors each with it's own driver but connected parallel except En signal. 

Picture for better concept understanding:
[image]

But at the same time will be having 28 Heater sensors which have to be monitored separately by Klipper as well as in Octoprint. Here is how I see it all should look like:

mental405, 2020-04-19:
without actually trying it. i suspect you can create 27 [heater hotend_n] sections, 27 [extruder_stepper extruder_n] sections and one [extruder]

KevinOConnor, 2020-04-19:
FWIW, it sounds like this is a significant project and I suspect you'll want a developer prepared to make code changes for it.

That said, it should be possible to define an `[extruder]` config section along with 27 `[extruder_stepper my_extra_stepper1]` type config sections.  It's also possible to define 27 `[heater_generic my_extra_heater1]` config sections to control each heater temperature.  This doesn't require any code changes, but would require setting up the appropriate gcode macros to bring all the heaters up to the desired printing temperature.  See the corresponding descriptions for extruder_stepper, heater_generic, and gcode_macro in config/example-extras.cfg for further details.

> Also distributes stp/dir signal to all the Extruder boards MCUs connected parallel.

FYI, that will be harder to make work with Klipper.  For Klipper, you'd be better off running 28 usb connections to the rpi (or 28 serial connections, or possibly 28 CAN bus connections if using the stm32f0).  The Klipper host software works best when it can directly communicate with all the micro-controllers.

-Kevin

countAleister, 2020-04-20:
@KevinOConnor 

Dear Kevin, thank you for your reply. Yes, ya right, it would be great to hire someone familiar with your solution to customise it for our own needs. If you can point on someone who would be interested to it will be highly appreciated.

countAleister, 2020-05-04:
Still looking for someone to exchange some money to the solution.

gnoblin, 2020-05-04:
what do you need 28 extruders in parallel for? I mean in terms of the actual product - what are you trying to solve? Thanks

countAleister, 2020-05-04:
> what do you need 28 extruders in parallel for? I mean in terms of the actual product - what are you trying to solve? Thanks

To print the same model multiplied to 28 ))))) My pleasure ))))

mental405, 2020-05-05:
28 baby yodas... at the same time

KevinOConnor, 2020-05-24:
I'm going to close this as it looks like the conversation has concluded.  Alas, I don't personally know of anyone.

Cheers,
-Kevin

closed 2020-05-24 by KevinOConnor