Klipper3d/klipper #2740: Usability: Must home before starting print

nstephenh, 2020-04-18:
When starting a print with octoklipper, if I haven't homed the printer since the last print completed, the printer returns a Must home axis first: 0.000 180.000 20.000 [61085.740] command and disconnects. When reconnecting it returns a m112 error and disconnects. I've had this happen multiple times in a row. Eventually it will reconnect and restarting the firmware and re-homing before starting the print works fine. 

My thoughts, after a day or so with klipper, is that this doesn't seem like an optimized workflow. My marlin printers auto-home as soon as I click print, instead of emergency stopping the printer and preventing it from re-connecting.

GarthSnyder, 2020-04-18:
This isn't normal. You are sending a G-code command that requires the axes to be homed, probably from your slicer or OctoPrint preamble. Look in the OctoPrint's Terminal pane - what is the actual command that precedes the "must home" error message?

Klipper seems to be a bit more stringent about requiring homing than Marlin, which explains why the same code might work under Marlin but not Klipper. My guess is that when you find and understand the issue, you will wonder why Marlin allows it without homing.

BlackStump, 2020-04-18:
In Octoprint the default settings do not give a new user a great experience, so you need to go through Octoprints settings to make sure it does not issue random M112 also as per the klipper install instructions make sure you did not miss this step in configuring Octoprint for Klipper
>In the Settings tab, navigate to the “Behavior” sub-tab and select the “Cancel any ongoing prints >but stay connected to the printer” option. Click “Save”.

You do not want Octoprint disconnecting from Klipper, Klipper has all the safety functions you need so avoid Octoprint taking on that role.

nstephenh, 2020-04-20:
@BlackStump I disabled that setting and it now doesn't disconnect. I still have to home it manually though. Thank you.

Nandox7, 2020-04-20:
> This isn't normal. You are sending a G-code command that requires the axes to be homed, probably from your slicer or OctoPrint preamble. Look in the OctoPrint's Terminal pane - what is the actual [...]

Note quite as I've used not Mrlin but RepetierFW for long time where this is possible and instead I wonder why Klipper does not allow it.
While is not something you'd be using for every single print but there are several instances where this is helpful. 
Like you need to emergency stop a print, carriage is somewhere in the bed, with RepetierFW I'd simply raise the Z, clean whatever mess was made, home and continue. With klipper I have the manually raise the Z axis (in my case rotating both leadscrews manually, clean the mess, hope the X axis is still leveled, home and level out the X axis.

GarthSnyder, 2020-04-20:
> @Nandox7  wrote: ...there are several instances where this is helpful. Like you need to emergency stop a print, carriage is somewhere in the bed, with RepetierFW I'd simply raise the Z, clean whatever mess was made, home and continue. With klipper I have the manually raise the Z axis (in my case rotating both leadscrews manually, clean the mess, hope the X axis is still leveled, home and level out the X axis.

I believe you can use the MANUAL_STEPPER command to achieve this effect in Klipper.

Nandox7, 2020-04-21:
Just read on MANUAL_STEPPER.
Doesn't seem to be an option, if I understood correct it would be for stepper drivers that are working outside of the kinematics definition of Klipper. In my example all the steppers are all already in use.

# Manual steppers (one may define any number of sections with a
# "manual_stepper" prefix). These are steppers that are controlled by
# the MANUAL_STEPPER g-code command. For example: "MANUAL_STEPPER
# STEPPER=my_stepper MOVE=10 SPEED=5". See the docs/G-Codes.md file
# for a description of the MANUAL_STEPPER command. The steppers are
# not connected to the normal printer kinematics.

closed 2020-04-25 by klipper-gitissuebot