Klipper3d/klipper #2741: Unable to read tmc uart 'stepper_x' register IFCNT, skr e3 mini v1.2

pengsloth, 2020-04-18:
Klipper has been working amazingly. I turned the printer on (an Ender 3 with an skr e3 mini v1.2) and got the following error. I wouldn't think this is actually an error with Klipper but I don't know how to troubleshoot the issue...

Recv: // Unable to read tmc uart 'stepper_x' register IFCNT
Recv: // Once the underlying issue is corrected, use the "RESTART"
Recv: // command to reload the config and restart the host software.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4495993/klippy.log)

KevinOConnor, 2020-04-19:
This means that Klipper was unable to communicate with a tmc2208 stepper motor driver.  This often happens when power is applied to the micro-controller, but not to the stepper motors.  Make sure power is fully up and double check the wiring.

-Kevin

pengsloth, 2020-04-22:
I've double-checked the wiring and power. I've even connected the X axis wires to a spare motor and still receive the same error. 

Switching the pin connections between the Y and X axis throws an error on the Y and not the X. So I think, being very new to 3D Printers, that it has something to do with the X driver itself.

KevinOConnor, 2020-04-24:
It sounds like there is a problem with the X stepper driver then - I'd triple check the wiring (look for shorts or loose/frayed wires/connectors).  It's also possible the X driver got damaged and is now inoperable.

-Kevin

CroveR99, 2020-04-24:
If you have drivers as like btt v3,0 you must soldered this 
[image].. same problem have got my friend and this solver it

pengsloth, 2020-04-24:
This is with the SKR E3 mini v1.2 it should be, and has been, plug in play
with no modification nessassary.

KevinOConnor, 2020-05-11:
I'm going to close this as it looks like the original question was answered (it appears to be a hardware issue with the X driver).

-Kevin

closed 2020-05-11 by KevinOConnor

jschuh, 2020-05-31:
Sorry for commenting on a closed bug, but this just happened to me and I wanted to leave some additional detail for anyone who hits this error on this specific board. Apparently this is one of the more common [hardware failures for the SKR Mini E3](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/155) (and it seems to typically be the X driver). As @KevinOConnor stated above, this means you have a dead TMC2209 driver, and you're going to need to replace the board. I had a working Marlin build for the same system, so I reflashed and verified from Pronterface (just because I wanted to double check that the driver was really dead).