Klipper3d/klipper #2744: No /tmp/printer nor /tmp/klippy.log

Hairyloon, 2020-04-18:
After putting /tmp/printer in the box, Octoprint won't give the option to select it.
Possibly this is because /tmp/printer does not exist.
I don't include a log, because the log does not exist: I'm guessing that klipper is failing to write to /tmp.
Possibly this is because I have done something stupid.

Flashing the firmware doesn't appear to have done what it should either, but I'll get back to you on that...

ld3300, 2020-04-18:
hmm, should be able to get a klippy.log even if the board isn't connected.  Does klipper show up if you run ps -A

Hairyloon, 2020-04-18:
I started again from scratch and that much of it worked this time...
Not got the printer to respond yet though...

Hairyloon, 2020-04-18:
> hmm, should be able to get a klippy.log even if the board isn't connected. Does klipper show up if you run ps -A

I started again from scratch and it worked, so I can no longer check that.
Thanks for trying.

ld3300, 2020-04-20:
What does the klippy log say now that you have the base klipper running?

Hairyloon, 2020-04-20:
It says the usual kind of stuff that you'd expect it to say: it is more or less working now.
This one is a problem that has gone away for me.

closed 2020-04-20 by Hairyloon

ld3300, 2020-04-21:
If you have any insights that might help someone else in the future please feel free to share.
Glad it seems things are working for you.

Hairyloon, 2020-04-21:
As I said in OP, it looks like klipper didn't have permission to write to /tmp.

My guess is that is because I didn't install as pi: I was under the impression that a root superuser was a root superuser, but apparently that is not the case.