Klipper3d/klipper #2745: [FEATURE] Extruder maximum volumetric speed?

mcmatrix, 2020-04-18:
I tried a larger nozzle today and stumbled upon the maximum volumetric rate limitation of the hotend.
It means that every hotend has a limited capacity which is expressed as the “maximum volumetric speed” it can handle. If you attempt to move more filament than this through the hotend, you’ll eventually encounter a variety of problems, including:
Low extrusion and uneven layers.
Extruder click and skips.
Nozzle and hotend jams.
Increased extruder motor heat.

[Calibrating maximum volumetric rate](http://projects.ttlexceeded.com/3dprinting_calibrating_volumetric_rate.html)

Some slicers have this maximum volumetric speed limit builtin (prusaslic3r for example)

I was wondering if this kind of feature is needed in Klipper?
Let's say a change in extruder kinematics to limit print move speed (if the user wants).
User can specify maximum volumetric speed in config or dynamically via gcode for the extruder. The unit is cubic mm in sec. By setting this limit the extruder will check a print move
and if the extruder volumetric speed will exceed the limit then the print move speed will be clamped down to the safe range.

ld3300, 2020-04-18:
Right now the nozzle size in klipper doesn't do much, it sets the maximum mm^2 extrusion, and I think not much else.
Do you have a different profile in your slicer for the nozzle size?  Since the extrusion amount and speed given to klipper as g-code commands it would be difficult to anticipate.  It is possible that the behavior of an square extrusion rate could implement a printing slow down.  I think that would be a semi-decent change to the move checking, and it may be too on the fly to anticipate and change.  The article you link talks about setting up the configuration in a slicer.  
I guess we could make it so the nozzle size could be changed after initial config load and allow a config section to define a ratio of print speed to nozzle size.  This would, however, just uniformly slow down the print regardless of need.  Wouldn't setting up the calibration in the slicer be more effective so the speed implementation could be slowed down only when needed?

Hywelmartin, 2020-04-18:
As mentioned in the article.. this is a feature that is indeed a part of slic3r/prusaslicer..

When we are at it..there is a buffert inside the hotend that can, on short moves, greatly exceed the maximum flow... for example my hotend is capable of 35mm³/s continuously but can burst up to 50 for shorter moves(it all depends on the model). 

The hotend capacity isn't determent by the nozzle size.. Extruder might be but I haven't seen that ..

mcmatrix, 2020-04-18:
I think the usage of volumetric rate will only slow down when needed.

My idea was that user can set maximum volumetric rate for the extruder.
Let's say after “free air” testing I'll find out that at temp 220C the extruder is able to 
push my filament at rate MVS=10.03mm<sup>3</sup>/s without skipping.
This max rate is set to the extruder before printing.
From the MVS I'm able to get safe extruder speed by using filament cross-section area.
safe speed = MVS / (pi * (Filament Diameter/2)<sup>2</sup>)

The extruder kinematics will anyway check each print move.
We can get the overall move length, how much plastic in mm will be extruded and at what speed. From this, we can calculate the volumetric rate for this concrete move and what will be the extrusion speed for this print move.
If the speed or rate will exceed the MVS then we just limit down this movement speed that so extrusion speed will be within the safe range.

lbibass, 2020-04-19:
Yeah, I agree. It's better to do that in the slicer, as the max flowrate can change drastically depending on the filament and temperature.

mcmatrix, 2020-04-19:
> Yeah, I agree. It's better to do that in the slicer, as the max flowrate can change drastically depending on the filament and temperature.

Yes, that's true. It's depending on different factors. It probably needs some additional GCODE that allows setting this flow limit dynamically before starting printing or even during printing if needed.

KevinOConnor, 2020-04-19:
FWIW, I'd say this is functionality better located in the slicer than in the firmware.  The slicer has better information (eg, heating rate and fan cooling rate) and is able to do better planning.

-Kevin

KevinOConnor, 2020-05-11:
I'm going to close this as it looks like the conversation has concluded.  Let us know if you get interesting results from your experiment.

-Kevin

closed 2020-05-11 by KevinOConnor

mcmatrix, 2020-05-11:
Thanks. I'll leave this topic for now.
I'm not pushing my printer that hard, trying to keep all speeds below skipping point :)

By testing, I got safe volumetric speed for my nozzle (0.6) at printing temperature.
It's around 12mm3/s and then calculated safe print speeds from that for different layer heights.
This seems to work quite well.

W-Floyd, 2021-07-13:
I understand why this is not going to be implemented, but an applicable situation is in a print farm with similar printers. In my case, some printers have less capable hotends that can sustain only a lower flow rate - rather than reslicing the model for each printer (which are otherwise identical), being able to set the flow rate would be useful. That being said, setting a low maximum speed is a good enough workaround in my case.