Klipper3d/klipper #2751: No manual control.

Hairyloon, 2020-04-19:
I have klipper mostly working: it printed a near perfect calibration cube yesterday, very slowly, but I think I sliced it too fine.

But I am having more serious issues, specifically the same as [url=https://github.com/KevinOConnor/klipper/issues/2132]this chap[/url], but that's causing added trouble because when the error occurs, the printer just stops where it is and the thing won't give back control to Octopi so I can move the print head out of the work.
There have been some occasions where I have had control, but I've not spotted any pattern to when it does.

Also, is there some way to change the "Home" position so that the bed comes this way instead of going that way?

ETA: extruder now showing "-95° ", so I think perhaps my thermistor is broken...

lbibass, 2020-04-19:
Yeah, it's probably your wiring that's causing some problems. The reason you can't control it is that it's a fatal error. If your thermistor fails,  it locks up the printer and shuts down all the heaters. This is to prevent fires. Look at the Anet A8, and all the fires it has caused if you want to see what happens if you don't have thermal runaway protection.

Hairyloon, 2020-04-20:
That is the obvious answer, and explains why it was reading -95°: I'd not put it back properly when I looked for a fault.
But I don't think it was the cause of the original problem because it printed some things fine, but consistently tripped out on others.
So I think there is some kind of bug in the G-code, or in my printer's interpretation of it.
I don't think this is a klipper related issue because it happens whatever I'm driving it with.

Hairyloon, 2020-04-20:
The point here though, since it has happened again, is that I have no means to control the printer after it has crashed.
Which means that I have no way to pull the print head off the work. Last time I ended up re-flashing the Marlin firmware because klipper kept trying to drive the print head down through the bed and I couldn't see another way around it apart from winding it up by hand.

ld3300, 2020-04-21:
Is that the entirety of your klippy log?  Can you update a new one.  It looks like it is unhappy with the pin aliases for arduino.

Hairyloon, 2020-04-21:
> Is that the entirety of your klippy log? 
I may have started a new log in between. Sorry.
>Can you update a new one. It looks like it is unhappy with the pin aliases for arduino.
I can, but if the problem isn't repeating then is it likely to show the problem?
Here's a clean log.  I could try to get the problem to repeat, but I'm not sure that I want to do that...
It is now printing nearly OK:
[image]
I don't know what caused that layer shift, but I suspect the printer rather than klipper.
[klippy.log](https://github.com/KevinOConnor/klipper/files/4510725/klippy.log)
[image]

ld3300, 2020-04-22:
if you re using octoprint with the print layer progress plugin there is a recent report of offset issues that seem to be caused by it.  I think they are still investigating why.

Hairyloon, 2020-04-22:
Unless they've put that into the default package, then I don't think that was it: I've not installed it AFAIK.
It's not done it again, maybe somebody joggled the table or something.

closed 2020-05-11 by KevinOConnor

Hairyloon, 2020-05-11:
> I'm going to close this as it looks like the original reporter found an explanation.
> 
> -Kevin

Not so much an explanation as a distraction: if the print fails with the print head above the model, I've still got no way to move it up out of the way: Klipper insists on homing before it'll move anything and that means going through the model...

Yes, I found an explanation as to why the print kept failing, so I haven't had the same problem since, but that doesn't change the original issue.