Klipper3d/klipper #2752: Feature Request: STM32 i2c read support

trevjonez, 2020-04-19:
As the title suggests, I would like to request i2c read support on STM32 controllers.

My immediate needs have workarounds available (add another mcu config that does support it) but in the longer term I would like to try and consolidate the configuration to only use a single MCU for a cleaner wiring configuration.

KevinOConnor, 2020-04-21:
Sounds useful, but an interested developer would need to take this on and submit it.

-Kevin

ld3300, 2020-04-21:
I have this goal to make a generic_i2c device that would be setup in the config file that would have a few features.  My idea is that it would be a register/data type of communication system and that the user could implement a pin as an i2c_attn line that would trigger the klipper host to read data when it is available, so a constant communication sequence doesn't have to be performed.  My personal interest is a matrixed button board I would like to implement for control, but would require both mcu level and higher level changes to implement.  Since i2c is already present I was hoping to build on that.  It could potentially, on a more general level, enable easier implementation control of special situations Where an extra processor could handle tasks, IO, etc. without needing additional mcu level support. It will take me quite a bit of digging around to figure it out.  I do have a weak start on it.

KevinOConnor, 2020-05-05:
FYI - @Arksine added this in PR #2830 .

-Kevin

closed 2020-05-05 by trevjonez